MarlinFirmware/Marlin #18781: [BUG] Garbled Screen when SDSUPPORT enabled

malcolmdsmith1, 2020-07-25:
Hi, 

**SETUP:** 
- RAMPS 1.4 board
- Adruino Mega 2560
- Discount LCD screen

**Issue:**
I am trying to enable the SD card to run gcode files from it. My screen gets garbled when I uncomment "#define_SDSupport". I tried slowing SPI transfer speed and that did not help. Screen works normal when this is commented (I had to add delay to initially make the screen work).
See attached txt file of my configuration file. Thanks!

[Configuration copy.txt](https://github.com/MarlinFirmware/Marlin/files/4976684/Configuration.copy.txt)

ellensp, 2020-07-26:
you are running a very old version of marlin.   (020000) 
You need to use a current version of marlin.

malcolmdsmith1, 2020-07-26:
Oh, I was under the impression this was the most up to date version - it’s the one I downloaded from their “latest release” link on the website.

ellensp, 2020-07-26:
https://marlinfw.org/meta/download/ 
Latest release comes with Config files  CONFIGURATION_H_VERSION 020005

boelle, 2020-07-26:
https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x

5 hours old

malcolmdsmith1, 2020-07-26:
Thanks! Ok, so it looks as though that allowed the SD card to show up on the menu, but now the screen is all garbled again and the delays that worked before are not anymore (I also tried other delay settings).

Also, when I try to access the menu it sends me back to the home screen and the encoder button will beep if I press it (doesn't do anything else). See attached config.h text file.

Thanks again!

[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/4978697/Configuration.h.txt)

closed 2020-09-01 by github-actions[bot]