MarlinFirmware/Marlin #18788: [FR] New symbol for Feedrate in LCD

positron96, 2020-07-26:
Am I the only one who finds "Fr" icon not very beautiful?
In my spare time, I've created a new symbol for feedrate, looks like "speed" or "fast forward" icon on consumer electronics. I believe it represents what changing feedrate percentage does - speeds all things up or slows them all down. 
If the community is interested, I can provide a pull request with changes to font.
[image]

SparkyDan555, 2020-07-26:
Nice work! Looks much better. Would love to see this implemented. Would probably prevent some confusion with "Flow Rate" for some beginners too!

sjasonsmith, 2020-07-28:
I was skeptical, but after I see if I like it. I agree that it reduced "Flow Rate" confusion. It is also language-agnostic.
I suggest you go ahead and post the pull request. The worst that will happen is you learn whether people like it!

sjasonsmith, 2020-11-12:
Looks like this is done.

closed 2020-11-12 by sjasonsmith