MarlinFirmware/Marlin #18790: [BUG] High reported print speed and unusual behaviour on Ender 3 V2

turmoni, 2020-07-26:
### Bug Description

When attempting to print many (all?) of my G-code files on my Ender 3 V2 with BLTouch connected (via the BLTouch port), the display reports a high print speed and printing behaviour is unusual (e.g. print head trying to move off the X axis at high speed, printing a single line on the right hand side, and freezing).

### My Configurations

I've tried various configurations (including stock example config), but have attached [the one I'm most recently using](https://github.com/MarlinFirmware/Marlin/files/4977522/ender3v2conf.zip).

### Steps to Reproduce

1. With an Ender 3 V2 with firmware from the bugfix-2.0.x branch as of bc688f27dc6db76021da584bc529f29ddb6f6cce (note: this is not when it started happening, just a state of the tree I know has this issue), attempt to print [the attached G-code file](https://github.com/MarlinFirmware/Marlin/files/4977521/smart_compact_temp_calibration.gcode.gz)

2. Observe the print speed on the display - it reports for me as 271%
3. Attempt to adjust the speed, observe as it goes to 999% on the first adjustment
4. Exit the tweak menu, observe as the print speed is reset to its original value
5. Cancel the print before your printer does something bad

**Expected behavior:** Successful printing at 100% speed, with the option to vary as needed

**Actual behavior:** As described in reproduction steps, inappropriate print speed, unable to change the print speed, printing not behaving as expected (either too fast, or freezing, or printing a line on the edge - I'm generally hesitant to let it run too long to find out exactly what it does)

#### Additional Information
From my investigations, the print speed is represented by the variable `feedrate_percentage` and that doesn't seem to be intentionally affected by anything other than user input and M220 (which is not in my G-code).

I have been able to reproduce this on every non-Creality build of Marlin I've tried, including my own with BLTouch enabled, my own without BLTouch enabled, and ones from GitHub. This behaviour occurs regardless of whether the BLTouch is plugged in or not. I've attempted to reset the settings and save to EEPROM multiple times, with no difference.

I wondered if there might be some weirdness with my SD card, so I created a new filesystem on it with just the single G-code file I've attached here and the problem remained.

The G-code is generated by SuperSlicer, which has this to say about the model:

> This tower is made with the 3D model created by gaaZolee with the CC BY-SA 3.0 licence.

So I believe the G-code would also be covered by this licence.

fleek, 2020-07-26:
Have you tried deleting the EEPROM.DAT file from the SD card?  I had this behaviour before.  But I realised that I need to delete the EEPROM.DAT file from the SD card whenever I need to upload new firmware.  Otherwise, the printer will behave erratically.

turmoni, 2020-07-26:
Thanks. Yes, I tried that, and also had this behaviour since https://github.com/MarlinFirmware/Marlin/pull/18758 went in (and before that after using a fresh filesystem on my SD card). I'm willing to believe there might be some state somewhere that's messing it up, but I've not been able to find it yet.

CRCinAU, 2020-07-26:
I'm also getting multiple reports of this - but haven't had a chance to sit down and debug it yet - so it doesn't seem to be an isolated issue... Will add more info as I discover further...

CRCinAU, 2020-07-26:
I'm getting reports that renaming the file to a smaller filename on the SD Card resolves this issue...

@turmoni - Are you able to confirm this at all?

If so, is this something as silly as a buffer overrun in something that handles the filename?

turmoni, 2020-07-26:
> I'm getting reports that renaming the file to a smaller filename on the SD Card resolves this issue...
> 
> @turmoni - Are you able to confirm this at all?

Yes, renaming the file to sctc.gcode seems to make it behave! (I didn't let it do a full print because I'm in the process of going to bed, but it's looking promising.)

I seem to have a separate issue with sorting so actually selecting the right file was tough, but I currently have three G-code files on the SD card - `smart_compact_temp_calibration.gcode`, `HotEnd_Insert_3mm.gcode`, and `sctc.gcode`. It _looks_ like `HotEnd_Insert_3mm.gcode` would also print correctly, and this may be entirely coincidence, but whilst being fairly long it fits on the display without needing to be truncated.

CRCinAU, 2020-07-26:
When you get some time, try seeing if you can figure out the maximum length of the filename before it goes boom...

Take an existing file, try something like 12345678901234567890123.gcode

My guess right now is that 24 characters or more causes it to go boom... I'm not sure if that includes the .gcode part or not...

turmoni, 2020-07-26:
The sorting issue is making it very hard to verify, since I'm not getting the file I selected (unsure if it might be related or not), but I think `twentythreetwentythreee.gcode` (23 characters before .gcode) still has an issue. Tomorrow I'll try `eighteeeneighteeen.gcode` and `sventeenseventeen.gcode`, although I'm sure I saw at least one of them seemingly working fine (which I'm aware isn't very diagnostically useful, but it's definitely bedtime).

CRCinAU, 2020-07-26:
@fleek - If you want to run your eyeballs over it, I'm thinking it has to do with this definition:
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/lcd/dwin/dwin.cpp#L84

Not having the physical hardware in front of me, I'm kinda guessing at the code bug...

CRCinAU, 2020-07-27:
I've been advised that the issue occurs at any filename length of 22 characters + the .gcode extension or longer.

Chapyboy, 2020-07-31:
> I've been advised that the issue occurs at any filename length of 22 characters + the .gcode extension or longer.
I can confirm this works, while skeptical at first I renamed Gcodes that were behaving in the same manner as you described. a quick rename and prints are coming out great. 
during one of my failed prints I had the printer connected via USB to the computer to try and see what was going on when it exhibited the glitch, at that instance absolutely no commands from pronterface would go though, ultimatly leading to a manual reset less damage to the printer

turmoni, 2020-08-01:
Just because I can, I thought I'd change `MENU_CHAR_LIMIT` and the only difference seems to be that with a lower value, less of the file name is shown on the menu screen. It doesn't seem to change which files trigger this issue.

CRCinAU, 2020-08-01:
> Just because I can, I thought I'd change `MENU_CHAR_LIMIT` and the only difference seems to be that with a lower value, less of the file name is shown on the menu screen. It doesn't seem to change which files trigger this issue.

Thanks for testing this.... I still can't get my hands on one of these to test right now - can't even place an order due to everything being out of stock.... Means my testing of theories is massively hampered without hands on hardware....

@thinkyhead  - I know you've got one on your desk - any chance you could take a look at this? I believe its something to do with a buffer overrun based on the length of the filename...

fleek, 2020-08-01:
Won't work even if you change the MENU_CHAR_LIMIT, its hardcorded for short filename to be 13 due to vfat, so although it can be displayed I believe there will be problems reading the file from the card.

ClockeNessMnstr, 2020-08-03:
with names exactly 22 or more characters my control panel is locking up completely immediately following G29 in the start code.  
Speed just says 100% need to turn off and on the power to recover. 
Sounds like part of this issue. 

I'm running nightly builds:
Ender3-V2-BLTouch-20200731.bin
Ender3-V2-BLTouch-20200801.bin

akfish, 2020-09-03:
I encountered some crazy behaviors with my custom firmware build for ender v2. Unfortunately I didn't realize it was relevant to this issue soon enough. My printer crashes the extruder into the frame multiple times during my 2 days of tinkering. 

Durning my painful process of troubleshooting. I observed that if I start SD print with g-codes M20, M23 and M24 manually from pronterface, it will print properly. I think the only difference here is that M20 give 8.3 format filenames:

```
>>>m20
SENDING:M20
Begin file list
HALF_D~1.GCO 2174948
FLAT_S~1.GCO 1357963
DG_PIN~1.GCO 946490
```

ellensp, 2020-09-03:
In Marlin/src/inc/Conditionals_post.h  is 
```
#if EITHER(SCROLL_LONG_FILENAMES, DWIN_CREALITY_LCD)	
  #define MAX_VFAT_ENTRIES (5)
#else
  #define MAX_VFAT_ENTRIES (2)
```
If this test failed, you would get 2 * 13 char long names maximum, ie 26 char
Perhaps you could try setting #define MAX_VFAT_ENTRIES (5) outside of this #if EITHER and see if that changes anything.

cosmoderp, 2020-09-05:
`void HMI_SelectFile(void)` inside `dwin.ccp` handles the file.  I have been making custom menus for this machine and have been able to faithfully reproduce this crash by having a menu option that sends a `G1 X10 Y10`

so first I simply try to print a long file name, then I cancel, and finally try running the G1 command and it crashes 100% of the time. with short file names this does not occur. 

also tried simply:
`#define MAX_VFAT_ENTRIES (5)` in `Conditionals_post.h`
and `"#include "../../inc/Conditionals_post.h"` in `dwin.cpp`
this did not work...still crashes

I think this is some sort of buffer issue as well but I dont think it can be entirely blamed on DWIN.

here are my own configs if it helps.
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/5177587/configs.zip)

thisiskeithb, 2020-09-05:
For those following this issue, please confirm https://github.com/MarlinFirmware/Marlin/pull/19268 fixes the reported problems.

CRCinAU, 2020-09-07:
Had one report that this fixes the issue - however can wait for more feedback, or close and reopen if a related issue re-emerges...

closed 2020-09-07 by thisiskeithb

pixelicous, 2020-10-07:
Not trying to print files with over 22characters but I might have that on my card, but I I also haven't tried to replace the SD card I'm using (original one)

I'm having the same issues on ender 3 v2 marlin 2.0.7,extruder flies into frame, weird speeds on fan and axis, z axis working together while extruder works when I change movement in menu??

Have tried to install bug fix
Using official demo example config, haven't changed a thing.. Will try in a few hours the proposed solution few posts up