MarlinFirmware/Marlin #18791: [BUG] Confusing and Ambigious description of NOZZLE_TO_PROBE_OFFSET

thierryzoller, 2020-07-26:
The (0,0) Coordinates are suggesting the Probe Somehow needs to be set as relative to Bed Origin. In reality, it needs to be set relative to the Nozzle position. Whatever the reasons for (0,0)

Suggest doing the following : 

1. Removing "(0,0)"  - or explain the purpose it is indicated
2. Add Text hat the character "O" is supposed to mean (Bed Origin) 

`* Z Probe to nozzle (X,Y) offset, relative to (0, 0).
*
* In the following example the X and Y offsets are both positive:
*
*   #define NOZZLE_TO_PROBE_OFFSET { 10, 10, 0 }
*
*     +-- BACK ---+
*     |           |
*   L |    (+) P  | R <-- probe (20,20)
*   E |           | I
*   F | (-) N (+) | G <-- nozzle (10,10)
*   T |           | H
*     |    (-)    | T
*     |           |
*     O-- FRONT --+
*   (0,0)`

GMagician, 2020-07-26:
@thierryzoller probe offset is relative to nozzle. If you read what is written in example you will se that taken 0,0 as bed origin and N as nozzle positionated at (10,10), probe in that situation is at 20,20. You have to write 10,10 in parameter...then relative to nozzle

sjasonsmith, 2020-07-26:
@thierryzoller since you have thought a out how you think this should change, why don’t you post a pull request with the edits?

I can’t promise you the change will be accepted, but that would be more productive than debating it here.

GMagician, 2020-07-26:
To be more honest 0,0 is origin. When nozzle is over such point, its coordinates are 0,0. Usually it is also the edge of the "printable area". In config example you may see the 3 letters O, N, P, they are origin, nozzle and probe.

And as told you in #18783 best approch to "cases" liki this are:

    MarlinFW.org Marlin Documentation
    RepRap.org Marlin Forum
    Tom's 3D Forums
    Facebook Group "Marlin Firmware"
    Facebook Group "Marlin Firmware for 3D Printers"
    Marlin Configuration on YouTube
    Marlin Discord server. Join link: https://discord.gg/n5NJ59y

> After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a bug report.

thierryzoller, 2020-07-27:
> @thierryzoller since you have thought about how you think this should change, why don’t you post a pull request with the dits?
1. I don't know all the use cases and scenarios that would have to be respected to make it clearer.
2. Never done a pull request my entire life

thierryzoller, 2020-07-27:
> @thierryzoller probe offset is relative to nozzle. If you read what is written in example you will se that taken 0,0 as bed origin and N as nozzle positionated at (10,10), probe in that situation is at 20,20. You have to write 10,10 in parameter...then relative to nozzle

Are you actually reading the issue and the config file? Let me help you by repeating it here in a single line it says " Z Probe to nozzle (X,Y) offset, relative to (0, 0)." <-- Do you see the Problem?  It's not relative to 0,0 (which on the graph is indicated at bed origin) so it is relative to NOZZLE.

thierryzoller, 2020-07-27:
I'll try to post an improvement here we can discuss, give me a bit of time.

sjasonsmith, 2020-07-27:
> Never done a pull request my entire life

@thierryzoller none of us had ever done a pull request prior our first one.

As I know you have observed, it isn't always easy to find people with time available to focus on your issues. There is a lot of power in fixing them yourself then sharing to help others in the future.

There is no single proper way to do it, but one method is documented in this guide:
https://marlinfw.org/docs/development/getting_started_pull_requests.html

GMagician, 2020-07-28:
@thierryzoller 

> Are you actually reading the issue and the config file? Let me help you by repeating it here in a single line it says " Z Probe to nozzle (X,Y) ### offset, relative to (0, 0)." <-- Do you see the Problem?

oh yeah now I got it

GMagician, 2020-07-28:
@thierryzoller plese check PR if now is clear

JoeTundra, 2020-07-30:
Good day. I'm a newb boomer trying to figure out the basics of 3d printing. I have an ender 3 V2, and got a not supported bed level sensor and thought I'd try and get it to work. That got me here.

I am another one who found the Nozzle to Probe offset instructions confusing. I took a stab at clarifying it. 

I think most confusion might be prevented simply by not having the nozzle to probe relative position, exactly the same as the 0,0 bed to nozzle relative position, in the example..

I think easiest solution is to simply start the instructions with the nozzle in the 0,0 position on the bed. That way, the only possible variable, is the probe. 

Have a good day.

GMagician, 2020-07-30:
@JoeTundra have you read new config description? check in bugfix and give feedback

JoeTundra, 2020-07-30:
@GMagician I have not yet. I will check it out now. Thanks.

boelle, 2020-07-31:
@thierryzoller PR #18828 is merged and have made it more clear, will close this as fixed, we can always reopen if its not clear enough.

closed 2020-07-31 by boelle

thierryzoller, 2020-07-31:
Thank you @GMagician, together with the explanations I think I would have understood it directly without playing around first.

PR #18828 (2020-07-29): Clarify Nozzle-to-Probe offset
This should fix #18791

If approved will follow all other cnf update