MarlinFirmware/Marlin #18795: [BUG] Marlin docs search doesn't find "wait for user"

onealexharms, 2020-07-26:
This is small, but it seems like it matters. Please delete and/or let me know if it shouldn't be here, or if it a feature request, or whatever.

### Bug Description

Searching the docs for "wait for user" found nothing, though "and wait for user." is in the one line description of Unconditional Stop

### My Configurations

Not applicable. 

### Steps to Reproduce

1. Go to the [docs](https://marlinfw.org/docs/gcode/M000-M001.html)
2. Search for "wait for user"

**Expected behavior:** [What you expect to happen]
The page for Unconditional Stop is found

**Actual behavior:** [What actually happens]
Nothing is found.

#### Additional Information
 
When you search instead of "wait", a bunch of results appear, and down the list you find Unconditional Stop, with the description "Stop and wait for user."

thinkyhead, 2020-08-27:
Thanks for the note! Small insignificant words like "for" are stripped from the search index, but are not ignored by the search results filter. It shouldn't be too hard to fix.

thinkyhead, 2020-08-28:
Ok, it should work a bit better now! Terms that have been stripped from the text are now ignored, and the search pattern will now match all search terms in order in the text even if they are far apart.

closed 2020-08-28 by thinkyhead

onealexharms, 2020-08-28:
Hey thanks! That’s gonna help folks I bet :)