MarlinFirmware/Marlin #18796: [BUG] filament motion sensor distance to out not counting correctly

brencerddwr, 2020-07-27:
### Bug Description

SKR mini e3 v1.2 with BTT smart filament sensor installed.  Used the used the runout debug switch and watched the debugging after I cut the filament.  After the movement detected stopped, the remaining to out number incremented up to >50 when I stopped it.  Printer was air printing.
### My Configurations
Ender 3 with SKR mini E3 v1.2
BTT smart filament sensor connected to e0-stop (P15) with power from ISP connecter
simple runout made with makerbot endstop wired the same way and configuration file appropriate for it, worked fine. 

[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4979807/configuration.zip)

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. Install firmware with runout debugging enabled
2. Use pronterface to start print from sd card
3. Cut filament
4. Watch debug output and distance remaining to out climb until I abort print

**Expected behavior:** [What you expect to happen]
When the sensor stops reporting movement, auto pause to replace filament

**Actual behavior:** [What actually happens]
printer keeps printing until stopped or the file is finished

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

brencerddwr, 2020-07-27:
This line is from runout.h:
runout_mm_countdown[e] -= (TEST(b->direction_bits, E_AXIS) ? -steps : steps) * planner.steps_to_mm[E_AXIS_N(e)];

I think it is where it is doing the calculations but it is beyond my understanding of code.  Will some kind soul translate this into something a mere mortal can understand?

brencerddwr, 2020-07-28:
I disabled linear advance in the firmware and left scurves enabled and that seemed to resolve the issue.  While setting up the debug branch to test it, I found out linear advance and scurves do not play well together.

closed 2020-07-28 by brencerddwr

Taomyn, 2020-08-12:
Just adding for anyone else that also comes looking regarding this issue, that this has been my experience as well - but I use the PT-DET socket. Disabled "linear advance" which I had never gotten around to setting up anyway, and now the smart sensor is working fully.