MarlinFirmware/Marlin #18797: [BUG] (2.0.6 crashes nozzle into bed)

cr20-123, 2020-07-27:
### Bug Description

After updating to 2.0.6 from 2.0.5.2 the nozzle crashes into the bed after doing the G29 and wait for hotend heating to finish in my start gcode. The start gcode worked on 2.0.5.2 without any issues.

### My Configurations
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/4979885/configs.zip)

### Steps to Reproduce

1. Use this start gcode for cura (change Z offset to your own)
[start gcode.txt](https://github.com/MarlinFirmware/Marlin/files/4979887/start.gcode.txt)

2. run the print

3. When the nozzle finishes heating after the G29 watch the nozzle move a little to the left then crash into the bed

**Expected behavior:** [What you expect to happen]
The nozzle to move over to the far left and do a purge line

**Actual behavior:** [What actually happens]
The nozzle moves a little to the left and goes straight into the bed

ilium007, 2020-07-27:
I can't see where in your start gcode you actually load the UBL mesh using an M420 to then perform the G29 on?? If I ever tried a G29 without loading the mesh slot the print head would crash into the bed.

This is the line I used in my start gcode for UBL mesh levelling:
`M420 S1 Z10 ; enable mesh levelling and load slot 1, 10mm fade height`

cr20-123, 2020-07-27:
I am using bilinear not UBL

ilium007, 2020-07-28:
You still have to enable bed leveling with the Bilinear method - see https://marlinfw.org/docs/gcode/M420.html

If you do not issue an M420 before the G29 the print head will crash into bed.

Again - this is my start gcode (I use bilinear by the way):

```
; Restore Bed Levelling
M117 Bed Levelling...
M420 S1 Z10 ; enable and load mesh, set fade height 10mm
G0 X0 Y15 F3000 ; return to origin
```

cr20-123, 2020-07-28:
I appreciate you trying to help, but you do not need to enable leveling before a G29. When you do a G29 it enables leveling. As I said in my original post it is something in Marlin 2.0.6 that is causing it. I have used many versions of marlin before it with the same start G-Code and I have never had any problems.

fleek, 2020-07-28:
You should have a G28 followed by the M420 command.  As the printer need to initialise the position of the hotend, otherwise you will start from a unknown position and potentially crash the hotend into the bed.

cr20-123, 2020-07-28:
I just added the start G-Code provided by TH3D which mine is based on and there is no M420 S1 in it.

cr20-123, 2020-07-28:
Update ABL_BILINEAR_SUBDIVISION is causing the problem. It is not properly calculating some of the points which is causing the bed leveling to act weird.
[Leveling data.txt](https://github.com/MarlinFirmware/Marlin/files/4990195/Leveling.data.txt)

cr20-123, 2020-08-06:
This is still an issue

closed 2020-08-06 by cr20-123