MarlinFirmware/Marlin #18798: [BUG] MANUAL_Y_HOME_POS Does Nothing

Dracrius, 2020-07-27:
### Bug Description

After installing a new hotend mount (Hero Me Gen5) I needed to correct the Y position of the nozzle as it now sticks out about 9mm. The pdf documentation for the mount says to enable/ set `#defineMANUAL_Y_HOME_POS-9` to achieve this but after trying about a half dozen times or more with different values I see absolutely no change in either the nozzle position ( `G1 X117.5 Y117.5` After homing) or the prob position of a G29 which is way off my `NOZZLE_TO_PROBE_OFFSET` because it thinks the nozzle is 9mm further back then it actually is.

The other suggested method of adding `M206 Y-9; Y axis offset` to start gcode on the other had appears to work. I tried this by sending my machine a `M206 Y-9` followed by a  `G1 X117.5 Y117.5` and the nozzle was centered.

### My Configurations

[Marlin Configs bugfix-2.x 270720.zip](https://github.com/MarlinFirmware/Marlin/files/4980504/Marlin.Configs.bugfix-2.x.270720.zip)

### Steps to Reproduce

1. Download current bugfix-2.0.x
2. Set a `#define MANUAL_Y_HOME_POS`
3. Compile and Flash
4. Auto Home and Run `G1 X117.5 Y117.5`
5. Disable Stepper motor and lower head to rest nozzle on 0 line of ruler. Then measure distance to edge.
6. Change `#define MANUAL_Y_HOME_POS` repeat step 3-5
7. Rinse and repeat but see no change in distance from edge!

**Expected behavior:** 

Offset Center/ Home position by 9mm towards the front of the bed!

**Actual behavior:** 

Nozzle considers the same position center no matter the `#define MANUAL_Y_HOME_POS`

ChipCE, 2020-07-27:
Did you try to reset the EEPROM after flashing?

Dracrius, 2020-07-28:
> 
> 
> Did you try to reset the EEPROM after flashing?

Just tried that for the heck of it but it made no difference. Not surprising as the EEPROM reports the correct default offset but it doesn't seem to use it.

Dracrius, 2020-07-28:
Appears after a EEPROM reset Home offsets are all reset to 0 not the firmware defined setting of -9 for y and 0 for the rest. After the reset I can use `M206 Y-9` to define the correct offset and save it to EEPROM using `M500`. My offset will now at least persist through reboots but there is an issue with marlin loading the firmware defined offset as a reset always take me back to 0 despite `#define MANUAL_Y_HOME_POS -9` being defined.

ManuelMcLure, 2020-07-28:
The M206 offsets and the setting of MANUAL_[XY]_HOME_POS are not and have never been the same thing. Home offsets are usually only used with CNC, not 3d printing.
Instead of using MANUAL_[XY]_HOME_POS try using MIN_[XY]_POS instead.

ChipCE, 2020-07-28:
@ManuelMcLure  Interesting, so they are doing the same thing ?. I'm using MANUAL_Y_HOME_POS and it work for me on 2.0.5.4 and 2.0.6.

ManuelMcLure, 2020-07-28:
@ChipCE in almost all cases they’re interchangeable. I believe if you have dual X carriages there’s some difference.

Dracrius, 2020-07-28:
What the heck could I test to figure out why `MANUAL_Y_HOME_POS` wouldn't work for me? Is it possibly mainboard related? I'm using a SKR Mini e3 v1.2. I did not build for release 2.0.6 but the current bugfix as of last night. Also @ManuelMcLure where is `MIN_Y_POS` defined it is not in either the Configuration.h or Configuration_adv.h. Also the mount/ fanduct in question (Hero Me Gen5) is an extremely popular one so if they shouldn't be using `MANUAL_Y_HOME_POS` I'd like to tell them before to many people do, but to me reading the marlin comments for `MANUAL_Y_HOME_POS` would lead me to believe that is exactly what it is meant for, defining offsets to the home position which in the case of a 3d printer is the nozzle.

Dracrius, 2020-07-28:
Also to quote marlin
```
// Manually set the home position. Leave these undefined for automatic settings.
// For DELTA this is the top-center of the Cartesian print volume.
```

Which would contradict your statement that it is primarily intended for cnc's although i guess there could be a delta CNC but I've never heard of one.

ChipCE, 2020-07-28:
@Dracrius try the stable release 2.0.6, that version works on my board (SKR). In my case i need to set the offset in config file, flash the firmware, reset EEPROM then reset the machine(or unplug it) to get it to work.

Dracrius, 2020-07-28:
I should be getting to bed so I'll try again in the morning but I'm very curious where Home Offsets are defined in  Configuration.h or Configuration_adv.h if it is not the HOME_POS settings. Are you 100% certain they are unrelated @ManuelMcLure I see no other setting that would define that and so far as I have noticed there is a default value for every EEPROM setting in one of the two configuration files! This is the kind of setting I wouldn't want reset just cause I reset the EEPROM as the chances that I remember I need to reset that are slim if a month down the line I reset my settings.

ManuelMcLure, 2020-07-28:
> Which would contradict your statement that it is primarily intended for cnc's although i guess there could be a delta CNC but I've never heard of one.

I never said that ``MANUAL_X_HOME_POS`` and ``MANUAL_Y_HOME_POS`` are for CNC - I said ``M206`` (workspace offsets) is for CNC. I called it "home offset" but that was a misnomer - I should have said "workspace offset".
You can use ``MANUAL_X_HOME_POS`` and ``MANUAL_Y_HOME_POS`` (or ``X_MIN_POS``/``Y_MIN_POS``) even with workspace offsets (and therefore ``M206``) disabled (``NO_WORKSPACE_OFFSETS``).
 
@ChipCE - with my ``X_MIN_POS`` set to -19 and ``Y_MIN_POS`` set to -21 after doing ``G28 X Y`` the LCD shows X=-19 Y=-21 which is correct. ``G0 X0 Y0`` then takes the nozzle to the front left corner of the bed.

ChipCE, 2020-07-28:
@ManuelMcLure Thank you! That is the diffirent, in config with MANUAL_X_HOME_POS, <code>G28 X Y</code> will move the nozzle to 0,0 by default.

@Dracrius Not sure if this can help, but this is my working config
https://github.com/ChipCE/Weird-bear-Marlin-BTT-SKR1_3

Dracrius, 2020-07-28:
@ManuelMcLure If you type `M503` and read the titles it is clearly labeled the home offset. I think the use case has changed fsince when ever you came to these conclusions. I would also think / prefer my G28 to home at what is the machines 0,0 and I think the machine should refer to it as such so I'll skip using `Y_MIN_POS` personally just doesn't seem right. Sure it gets the job done but so does the home offset which as far as I can tell designed for these use cases where the nozzle/ home reference is what has moved not the bed.

ManuelMcLure, 2020-07-28:
@Dracrius Whatever works, but don't be surprised if other people get confused by your setup or you run into some obscure bug with leveling or probing or such. The use of negative ``X_MIN_POS``/``Y_MIN_POS`` (or ``MANUAL_X_HOME_POS``/``MANUAL_Y_HOME_POS``) is much more common and tested and less likely to break.

Dracrius, 2020-07-28:
@ManuelMcLure well `MANUAL_Y_HOME_POS` is not working as expected and unfortunately this is soon to be far from an uncommon use! I honestly don't understand why you think that setting shouldn't be used!! Absolutely no part of the documentation states that that is for CNC and it is not or at least is no longer a "workspace offset" as even the [Marlin Wiki for M206](https://marlinfw.org/docs/gcode/M206.html) is called **M206 - Set Home Offsets**.

This is the instructions that I am certain at least 723 people (# of Favorites on https://www.thingiverse.com/thing:4460970)
```

This Home setting can be madein firmware, but it also can be done via GCode.The firmware setting can be edited/added to the configuration.h file. Based upon the Gantry Adapter you are using, you would add one of these two lines in the configuration.h file, then recompile and upload to your printer.#defineMANUAL_Y_HOME_POS-9
Hero Me Gen5 Master SuiteJuly 13, 2020 Version 1.4Page 23of 29You can set the Home setting via your slicer software. Based upon the Gantry Adapter you are using, you would add one of these two lines to the Start GCode in your slicer’s settings:M206 Y-9; Y axis offsetIn case these values do not place the hotend’s nozzle just inside the edge of the build plate when homing the Y axis, you can adjust the Y offset value up or down to work for your setup. The Marlin firmware detailed instructions on setting the printer’s Homeoffset are here:https://marlinfw.org/docs/gcode/M206.html
```
From[ this pdf](https://cdn.thingiverse.com/assets/b6/33/c2/db/e9/Hero_Me_Gen5_Parts_Cross_Reference_and_Assembly_Instructions.pdf) posted as the install instructions!

Dracrius, 2020-07-28:
> 
> 
> In my case i need to set the offset in config file, flash the firmware, reset EEPROM then reset the machine(or unplug it) to get it to work.

This worked even on my Bugfix build. With the power cycle after the reset it finally began acting as normal. I am surprised it wasn't doubling up `MANUAL_Y_HOME_POS` with Home Offset as I had tried resetting last night and testing right away and it didn't work so I added the `M206 Y-9` and saved and then ran a test that came out good. I then power cycled and tested again I would of though both should have been working at that point for a -18 offset but I didn't see that.

I am now unsure if I should count this solved or count it as a bug. While simply needing to reset the EEPROM could make sense after an update, specifically needing to reset and power cycle seems a bit more like a bug especially since as far as I remember its not like even resetting your EEPROM is part of most update guides.

But thanks for the suggestion @ChipCE

Dracrius, 2020-07-28:
Nevermind just tweaked the `MANUAL_Y_HOME_POS` many times as it wasn't perfectly center and after flashing I ran a test each time without a reset and things worked fine. I now have to assume something was just off with my EEPROM coming from last weeks bugfix that didn't mesh well with my more or less 2.0.6 build but post 2.0.6 updating does not require EEPROM resetting.

closed 2020-07-28 by Dracrius