MarlinFirmware/Marlin #18799: [BUG] DOUBLECLICK_FOR_Z_BABYSTEPPING and MOVE_Z_WHEN_IDLE does not work together

ChipCE, 2020-07-27:
### Bug Description

With DOUBLECLICK_FOR_Z_BABYSTEPPING and MOVE_Z_WHEN_IDLE enabled, when printing MOVE_Z showing instead of Z_BABYSTEPPING.

### My Configurations

I noticed the behavior since 2.0.5.4 and it appear in 2.0.6 too.

<pre>
#define BABYSTEPPING
#if ENABLED(BABYSTEPPING)
  //#define INTEGRATED_BABYSTEPPING
  //#define BABYSTEP_WITHOUT_HOMING
  //#define BABYSTEP_XY
  #define BABYSTEP_INVERT_Z false
  //#define BABYSTEP_MILLIMETER_UNITS
  #define BABYSTEP_MULTIPLICATOR_Z  1
  #define BABYSTEP_MULTIPLICATOR_XY 1

  #define DOUBLECLICK_FOR_Z_BABYSTEPPING
  #if ENABLED(DOUBLECLICK_FOR_Z_BABYSTEPPING)
    #define DOUBLECLICK_MAX_INTERVAL 1250
    //#define BABYSTEP_ALWAYS_AVAILABLE
    #define MOVE_Z_WHEN_IDLE
    #if ENABLED(MOVE_Z_WHEN_IDLE)
      #define MOVE_Z_IDLE_MULTIPLICATOR 1
    #endif
  #endif

  //#define BABYSTEP_DISPLAY_TOTAL          // Display total babysteps since last G28

  #define BABYSTEP_ZPROBE_OFFSET          // Combine M851 Z and Babystepping
  #if ENABLED(BABYSTEP_ZPROBE_OFFSET)
    //#define BABYSTEP_HOTEND_Z_OFFSET      // For multiple hotends, babystep relative Z offsets
    //#define BABYSTEP_ZPROBE_GFX_OVERLAY   // Enable graphical overlay on Z-offset editor
  #endif
#endif
</pre>

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[Marlin.tar.gz](https://github.com/MarlinFirmware/Marlin/files/4980821/Marlin.tar.gz)

### Steps to Reproduce

1. Start printing
2. Double click

**Expected behavior:** [What you expect to happen]
* MOVE_Z show when not printing if double clicked
* Z_BABYSTEPPING show when printing if double clicked

**Actual behavior:** [What actually happens]
In both case(printing or not) MOVE_Z is showed up.

boelle, 2020-07-27:
you configs are in tar.gz format, makes it more dificult for windows people

plese use zip as the  template tells you to

ChipCE, 2020-07-27:
> you configs are in tar.gz format, makes it more dificult for windows people
> 
> plese use zip as the template tells you to

Oups. Updated the files.

thinkyhead, 2020-08-27:
The build succeeds now, so the issue is resolved.

closed 2020-08-27 by thinkyhead