MarlinFirmware/Marlin #18800: [FR] Ender 3V2 : Adding relative motion to controller menu

fleek, 2020-07-27:
### Description

Currently, the controller offers the options : Prepare -> Move -> Move X,Y,Z options.  However, these moves are absolute.  In my previous Anet A8 LCD controller, the move options are relative to the current hotend position.

I don't know in what circumstances will the current absolute move options be useful, but I think relative move options might be more useful to fine-tune the positions of the hotend.

### Feature Workflow

1. Adding relative move options to the controller menu.  Keeping the existing absolute move options for X,Y and Z.  However, move the Extruded option to the new relative option menu.
2.  Change the current move option to be more intuitive to specify 'Move Absolute' instead of 'Move'
3.  Include menu option under Prepare -> Move relative -> Move relative X,Y,Z, E
The relative moves will be governed by the soft endstops if turned on.  The registered current coordinates of X,Y,Z position will be display and the relative move distance can be selected.  I have not decided on the menu design yet.  

#### Additional Information

I can help work on this feature.  However, I would need tools to edit the icon files from the DWIN DGUS if someone can point me where to obtain them.  It will take me sometime to implement this feature as I cannot commit full time to this feature, any help will be appreciated.

sjasonsmith, 2020-07-28:
I removed the "Feature Request" label because I thought surely you were mistaken. The moves on the LCD are always relative.

Now I see that you are saying this is the Ender 3 V2 display. Jogging moves do not sound useful if they are absolute, but it still seems like surely this is a bug, and not a feature request.

I think we need more information to understand the problem, and for more V2 users to verify the behavior.

Pictures of the menu which causes problems would be helpful.

sjasonsmith, 2020-07-28:
Admittedly, I know so little about this display that I don't know whether Marlin controls what is on that display, or if it is being controlled by separate display firmware.

fleek, 2020-07-28:
@sjasonsmith After autohoming, when you navigate to the menu option Prepare->Motion, you will see the following numbers being shown on the X,Y and Z.  I think this proves that it is absolute rather than relative, if it is relative, all the values should be zero right?
 
[image]

sjasonsmith, 2020-07-28:
@fleek I've never used that interface. I assume that if you tap "MoveX" you are then provided a means to adjust that in small _relative_ units. But again, I've not used it, so maybe not!

boelle, 2020-07-28:
@fleek please edit your title and replace [FR] with [BUG]

fleek, 2020-07-30:
@sjasonsmith In the dwin.cpp file, under the HMI_Move_X subroutine :
`
  current_position[X_AXIS] = HMI_ValueStruct.Move_X_scale / 10;
`
If I am not wrong, it is a absolute position move.

sjasonsmith, 2020-07-30:
@fleek I am not doing anything with this issue. I do not have the hardware to attempt to help.

cosmoderp, 2020-08-02:
I actually plan to add this to DWIN.cpp as part of my menus overall that I am working on but as an option you have to enable.  I have made some menu changes already but I have to work too so its a bit slow.

closed 2020-09-07 by github-actions[bot]