MarlinFirmware/Marlin #18801: [FR] Maximum_Stepper_Rate per axis

3iswolf, 2020-07-27:
I installed a piezo-driven dispensing system on the E-Axis. I use the Step-Signal to trigger the system. The hole dispensing calculation is done external and the maximum frequency of step signals is limited. If two step signal are too close, the second step signal will be ignored. It's like an loss of steps with a stepper. 

The 3D-Printer is an CoreXY  with TMC2130's and i want to go fast on the X/Y-Plane. The dispensing system is limited to frequency of 2000 Hz. 

I can calculate the maximum travel speed of the axis so the steps won't be faster than 2000 Hz. But Jerk and Acceleration could come faster. I probed the step signal with an Oscilloscope. 2000 Hz equals 0,5 ms. I get always a double Step Pulse in 0,1 ms.

[image]

The hard limit in Marlin #define MAXIMUM_STEPPER_RATE is the perfect solution and works perfect for the e-axis but this is a global setting and MAXIMUM_STEPPER_RATE 2000 would limit my X/Y axis to only 12,5 mm/s.

Is there a way to set #define MAXIMUM_STEPPER_RATE per axis? 

_______________________________________________________________________________________________________________________

In the are of  #define DRIVER_TYPE in configuration.h. I should be noted that MAXIMUM_STEPPER_RATE, MINIMUM_STEPPER_PULSE and MINIMUM_STEPPER_POST_DIR_DELAY are changed to value of the slowest entered stepper driver type. 

Example: 

//#define X_DRIVER_TYPE  TB6560
//#define Y_DRIVER_TYPE  TMC2130
//#define Z_DRIVER_TYPE  A4988

All axis behave like TB6560. I tried this a different solution for my problem. 

Thanks!

sjasonsmith, 2020-07-27:
Do you have linear advance enabled? If so, disable it.

sjasonsmith, 2020-07-27:
I have seen double-pulses like this cause by linear advance. It is something I intend to debug but have not gotten around to yet.

sjasonsmith, 2020-07-27:
This is a capture I took the other day. This is with square wave stepping enabled, so each edge is a step. Eached circled area represents a double-step, which might be the same thing you are seeing.
[image]

3iswolf, 2020-07-27:
Thanks for the comments. I tried some different configurations. 

You're right. Linear advanced did this double pulse. 

In my current setup. I have a normal Extruder on E0 and the dispensing system E1. 

I set M900 T0 K1.2 and M900 T1 K0 after flashing to disable Lin_adv for the dispenser. I don't see a way to set different K Values in Marlin directly. 

But this was not enough. After your posts i disabled Linear advanced completely and it stops the double pulses.  Thanks! 

So, how can i be sure, that i will not get other 2000 Hz? Acceleration and Jerk makes me some headache.  

When Steps/mm are 50.
DEFAULT_MAX_FEEDRATE 40
DEFAULT_MAX_ACCELERATION ?
DEFAULT_EJERK ? or JUNCTION_DEVIATION_MM ?

sjasonsmith, 2020-07-27:
You still saw the double pulse with K=0? Did behavior change at all? 

I’m not sure about those other settings. I wouldn’t thing you would see anything higher than the max travel rate, but I don’t know exactly how those features work.

sjasonsmith, 2020-07-28:
I am re-labeling this as a Linear Advance bug, rather than a Feature Request. I can't modify the titles, but resolving the double pulse is a lot more feasible solution than setting separate step rates per axis.

In the image I posted, the Linear Advance interrupt rate was faster than the movement rate. E moves were deferred to the LA interrupt, which would cause a double-step every time an actual step (not a LA step) was called for.

I'm not sure what the LA behavior is when K=0. We shouldn't be handling any LA steps, so the double-step is coming from somewhere else (if it actually remains with K=0). One theory is that move steps are still deferred to the LA ISR, but the move ISR is executing faster than the LA ISR, causing multiple E steps to be queued up and applied in a batch.

AnHardt, 2020-07-28:
> 
```
/**
 * Default Axis Steps Per Unit (steps/mm)
 * Override with M92
 *                                      X, Y, Z, E0 [, E1[, E2...]]
 */
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 4000, 500 }

/**
 * Default Max Feed Rate (mm/s)
 * Override with M203
 *                                      X, Y, Z, E0 [, E1[, E2...]]
 */
#define DEFAULT_MAX_FEEDRATE          { 300, 300, 5, 25 }
```
These together define a maximum steprate per axis. Together with `DISABLE_MULTI_STEPPING` you can get rid of multiple steps per stepper interrupt on X,Y,Z and for E if also `LIN_ADVANCE` is off. That will limit your max steprate to the max interrupt rate of the stepper interrupt - that varies with the used processor type and MHz.

Then Jerk speeds and accelerations are not involved in this. They are, for E, when LA is in use.

AnHardt, 2020-07-28:
> I get always a double Step Pulse in 0,1 ms.

That would mean above 10KHz. That's about the max frequency of the stepper interrupt for an AVR 16 MHz processor. Above that i expect multistepping.

sjasonsmith, 2020-07-28:
Good point about multi-stepping. I assumed it would not have been an issue for this axis, since the described speeds are so low.

@3iswolf could you attach your configuration files so we can are all of your settings?

closed 2020-09-03 by github-actions[bot]