MarlinFirmware/Marlin #18802: 'X_HOME_BUMP_MM' was not declared in this scope

jerryfudd, 2020-07-27:
Just trying to compile 2.0.6 with the example config for the BTT SKR Mini 1.2 example files - my only change being 'STM32F103RC_btt_512K' in platformio.ini and I'm getting the following when trying to compile.

Marlin/src/module/motion.cpp:77:71: error: 'X_HOME_BUMP_MM' was not declared in this scope
 #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval<T> NAME##_P = { X_##OPT, Y_##OPT, Z_##OPT }
Compiling .pio/build/STM32F103RC_btt_512K/src/src/sd/SdFatUtil.cpp.o
                                                                       ^
Marlin/src/module/motion.cpp:83:1: note: in expansion of macro 'XYZ_CONSTS'
 XYZ_CONSTS(float, home_bump_mm,   HOME_BUMP_MM);
 ^~~~~~~~~~
Marlin/src/module/motion.cpp:77:71: note: suggested alternative: 'HOMING_BUMP_MM'
 #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval<T> NAME##_P = { X_##OPT, Y_##OPT, Z_##OPT }
                                                                       ^
Marlin/src/module/motion.cpp:83:1: note: in expansion of macro 'XYZ_CONSTS'
 XYZ_CONSTS(float, home_bump_mm,   HOME_BUMP_MM);
 ^~~~~~~~~~
Marlin/src/module/motion.cpp:77:80: error: 'Y_HOME_BUMP_MM' was not declared in this scope
 #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval<T> NAME##_P = { X_##OPT, Y_##OPT, Z_##OPT }
                                                                                ^
Marlin/src/module/motion.cpp:83:1: note: in expansion of macro 'XYZ_CONSTS'
 XYZ_CONSTS(float, home_bump_mm,   HOME_BUMP_MM);
 ^~~~~~~~~~
Marlin/src/module/motion.cpp:77:80: note: suggested alternative: 'HOMING_BUMP_MM'
 #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval<T> NAME##_P = { X_##OPT, Y_##OPT, Z_##OPT }
                                                                                ^
Marlin/src/module/motion.cpp:83:1: note: in expansion of macro 'XYZ_CONSTS'
 XYZ_CONSTS(float, home_bump_mm,   HOME_BUMP_MM);
 ^~~~~~~~~~
Marlin/src/module/motion.cpp:77:89: error: 'Z_HOME_BUMP_MM' was not declared in this scope
 #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval<T> NAME##_P = { X_##OPT, Y_##OPT, Z_##OPT }
                                                                                         ^
Marlin/src/module/motion.cpp:83:1: note: in expansion of macro 'XYZ_CONSTS'
 XYZ_CONSTS(float, home_bump_mm,   HOME_BUMP_MM);
 ^~~~~~~~~~
Marlin/src/module/motion.cpp:77:89: note: suggested alternative: 'HOMING_BUMP_MM'
 #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval<T> NAME##_P = { X_##OPT, Y_##OPT, Z_##OPT }
                                                                                         ^
Marlin/src/module/motion.cpp:83:1: note: in expansion of macro 'XYZ_CONSTS'
 XYZ_CONSTS(float, home_bump_mm,   HOME_BUMP_MM);
 ^~~~~~~~~~
Compiling .pio/build/STM32F103RC_btt_512K/src/src/sd/SdFile.cpp.o
*** [.pio/build/STM32F103RC_btt_512K/src/src/module/motion.cpp.o] Error 1

ellensp, 2020-07-27:
You have to download from the link on the release page ie https://github.com/MarlinFirmware/Marlin/releases/tag/2.0.6
ie https://github.com/MarlinFirmware/Marlin/archive/2.0.6.zip   
Currently 2.0.x does not point to 2.0.6

boelle, 2020-07-27:
Use [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) release versions are just a frozen copy of that so does not make much sense to use them if you ask me

boelle, 2020-07-27:
and if its not much trouble could you add [BUG] at the start of the title ?

jerryfudd, 2020-07-27:
> Configurations, please
> Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
> requested in the Issue template) and drop them into your next reply.
> We'll check them over and see if anything is amiss.

No point in zipping, here they are -> [https://github.com/MarlinFirmware/Configurations/tree/release-2.0.6/config/examples/Creality/Ender-3/BigTreeTech%20SKR%20Mini%20E3%201.2](url)

After not being able to compile with my changes I tried compiling without modifying - I downloaded from the 2.0.6 link [https://github.com/MarlinFirmware/Marlin/archive/2.0.x.zip](url) updated platformio.ini for my board 'STM32F103RC_btt_512K' and tried compiling.

I didn't add [BUG] as I was yet to rule out end user stupidity or lack of knowledge so was looking for general advise first.

jerryfudd, 2020-07-27:
> You have to download from the link on the release page https://github.com/MarlinFirmware/Marlin/releases/tag/2.0.6
> ie https://github.com/MarlinFirmware/Marlin/archive/2.0.6.zip
> Currently 2.0.x does not point to 2.0.6

Good shout - I hadn't noticed that!

now when using your link and the config files mentioned above to @boelle I get this

`Marlin/src/feature/bedlevel/bedlevel.cpp:235:3: fatal error: opening dependency file .pio/build/STM32F103RC_btt_512K/src/src/feature/bedlevel/bedlevel.cpp.d: No such file or directory
   }
   ^
compilation terminated.
Compiling .pio/build/STM32F103RC_btt_512K/src/src/feature/powerloss.cpp.o
Compiling .pio/build/STM32F103RC_btt_512K/src/src/feature/probe_temp_comp.cpp.o
*** [.pio/build/STM32F103RC_btt_512K/src/src/feature/bedlevel/bedlevel.cpp.o] Error 1
In file included from Marlin/src/feature/../inc/MarlinConfig.h:47:0,
                 from Marlin/src/feature/encoder_i2c.cpp:31:
Marlin/src/feature/../inc/../core/serial.h:324:1: fatal error: opening dependency file .pio/build/STM32F103RC_btt_512K/src/src/feature/encoder_i2c.cpp.d: No such file or directory
 }
 ^
compilation terminated.
*** [.pio/build/STM32F103RC_btt_512K/src/src/feature/encoder_i2c.cpp.o] Error 1
In file included from Marlin/src/feature/../inc/MarlinConfig.h:47:0,
                 from Marlin/src/feature/fanmux.cpp:28:
Marlin/src/feature/../inc/../core/serial.h:324:1: fatal error: opening dependency file .pio/build/STM32F103RC_btt_512K/src/src/feature/fanmux.cpp.d: No such file or directory
 }
 ^
compilation terminated.
*** [.pio/build/STM32F103RC_btt_512K/src/src/feature/fanmux.cpp.o] Error 1
In file included from Marlin/src/feature/../inc/../core/boards.h:24:0,
                 from Marlin/src/feature/../inc/MarlinConfigPre.h:35,
                 from Marlin/src/feature/joystick.cpp:27:
Marlin/src/feature/../inc/../core/macros.h:310:5: fatal error: opening dependency file .pio/build/STM32F103RC_btt_512K/src/src/feature/joystick.cpp.d: No such file or directory
     }
     ^
compilation terminated.
*** [.pio/build/STM32F103RC_btt_512K/src/src/feature/joystick.cpp.o] Error 1
In file included from Marlin/src/feature/../inc/MarlinConfig.h:47:0,
                 from Marlin/src/feature/filwidth.cpp:23:
Marlin/src/feature/../inc/../core/serial.h:324:1: fatal error: opening dependency file .pio/build/STM32F103RC_btt_512K/src/src/feature/filwidth.cpp.d: No such file or directory
 }
 ^
compilation terminated.
*** [.pio/build/STM32F103RC_btt_512K/src/src/feature/filwidth.cpp.o] Error 1
In file included from Marlin/src/feature/../inc/../core/boards.h:24:0,
                 from Marlin/src/feature/../inc/MarlinConfigPre.h:35,
                 from Marlin/src/feature/max7219.cpp:38:
Marlin/src/feature/../inc/../core/macros.h:310:5: fatal error: opening dependency file .pio/build/STM32F103RC_btt_512K/src/src/feature/max7219.cpp.d: No such file or directory
     }
     ^
compilation terminated.
*** [.pio/build/STM32F103RC_btt_512K/src/src/feature/max7219.cpp.o] Error 1
In file included from Marlin/src/feature/../inc/../core/boards.h:24:0,
                 from Marlin/src/feature/../inc/MarlinConfigPre.h:35,
                 from Marlin/src/feature/power_monitor.cpp:23:
Marlin/src/feature/../inc/../core/macros.h:310:5: fatal error: opening dependency file .pio/build/STM32F103RC_btt_512K/src/src/feature/power_monitor.cpp.d: No such file or directory
     }
     ^
compilation terminated.
In file included from Marlin/src/feature/../inc/MarlinConfig.h:47:0,
                 from Marlin/src/feature/fwretract.cpp:27:
Marlin/src/feature/../inc/../core/serial.h:324:1: fatal error: opening dependency file .pio/build/STM32F103RC_btt_512K/src/src/feature/fwretract.cpp.d: No such file or directory
 }
 ^
compilation terminated.
*** [.pio/build/STM32F103RC_btt_512K/src/src/feature/power_monitor.cpp.o] Error 1
*** [.pio/build/STM32F103RC_btt_512K/src/src/feature/fwretract.cpp.o] Error 1
In file included from Marlin/src/feature/../inc/MarlinConfig.h:47:0,
                 from Marlin/src/feature/host_actions.cpp:23:
Marlin/src/feature/../inc/../core/serial.h:324:1: fatal error: opening dependency file .pio/build/STM32F103RC_btt_512K/src/src/feature/host_actions.cpp.d: No such file or directory
[... 51 lines cut ...]
compilation terminated.
*** [.pio/build/STM32F103RC_btt_512K/src/src/feature/pause.cpp.o] Error 1
FileNotFoundError: [Errno 2] No such file or directory: '/Users/daniel/Documents/Visual Studio Code/Marlin-2.0.x/.pio/build/STM32F103RC_btt_512K/.sconsign37.db.dblite':
  File "/Users/daniel/.platformio/packages/tool-scons/script/../engine/SCons/Script/Main.py", line 1381:
    _exec_main(parser, values)
  File "/Users/daniel/.platformio/packages/tool-scons/script/../engine/SCons/Script/Main.py", line 1344:
    _main(parser)
  File "/Users/daniel/.platformio/packages/tool-scons/script/../engine/SCons/Script/Main.py", line 1119:
    nodes = _build_targets(fs, options, targets, target_top)
  File "/Users/daniel/.platformio/packages/tool-scons/script/../engine/SCons/Script/Main.py", line 1318:
    jobs.run(postfunc = jobs_postfunc)
  File "/Users/daniel/.platformio/packages/tool-scons/script/../engine/SCons/Job.py", line 113:
    postfunc()
  File "/Users/daniel/.platformio/packages/tool-scons/script/../engine/SCons/Script/Main.py", line 1315:
    SCons.SConsign.write()
  File "/Users/daniel/.platformio/packages/tool-scons/script/../engine/SCons/SConsign.py", line 117:
    syncmethod()
[... cut ...]
Traceback (most recent call last):
  File "/Users/daniel/.platformio/packages/tool-scons/script/../engine/SCons/dblite.py", line 143, in __del__
  File "/Users/daniel/.platformio/packages/tool-scons/script/../engine/SCons/dblite.py", line 140, in close
  File "/Users/daniel/.platformio/packages/tool-scons/script/../engine/SCons/dblite.py", line 161, in sync
FileNotFoundError: [Errno 2] No such file or directory: '/Users/daniel/Documents/Visual Studio Code/Marlin-2.0.x/.pio/build/STM32F103RC_btt_512K/.sconsign37.db.dblite'`

jerryfudd, 2020-07-27:
[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/4981727/Archive.zip)

ellensp, 2020-07-27:
try again. real marlin 206 with your config files works fine for me.
You might have to remove .pio folder in Marlin.

jerryfudd, 2020-07-27:
> try again. real marlin 206 with your config files works fine for me.
> You might have to remove .pio folder in Marlin.

Thats so weird, its compiling now - could it not have been pulling a dependancy earlier?

> and if its not much trouble could you add [BUG] at the start of the title ?

like said above - no BUG :D

thanks @ellensp

ellensp, 2020-07-27:
Can we close this then?

jerryfudd, 2020-07-27:
> Can we close this then?

please do, I've just made all of my normal changes and still compiling now without issue.

thanks for the help!

Dan

closed 2020-07-27 by ellensp