MarlinFirmware/Marlin #18803: Manual mesh bed levelling

shayan123869, 2020-07-27:
Manual mesh bed levelling not working from LCD when dual x carriage is defined. When clicked on bed levelling It just does auto home and goes to home screen. I have my own idex 3d printer running on ramps 1.4 with stepper expansion board....previous version of marlin 1.1.9 works fine but the issue is in marlin 2....it would be of great help if this is fixed because previous version of marlin 1.1.9 does not a option for setting second nozzle Z offset but marlin 2 and higher version have the option to set Z offset but mesh bed levelling itself not working...thanking you.

boelle, 2020-07-27:
And use [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) not old outdated versions

shayan123869, 2020-07-27:
Thank you for your response....in 15 to 20 mins I will upload the
configuration.h and configurationadv.h files

Thanking you
Shayan
INDIA

shayan123869, 2020-07-27:
I have attached the zip files. I have added //shayan where ever I have done changes for reference purpose and also I have added E2 extruder to RAMPS board in pins.h

in configurationadv.h I have only altered dual x carriage (values only) while all others remain untouched.

[Marlin-bugfix-2.0.x_SHAYAN.zip](https://github.com/MarlinFirmware/Marlin/files/4981818/Marlin-bugfix-2.0.x_SHAYAN.zip)

boelle, 2020-07-27:
have you tried [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x)?

boelle, 2020-07-27:
and remember you can use a tool like winmerge to copy settings from old to new files

shayan123869, 2020-07-27:
i have tried all till 2.0.5 including bug fix branches.....as you say let me try winmerg to copy files from 1.1.9 to marlin 2.0.6 and revert back.

shayan123869, 2020-07-27:
i will try attaching a small video for better reference

borland1, 2020-07-27:
Shayan, 
Under `#define LCD_BED_LEVELING`
Did you try enabling...
`#define MESH_EDIT_MENU`
to see if that makes any difference?

shayan123869, 2020-07-27:
[20200727_232016 (1).zip](https://github.com/MarlinFirmware/Marlin/files/4983650/20200727_232016.1.zip)
dear sir....yes I tried mesh edit menu also but no changes.i am attaching a small 12 seconds video clip for reference purpose...sorry for the bad video quality.

borland1, 2020-07-27:
OK, I think you will find manual leveling much easier to do with LCD menu. so you might want to enable MESH_EDIT_MENU.
Also, try enabling MANUAL_PROBE_START_Z 0.2
```
#define PROBE_MANUALLY    //shayan
//#define MANUAL_PROBE_START_Z 0.2
```
Also consider, by default, you need to manually load your mesh settings previously saved to EEPROM.  Or you can modify Configuration_adv.h to use Startup G-code..
```
/**
 * Startup commands
 *
 * Execute certain G-code commands immediately after power-on.
 */
#define STARTUP_COMMANDS "M420 S1" 
```

shayan123869, 2020-08-21:
hi,
sorry for the late reply as I was carried away by my work... I tried enabling both 

#define PROBE_MANUALLY    //shayan
//#define MANUAL_PROBE_START_Z 0.2

and 

#define STARTUP_COMMANDS "M420 S1" 

but still, the same issue the manual bed leveling not working. Can any please help me out because marlin2.0.6 has an option to adjust nozzle 2 Z offset while performing bed leveling which is very helpful for IDEX. without the option to adjust nozzle 2 Z offset i will have to adjust it in the hardware which is a frustrating process.

thank you

closed 2020-09-28 by github-actions[bot]