MarlinFirmware/Marlin #18805: [FR] Set chamber fan speed with gcode

ogland, 2020-07-27:
I'd like to be able to set the fan percentage for the chamber fan manually via gcode, in order to have it vent differently for different material types.

If I choose to print with PLA, I might want the chamber fan at full speeds at all times, so I'd set the chamfer fan to max in the PLA filament gcode. 

For ABS I'd like it completely off or set it to a low percentage, in order to filter it through my HEPA filter in front of the fan.

1. Set fan speed in filament gcode
2. Start print
3. Good looking print

#### Additional Information

I successfully added another fan in an old version of marlin 2.0 which let me control it like this, but it was a major hassle for a non coder like me and very hard to move over to newer versions.

ellensp, 2020-07-27:
You don't set the fan speed you set the temperature you want with M141 or M191

ogland, 2020-07-27:
> You don't set the fan speed you set the temperature you want with M141 or M191

Yes. I'm aware of the current functionality. It isn't good for me though. My vent fan is very powerful and I'm afraid ABS would warp very much if it suddenly kicked on at 100%.

ellensp, 2020-07-27:
Yes setting a temperature it needs an active heater, good point.

thisiskeithb, 2020-07-28:
Would [M106](https://marlinfw.org/docs/gcode/M106.html) suffice? You can set the fan index using the `P` parameter.

ogland, 2020-07-28:
M106 would be great but you can't control it like that for now. I managed to add another fan like that to an older version but it was a lot of work that I didn't fully understand.

ogland, 2020-07-29:
Turns out it's fixed already: https://github.com/MarlinFirmware/Marlin/issues/12961

closed 2020-07-29 by ogland