MarlinFirmware/Marlin #18807: [BUG] 2.0.6 Boot screen shows "Marlin 2.0.5.4"

salacpavel, 2020-07-27:
### Bug Description

As a title says

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

Compile 2.0.6 and flash it
Power on

**Expected behavior:** 
Default boot screen should report Marlin firmware 2.0.6

**Actual behavior:**
Reports 2.0.5.4 instead

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4981884/Marlin.zip)

ellensp, 2020-07-27:
That is not Marlin 2.0.6
You can only download marlin 2.0.6 via GIT or from this link https://github.com/MarlinFirmware/Marlin/archive/2.0.6.zip
2.0.x.zip has not been updated to 2.0.6 yet.

boelle, 2020-07-27:
could we close this one allready?

ellensp, 2020-07-27:
I was going to give them 30 mins to respond.

salacpavel, 2020-07-27:
I have downloaded Marlin from https://marlinfw.org/meta/download/
It refers to 2.06 right there:
[image]
Indeed, configuration files are plain copy & paste from my 2.0.5.3 build - but that should not make a difference here

boelle, 2020-07-27:
I'm tempted to agree with @salacpavel here

i use bugfix 2.0.x so i have never cared much for version numbers

salacpavel, 2020-07-27:
In any case, i.e. version.h is different between versions downloaded from Marlinfw.org download page and the package linked by ellensp.
My guess is marlinfw downloadpage has only updated number, but dowload link still points to a previous version.

ellensp, 2020-07-27:
as I already stated 2.0.x.zip has not been updated to 2.0.6 yet. That page links through to 2.0.x.zip
I have submitted a PR to update it, but things take time.

closed 2020-07-27 by ellensp