MarlinFirmware/Marlin #18809: [BUG] MAX31865 Incorrect Calibration Resistance Value

zervin, 2020-07-27:
[RelatedConfigs.zip](https://github.com/MarlinFirmware/Marlin/files/4982832/RelatedConfigs.zip)

The max31865 calibration resistance is set to 400 in the temperature.cpp. I am able to get the system to read temperatures, but the values are incorrect. The MAX31865 typically uses a 430 for reference. Modifying the temperature.cpp from 400 to 430 results in the correct calculation. The current ambient temprature in the space is 29.4c. At 400, the extruder reads 9c, and 430 it reads 30c.

sjasonsmith, 2020-08-09:
The MAX31865 manual mentions 400 ohms, although I also see that AdaFruit installs 430 ohm.

Perhaps this needs a configuration option, rather than a simpe value change?

zervin, 2020-08-09:
Quite possible. It probably should as least be documented somewhere. I didn't find a clear rationale for one versus the other at the time. I only have 1 type, so I can't test if it matters when switching between pt1000 and pt100, or if it's maybe a different board revision.

AnHardt, 2020-08-09:
> I only have 1 type, so I can't test if it matters when switching between pt1000 and pt100, or if it's maybe a different board revision.

You can't simply connect a PT1000 to a module made for PT100. The modules for PT1000 differ in having a 10 times higher reference resistor. The [Adafruit PT1000 modules](https://www.adafruit.com/product/3648) have a 4300 Ohm resistor.

The 400 Ohm value comes from the [MAXIM-datasheet](https://cdn-learn.adafruit.com/assets/assets/000/037/309/original/MAX31865.pdf) Page 20. It reads: "In the typical case of a PT100 RTD with a 400 Ohm high precision low drift reference resistor, ...". Its just a suggestion to get the maximum resolution over the complete range of temperatures. Higher values of the resistor (like 430) result in a reduced selfheating of the PT-resistor. As always you have to search for a compromise.

Is anyone aware of a MAX31865-module not being from Adafruit (or a clone)?

There is a [sub-board for the DUET](https://reprapworld.se/products/electronics/duet3d/duet3d_pt100_daughterboard_for_duet/) where i have not data for. But as long Marlin is not running on the DUET-Baoards - ... .

wallaceowen, 2020-09-02:
All I needed to do to get a PT100 from Adafruit (with the 430 ohm resistor) was change the 400 to 430.  But it seems like bad practice of the first order to have that raw 400 numerical constant in a .cpp file

I would think that simply the fact that you've got a hard-coded number with no docs would be sufficient warrant to make a change that creates a #define for an appropriately named constant in a header file.  Then I can #define an alternate value in Marlin/Configuration.h, so I can override it and not have to go searching through the code for magic numbers.

Quas7, 2020-09-12:
maybe adding another temperature probe number for MAX31856 with 430Ohm reference seems logical to me

AnHardt, 2020-09-12:
[image]
It's unlikely a module with a 400 Ohm reference resistor is existent. If @Bob-the-Kuhn has not one he tested with, the 400 is just an error.

wallaceowen, 2020-10-12:
I've been asked if I want to add a comment to keep this issue alive.

I'm kind of discouraged that it's down to me.

I found the issue and have modified my local checkout of 2.0.6, so this doesn't impact me.

I'm discouraged because no one has stepped up to make a change to the code to either
1. do the minimum and change the magic number 400 to the magic number 430 in two places
2. do the right thing and #define a constant or an enumeration of the values used by various vendors (sounds like it rounds to one).

sad. not complaining, I know it's because lack of resources.  and that's why I'm sad.

sjasonsmith, 2020-10-12:
I posted a pull request to allow this configuration. I don't use these parts, so did not actually test it, but it is a low risk change.

This change is not important to me personally, so if you with to see it (or something similar) merged please go share your support or concerns on that PR.

#19695

boelle, 2020-10-13:
@zervin did the PR from jason help?

zervin, 2020-10-13:
Yes, I think that is the proper way to define it from the looks of the commit that will do nicely. It will be much easier to change versus digging through source. Thanks!

closed 2020-10-13 by zervin