MarlinFirmware/Marlin #18813: [BUG] Unable to Assign FAN_PIN to E0_AUTO_FAN_E0_PIN for Auto Fan Feature with Single Fan Printers

borland1, 2020-07-27:
### Bug Description
Enabling extruder Auto Fan feature in Configuration_adv.h as follows..

`#define E0_AUTO_FAN_PIN 9  // Fan Connector (pin D9) on MKS_GEN_L board`

produces a complier error "Cannot assign FAN_PIN to E0_AUTO_FAN_PIN".

This is for single extruder, single fan, printer.

I have tried disabling FAN_PIN and enabling Auto Fan in Configuration_adv.h as follows...
```
#define FAN_PIN -1
#define E0_AUTO_FAN_PIN 9
```
and this compiles with success, and Auto Fan works, but Fan Speed cannot be set with LCD menu.  LCD menu also shows two Fans when only one is configured.

### My Configurations

Config files ( bug-fix-2.0.x ):

[ConfigFiles.zip](https://github.com/MarlinFirmware/Marlin/files/4984857/ConfigFiles.zip)

borland1, 2020-08-01:
I found a workaround for this bug where the LCD's Temperature menu showing two Fans, when only one is configured.  I posted the workaround in issue #18529 .

GMagician, 2020-08-01:
I think this is normal...
E0_AUTO_FAN_PIN can only be used on pins not used by fan. First fan can' be used because is supposed to be used by...fan itself

closed 2020-09-06 by github-actions[bot]