MarlinFirmware/Marlin #18816: [BUG] Archim 2 will not upload compiled firmware via VSCode

AFprinter, 2020-07-28:
### Bug Description

Upon completion of compiling Marlin in VSCode it will not upload to the board via USB.

### Error message produced
CONFIGURATION: https://docs.platformio.org/page/boards/atmelsam/due.html
PLATFORM: Atmel SAM 4.5.1 > Arduino Due (Programming Port)
HARDWARE: AT91SAM3X8E 84MHz, 96KB RAM, 512KB Flash
DEBUG: Current (atmel-ice) External (atmel-ice, blackmagic, jlink, stlink)
PACKAGES:
 - framework-arduino-sam 1.6.12
 - framework-cmsis 1.40500.0 (4.5.0)
 - framework-cmsis-atmel 1.2.0
 - tool-bossac 1.10700.190624 (1.7.0)
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 5 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <TMCStepper> 0.7.1
|   |-- <SPI> 1.0
|-- <LiquidCrystal> 1.5.0
|   |-- <Wire> 1.0
|-- <Wire> 1.0
|-- <SPI> 1.0
Building in release mode
Checking size .pio\build\DUE_archim\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   9.4% (used 9284 bytes from 98304 bytes)
Flash: [=====     ]  45.8% (used 239972 bytes from 524288 bytes)
Configuring upload protocol...
AVAILABLE: atmel-ice, blackmagic, jlink, sam-ba, stlink
CURRENT: upload_protocol = sam-ba
Looking for upload port...
Auto-detected: COM3
Forcing reset using 1200bps open/close on port COM3
Uploading .pio\build\DUE_archim\firmware.bin
**### bossac: missing file
Try 'bossac -h' or 'bossac --help' for more information
*** [upload] Error 1**

### Steps to Reproduce

1. Compile Marlin
2. Try to upload to Archim 2 board

**Expected behavior:**Completion of Upload of firmware to the board
**Actual behavior:** Fails

ellensp, 2020-07-28:
In platformio is 

```
[common_DUE_archim]
platform      = atmelsam
board         = due
src_filter    = ${common.default_src_filter} +<src/HAL/DUE>
build_flags   = ${common.build_flags}
  -DARDUINO_SAM_ARCHIM -DARDUINO_ARCH_SAM -D__SAM3X8E__ -DUSBCON
extra_scripts = ${common.extra_scripts}
  Marlin/src/HAL/DUE/upload_extra_script.py
```

It doesn't seem to need to the upload_extra_script.py anymore. In fact it seems to break windows uploading. 
So change to 
```
[common_DUE_archim]
platform      = atmelsam
board         = due
src_filter    = ${common.default_src_filter} +<src/HAL/DUE>
build_flags   = ${common.build_flags}
  -DARDUINO_SAM_ARCHIM -DARDUINO_ARCH_SAM -D__SAM3X8E__ -DUSBCON
extra_scripts = ${common.extra_scripts}
```

Can you confirm that this works for you also ?

ellensp, 2020-07-28:
Waiting on conformation before releasing PR

AFprinter, 2020-07-28:
I deleted the extra line as recommended and it actually connected this time but wont upload.  Here's the new error. 

Configuring upload protocol...
AVAILABLE: atmel-ice, blackmagic, jlink, sam-ba, stlink
CURRENT: upload_protocol = sam-ba
Looking for upload port...
Auto-detected: COM3
Forcing reset using 1200bps open/close on port COM3
Uploading .pio\build\DUE_archim\firmware.bin
Atmel SMART device 0x285e0a60 found

SAM-BA operation failed
*** [upload] Error 1

AFprinter, 2020-07-28:
To be clear I can upload a Marlin Bugfix from March that I've been using via Arduino IDE just fine, even as of today.  As of July 7 Bugfix I cannot use Arduino IDE for compiling or upload, something changed.  And I've tried using the atmel-ice protocol as well as the SAM-BA that seems to be the default as selected in the platform.ini and both fail.  The SAM-BA seems to get the furthest as it actually seems to make the connection to the board, it's just not pushing the file now, as per my last post.

AFprinter, 2020-07-29:
So after messing with this a little bit, I changed 
board         = due 
to
board         = dueUSB

and **I got it to upload** but now after restarting the printer the screen is just displaying random crap on the LCD (yes I double checked that the correct LCD was uncommented) 
[image]
and the **y axis only moves in one direction** (endstops are open) no matter which way I tell it to move in octoprint (yes I can connect via octoprint).  I can move the x axis in both directions and the z axis in both directions.  I can also interface with Marlin via the Octoprint terminal and use commands. 

[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/4991745/Configurations.zip)

jeffeb3, 2020-07-31:
Those are some goofy symptoms. Have you compared the configuration and pins files from the two versions?

colinoflynn, 2020-09-03:
The issue with the pins is also called out in #19215, these are the same root problems.

Archim2 requires a special Arduino DUE environment, as it uses pins 92+ (these pins are beyond the normal Arduino DUE pins). See line https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/Makefile#L698 for example showing the inclusion of `variants.cpp` which I assume solves the problem(?). I put more details in #19215.

I'm not sure on the correct way to solve this as I'm not too familiar with either PlatformIO build system, or Marlin (new to both!).

jeffeb3, 2020-10-05:
#19545 and #19546 might have helped these two issues. We had the same problems (bossac error and steopers not moving) and they seem to be fixed now.

sjasonsmith, 2020-10-28:
Several related fixes were introduced recently, and there haven’t been any replies regarding this still being an issue since before those were made.

I am going to close this, as I suspect it is fixed from comments made by @jeffeb3 and @colinoflynn.

Please let me know if this action was incorrect.

closed 2020-10-28 by sjasonsmith