MarlinFirmware/Marlin #18818: [FR] Save to SDcard for thermal eroors.

firepower9966, 2020-07-28:
#4732 # Description

When thermal error occurs with hotend such as Thermal runaway a error message is displayed and alarm sounded. 
you need to reset printer to resume.

Is it possible to also save to SDCARD the same as power loss recovery does to allow you to resume print when you power up.

AnHardt, 2020-07-28:
Thermal errors (MIN- MAX. TEMP, Heating failed, Thermal runaway) do not occur once the system is correctly tuned - except the hardware, the parameters have been tuned for, has changed (hardware failure). In that case a simple resume, without checking the hardware before, is not very wise.

firepower9966, 2020-07-28:
sometimes i just get them very rarely , if it could allow resume and save a print then it would be a benefit, especially when the code is already their and just needs to be called. maybe have it an option in config to enable.

sjasonsmith, 2020-08-09:
> sometimes i just get them very rarely and intermittent 

This is likely caused by a broken thermistor wire which loses connection when the bed or nozzle are in certain positions.

firepower9966, 2020-08-09:
I have checked my wiring and unplugged and plugged back in thermistor and heaters, have not had fault since then, so this FR is really not need as other have stated.