MarlinFirmware/Marlin #18820: [BUG-I think] I keep getting this error no matter what versions I use

Grafur1, 2020-07-28:
Marlin\src\HAL\STM32F1\SPI.cpp:52:40: error: 'spi_dev' was not declared in this scope
 static const spi_pins* dev_to_spi_pins(spi_dev *dev);
                                        ^~~~~~~
Marlin\src\HAL\STM32F1\SPI.cpp:52:40: note: suggested alternative: 'gpio_dev'
 static const spi_pins* dev_to_spi_pins(spi_dev *dev);
                                        ^~~~~~~
                                        gpio_dev
Marlin\src\HAL\STM32F1\SPI.cpp:52:49: error: 'dev' was not declared in this scope
 static const spi_pins* dev_to_spi_pins(spi_dev *dev);
                                                 ^~~
Marlin\src\HAL\STM32F1\SPI.cpp:52:49: note: suggested alternative: 'div'
 static const spi_pins* dev_to_spi_pins(spi_dev *dev);
                                                 ^~~
                                                 div
Marlin\src\HAL\STM32F1\SPI.cpp:53:29: error: variable or field 'configure_gpios' declared void
 static void configure_gpios(spi_dev *dev, bool as_master);
                             ^~~~~~~
Marlin\src\HAL\STM32F1\SPI.cpp:53:29: error: 'spi_dev' was not declared in this scope
Marlin\src\HAL\STM32F1\SPI.cpp:53:29: note: suggested alternative: 'gpio_dev'
 static void configure_gpios(spi_dev *dev, bool as_master);
                             ^~~~~~~
                             gpio_dev
Marlin\src\HAL\STM32F1\SPI.cpp:53:38: error: 'dev' was not declared in this scope
 static void configure_gpios(spi_dev *dev, bool as_master);
                                      ^~~
Marlin\src\HAL\STM32F1\SPI.cpp:53:38: note: suggested alternative: 'div'
 static void configure_gpios(spi_dev *dev, bool as_master);
                                      ^~~
                                      div
compilation terminated due to -fmax-errors=5.
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\SPI.cpp.o] Error 1
In file included from c:\users\emil\marlin\marlin\src\hal\hal.h:26:0,
                 from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\STM32F1\HAL.cpp:30:
c:\users\emil\marlin\marlin\src\hal\stm32f1\hal.h:81:4: error: #error "SERIAL_PORT cannot be 0. (Port 0 does not exist.) Please update your configuration."
   #error "SERIAL_PORT cannot be 0. (Port 0 does not exist.) Please update your configuration."
    ^~~~~
In file included from c:\users\emil\marlin\marlin\src\hal\hal.h:26:0,
                 from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\STM32F1\HAL_SPI.cpp:35:
c:\users\emil\marlin\marlin\src\hal\stm32f1\hal.h:81:4: error: #error "SERIAL_PORT cannot be 0. (Port 0 does not exist.) Please update your configuration."
   #error "SERIAL_PORT cannot be 0. (Port 0 does not exist.) Please update your configuration."
    ^~~~~
In file included from c:\users\emil\marlin\marlin\src\hal\hal.h:26:0,
                 from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\STM32F1\Servo.cpp:25:
c:\users\emil\marlin\marlin\src\hal\stm32f1\hal.h:81:4: error: #error "SERIAL_PORT cannot be 0. (Port 0 does not exist.) Please update your configuration."
   #error "SERIAL_PORT cannot be 0. (Port 0 does not exist.) Please update your configuration."
    ^~~~~
In file included from c:\users\emil\marlin\marlin\src\hal\hal.h:26:0,
                 from c:\users\emil\marlin\marlin\src\inc\marlinconfig.h:30,
                 from Marlin\src\HAL\STM32F1\dogm\u8g_com_stm32duino_fsmc.cpp:29:
c:\users\emil\marlin\marlin\src\hal\stm32f1\hal.h:81:4: error: #error "SERIAL_PORT cannot be 0. (Port 0 does not exist.) Please update your configuration."
   #error "SERIAL_PORT cannot be 0. (Port 0 does not exist.) Please update your configuration."
    ^~~~~
In file included from c:\users\emil\marlin\marlin\src\hal\hal.h:26:0,
                 from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\STM32F1\eeprom_bl24cxx.cpp:28:
c:\users\emil\marlin\marlin\src\hal\stm32f1\hal.h:81:4: error: #error "SERIAL_PORT cannot be 0. (Port 0 does not exist.) Please update your configuration."
   #error "SERIAL_PORT cannot be 0. (Port 0 does not exist.) Please update your configuration."
    ^~~~~
In file included from c:\users\emil\marlin\marlin\src\hal\hal.h:26:0,
                 from c:\users\emil\marlin\marlin\src\inc\marlinconfig.h:30,
                 from Marlin\src\HAL\STM32F1\dogm\u8g_com_stm32duino_swspi.cpp:21:
c:\users\emil\marlin\marlin\src\hal\stm32f1\hal.h:81:4: error: #error "SERIAL_PORT cannot be 0. (Port 0 does not exist.) Please update your configuration."
   #error "SERIAL_PORT cannot be 0. (Port 0 does not exist.) Please update your configuration."
    ^~~~~
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\HAL.cpp.o] Error 1
Marlin\src\HAL\STM32F1\HAL_SPI.cpp: In function 'void spiInit(uint8_t)':
Marlin\src\HAL\STM32F1\HAL_SPI.cpp:87:27: error: 'SPI_CLOCK_DIV4' was not declared in this scope
     #define SPI_CLOCK_MAX SPI_CLOCK_DIV4
                           ^
Marlin\src\HAL\STM32F1\HAL_SPI.cpp:93:37: note: in expansion of macro 'SPI_CLOCK_MAX'
     case SPI_FULL_SPEED:    clock = SPI_CLOCK_MAX ;  break;
                                     ^~~~~~~~~~~~~
Marlin\src\HAL\STM32F1\HAL_SPI.cpp:87:27: note: suggested alternative: 'SPI_CLOCK_MAX'
     #define SPI_CLOCK_MAX SPI_CLOCK_DIV4
                           ^
Marlin\src\HAL\STM32F1\HAL_SPI.cpp:93:37: note: in expansion of macro 'SPI_CLOCK_MAX'
     case SPI_FULL_SPEED:    clock = SPI_CLOCK_MAX ;  break;
                                     ^~~~~~~~~~~~~
Marlin\src\HAL\STM32F1\HAL_SPI.cpp:95:37: error: 'SPI_CLOCK_DIV8' was not declared in this scope
     case SPI_QUARTER_SPEED: clock = SPI_CLOCK_DIV8 ; break;
                                     ^~~~~~~~~~~~~~
Marlin\src\HAL\STM32F1\HAL_SPI.cpp:95:37: note: suggested alternative: 'SPI_CLOCK_MAX'
     case SPI_QUARTER_SPEED: clock = SPI_CLOCK_DIV8 ; break;
                                     ^~~~~~~~~~~~~~
                                     SPI_CLOCK_MAX
Marlin\src\HAL\STM32F1\HAL_SPI.cpp:96:37: error: 'SPI_CLOCK_DIV16' was not declared in this scope
     case SPI_EIGHTH_SPEED:  clock = SPI_CLOCK_DIV16; break;
                                     ^~~~~~~~~~~~~~~
Marlin\src\HAL\STM32F1\HAL_SPI.cpp:96:37: note: suggested alternative: 'SPI_CLOCK_MAX'
     case SPI_EIGHTH_SPEED:  clock = SPI_CLOCK_DIV16; break;
                                     ^~~~~~~~~~~~~~~
                                     SPI_CLOCK_MAX
Marlin\src\HAL\STM32F1\HAL_SPI.cpp:97:37: error: 'SPI_CLOCK_DIV32' was not declared in this scope
     case SPI_SPEED_5:       clock = SPI_CLOCK_DIV32; break;
                                     ^~~~~~~~~~~~~~~
Marlin\src\HAL\STM32F1\HAL_SPI.cpp:97:37: note: suggested alternative: 'SPI_CLOCK_MAX'
     case SPI_SPEED_5:       clock = SPI_CLOCK_DIV32; break;
                                     ^~~~~~~~~~~~~~~
                                     SPI_CLOCK_MAX
compilation terminated due to -fmax-errors=5.
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\HAL_SPI.cpp.o] Error 1
In file included from c:\users\emil\marlin\marlin\src\hal\hal.h:26:0,
                 from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\STM32F1\eeprom_flash.cpp:32:
c:\users\emil\marlin\marlin\src\hal\stm32f1\hal.h:81:4: error: #error "SERIAL_PORT cannot be 0. (Port 0 does not exist.) Please update your configuration."
   #error "SERIAL_PORT cannot be 0. (Port 0 does not exist.) Please update your configuration."
    ^~~~~
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\dogm\u8g_com_stm32duino_fsmc.cpp.o] Error 1
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\Servo.cpp.o] Error 1
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\eeprom_bl24cxx.cpp.o] Error 1
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\dogm\u8g_com_stm32duino_swspi.cpp.o] Error 1
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\eeprom_flash.cpp.o] Error 1``

I trying to make it for a BTT SKR-Mini-E3. for this one, I'm using 2.0.6, the only change I made: 

#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_BTT_SKR_MINI_E3_V1_2
#endif

ellensp, 2020-07-28:
Something is wrong with your build environment. Uninstall platformio and remove .platfomio directory. Then reinstall platformio

Grafur1, 2020-07-28:
It worked, I did not know about the .platfomio folder. Thank you so much for your help. 👍

closed 2020-07-28 by Grafur1