MarlinFirmware/Marlin #18821: [BUG] Can't live adjust z-offset (skr 1.4 turbo)

leewaa, 2020-07-28:
I'm running the latest version of Marlin for the SKR 1.4 Turbo

I cannot live adjust the Z-Offset during a print. This is normally the way to go when getting the z-offset set correctly initially.

I had even tried the paper method before hand but for what ever reason the z-offset is way off even after doing this (Initially set with this method: https://3dprinting.stackexchange.com/a/5858)

Even after setting this the z-offset is still incorrect during printing. Normally I would (with ender 5 plus standard firmware) I would live adjust the z-offset to get it perfect.

However with Marlin, this does not work, either with Gcode commands (M851, M500) Or using the display (configuration -> z-offset, then -> store settings).

IT would be nice to have this feature working just as with the standard Creality firmware does where one can easily make fine adjustments to the Z-Offset during the print.

Maybe its a problem with the board I am using ? (skr 1.4 turbo) or the screen ? (TFT35)
Right now I have to start a print, check if tis printing correctly, if not cancel the print, adjust the z-offset, start another print, etc etc.....

ellensp, 2020-07-28:
Please attach your configuration files.

leewaa, 2020-07-28:
Sure: [config.zip](https://github.com/MarlinFirmware/Marlin/files/4987165/config.zip)

ellensp, 2020-07-28:
Your Configuration shows CONFIGURATION_H_VERSION 020000  this is a really old version of marlin. 
current is 20006
Please try a current version.

boelle, 2020-07-28:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

leewaa, 2020-07-28:
Oh, thanks for pointing that out!

Actually I complete forgot (sorry) I had taken the firmware from Bigtreetech: https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/tree/master/BTT%20SKR%20V1.4/Firmware/Marlin-2.0.x-SKR-V1.4-Turbo

I guess their not up to date.

I will test it out when I get time tonight or tomorrow and report back!! Maybe just close it since I am a different fork of Marlin (from big tree tech) and if the problem re-occurs I will just open up the issue again :)

ellensp, 2020-07-28:
when you get firmware from anywhere other than the official marlin repository https://github.com/MarlinFirmware/Marlin, then you need to get assistance from whomever runs that repository. Or switch to real Marlin.

closed 2020-07-28 by ellensp

leewaa, 2020-07-28:
> when you get firmware from anywhere other than the official marlin repository https://github.com/MarlinFirmware/Marlin, then you need to get assistance from whomever runs that repository. Or switch to real Marlin.

Yes this goes without saying. Sorry my mistake!