MarlinFirmware/Marlin #18825: Ender 3v2 - DWIN_SET - "1.English.jpg" not updating during flash

mjmurray68, 2020-07-28:
### Bug Description

When flashing DWIN_SET from the current Configuration for Ender V2, the display menu does not reflect the menu changes in the 1.English.jpg file. Specifically the "Store Configuration" menu item:
[image]

After flashing, the menu item remains "Storage configuration" although the image selection being made from dwin.ccp appears to be correct as the word "Storage" is being cut off: 
[image]
[image]

### My Configurations

[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/4988130/Configs.zip)

### Steps to Reproduce

1. Copy latest "DWIN_Set Folder from Ender 3 V2 Configurations to root of SD Card
2. Insert SD card into Display SD Slot
3. Power on Printer
4. Wait for Blue Screen to Turn Orange.
5. Power off Printer/Remove SD Card
6. Power on Printer/Navigate to Control Menu

**Expected behavior:** [What you expect to happen]
Expect "Storage Configuration" menu item to be changed to "Store configuration"

**Actual behavior:** [What actually happens]
Menu Item does not change after flash.

#### Additional Information

Available as requested.

fleek, 2020-07-28:
I don't think updating the .jpg file without recompiling the .bin file will work.  I believe you need a tool to combine the icon files and .jpgs into the bin file which is updated into the DWIN DGUS device.

mjmurray68, 2020-07-28:
@fleek I totally concur, but I didn't update the image, it was part of the Configurations release. So it may mean that the .bin file for the display, in the DWIN_SET directory hasn't been recompiled in the current Configurations release.

fleek, 2020-07-28:
@mjmurray68 Yes, there is no clarified workflow for updating the DWIN DGUS menu pages and icons, though some people have figured out how to update it, but they are not sharing. I have visited the DWIN DGUS website and downloaded tools to try to change the icons and amend the messages but to no avail.  It will probably take some time.  So hopefully you can live with this minor inconvenience atm until we figure out the tools required.  Even writing to Creality has not been forthcoming in supporting my endeavor.

mjmurray68, 2020-07-28:
@fleek Absolutely can live with it! I just wanted to make sure, given that the image changed, that the intent wasn't to have an updated DWIN DGUS bin as part of the release, and have it not working.

fleek, 2020-07-28:
@mjmurray68 Sorry for having misled you but the developers and contributors are all still figuring out this DWIN DGUS thingy.  Hopefully we can get more info and people with the knowledge are willing to share the actual workflow. As of now, we cannot commit to providing a solution.  Hopefully, you can close this issue but no worries, I will be working on the M600 issue and also the relative movement issue which will potential solve this issue of yours.

mjmurray68, 2020-07-28:
Not a Bug - Closing pending future DWIN DGUS Workflow.

closed 2020-07-28 by mjmurray68

thinkyhead, 2020-07-29:
The only tools available for converting / editing the ICO files (and others) are for Windows/ I don't have a working Windows VM or WINE layer right now so I haven't tried them. For the menu strings in the JPG file I did some small tweaks to both the graphics and to the code to align them a bit better. Any changes to the language JPG files will need to be accompanied by changes to the `dwin.cpp` file which draws them.

I've simplified the menu item drawing a lot in `dwin.cpp` compared to the original code from Creality, which had a lot of repetition in it. The cleanest way to manage new string pictures is to follow the same pattern that I've established there.

All that said, my intention is to implement all the `MarlinUI` methods for the Ender 3 V2 screen so it can display the entire Marlin menu tree complete with select screens, edit screens, etc. In support of that effort I need to figure out whether the screen can display real fonts complete with accents so that it can display all of Marlin's languages, including Chinese — and then we don't have to put any menu titles into JPG files.

fleek, 2020-07-30:
@thinkyhead This is what I can pull from DWIN's website, seems that it can display fonts directly through the serial interface, but there is no option to select fonts.
[T5UIC1 Kernel Application Guide.pdf](https://github.com/MarlinFirmware/Marlin/files/4999113/T5UIC1.Kernel.Application.Guide.pdf)

If I read this documentation.  I believe to update the files on the smart screen, you need to take the screen out of the housing and there should be a SD card slot on the screen itself.  Not sure if the current firmware bootloader does read the DWIN_SET to update the smart screen, but I believe it probably does not.

mjmurray68, 2020-07-30:
@fleek - as part of my workflow, I had cracked open my display as well, and flashed with the DWIn_Set from Configurations, as per the README in the DWIN_Set Directory. I raised the issue when I saw the changes made by @thinkyhead referenced above didn't come across. Or more accurately, the changes to the dwin.cpp took, but not the changes to the image file). My end goal wasn't to correct the grammar of "Store Configurations" but rather to better understand if changes could be made to the UI, perchance to be able to leverage Manual Mesh Bed Leveling. I'm happy to wait.

fleek, 2020-07-30:
@mjmurray68 Sorry for hijacking your thread, but my intention is to share my investigations with Scott (thinkyhead), hoping I can help in anyway to advance the development of the MarlinUI.  Today, the smart display is actually a hindrance to incorporating certain functionalities of Marlin such as filament sensor (M600) which I needed and also some form of custom user menu which was available in my previous Anet A8.

cosmoderp, 2020-08-05:
@mjmurray68 why did you close this? it had an easy fix. I have been working on these menus for days and editing them is fun.  I am making a few new menus and trying to get rid of some repeated code. anyway for now here is your fix for this.  I still need to learn how to github. Until I am done reworking some of these menus I will at least give you this one fix since it seems to immediately annoy some people XD.

just change:
DWIN_Frame_AreaCopy(1, 131 + 17, 89, 271 -  3, 479 - 377 - 1, LBLX, CLINE(3));// "Store Configuration"
to
DWIN_Frame_AreaCopy(1, 131, 89, 271 -  3, 479 - 377 - 1, LBLX, CLINE(3));// "Store Configuration"
and it now reads correct

cosmoderp, 2020-08-06:
sorry for hijacking as well but there is nowhere else where people have info on this....@fleek since you seem to have found the proper version of DWIN icon file generator what version of DGUS tools did this come from.  there has to be one with a more up to date version.  best t5 one I found only had icon app v2?

cosmoderp, 2020-08-11:
I solved this and will also post here for people that have trouble in the future.
#1 the creality provided SD card does not work
#2 make the card you choose is formatted fat32 AND also your allocation size is 4k
I could not get it to work and when you format it fat 32 and its not 4k it will act like its flashing but in reality its failing every time
once I set the 4k allocation size bingo

you only need the single file you want to update so if you just want to update store config only add  1_English.JPG to the DWIN_SET folder and nothing else.