MarlinFirmware/Marlin #18829: [BUG] Compile error when using Neopixels on STM32F103 and Marlin 2.0.6

rselec, 2020-07-28:
### Bug Description

Standard Neopixel lib is not compiling with STM32F103 based boards, because something with the STM32 and the NEOPIXEL lib is incompatible.

There is another lib avalible which works:
https://github.com/ccccmagicboy/Adafruit_NeoPixel#meeb_3dp_use

Same behaviour also on Marlin 2.0.x bugfix from 24th July 2020.

### My Configurations

Board: BIGTREETECH SKR Mini E3 V2.0 https://www.biqu.equipment/products/bigtreetech-skr-mini-e3-v2-0-32-bit-control-board-integrated-tmc2209-uart-for-ender-3

See Zip

### Steps to Reproduce

Use Marlin release 2.0.6 and my config enabling NEOPIXELS.

**Expected behavior:** Compiles without errors and Neopixel LEDs should work

**Actual behavior:**
In file included from Marlin\src\feature/leds/neopixel.h:34:0,
                 from Marlin\src\feature/leds/leds.h:33,
                 from Marlin\src\MarlinCore.cpp:108:
.pio\libdeps\STM32F103RC_btt_512K\Adafruit NeoPixel/Adafruit_NeoPixel.h:361:3: error: 'GPIO_TypeDef' does not name a type
   GPIO_TypeDef *gpioPort;       ///< Output GPIO PORT
   ^~~~~~~~~~~~
*** [.pio\build\STM32F103RC_btt_512K\src\src\MarlinCore.cpp.o] Error 1

rselec, 2020-07-28:
[my_config.zip](https://github.com/MarlinFirmware/Marlin/files/4989359/my_config.zip)

brencerddwr, 2020-07-28:
This is a known issue with adafruit neopixel library, in your platformio.ini file find the NEOPIXEL_LED line and change it to the following:

`# NEOPIXEL_LED            = Adafruit NeoPixel@1.5.0'
'NEOPIXEL_LED            =https://github.com/bigtreetech/Adafruit_NeoPixel'

That should get you compiling, This is something that big tree tech needs to submit a pull request to adafruit to resolve..

thisiskeithb, 2020-07-28:
Just a couple weeks ago, Neopixels were not supported for `STM32F1` (https://github.com/MarlinFirmware/Marlin/issues/18676), so I'm not sure why they were removed from the `lib_ignore` list.

sjasonsmith, 2020-07-28:
@rhapsodyv

rhapsodyv, 2020-07-28:
> Just a couple weeks ago, Neopixels were not supported for `STM32F1` (#18676), so I'm not sure why they were removed from the `lib_ignore` list.

The new dependency script ignore all unused known libs. Seems his problem is using a wrong version.

> 'NEOPIXEL_LED =https://github.com/bigtreetech/Adafruit_NeoPixel'

Is it worked?

thisiskeithb, 2020-07-28:
The existing neopixel libraries weren’t compatible on `STM32F1`, particularly the SKR Mini E3 boards (well, they mostly worked, but were still bugged), which is why they were explicitly ignored.

thisiskeithb, 2020-07-28:
Maybe there should be a sanity check instead until neopixels fully work on that HAL?

rhapsodyv, 2020-07-28:
We can override the Adafruit_NeoPixel lib in the BTT env.
Just need to add it to BTT envs, so our script will not try to use the wrong version.
```
lib_deps=Adafruit NeoPixel=https://github.com/bigtreetech/Adafruit_NeoPixel'
```

thisiskeithb, 2020-07-28:
BigTreeTech's version is also bugged. See https://github.com/MarlinFirmware/Marlin/issues/18676#issuecomment-660338838

rhapsodyv, 2020-07-28:
> BigTreeTech's version is also bugged. See [#18676 (comment)](https://github.com/MarlinFirmware/Marlin/issues/18676#issuecomment-660338838)

Hum... So the user can't use Neopixels in any STM32F1 env, right?

If so, I agree 100% with you. We need add sanity check.

thisiskeithb, 2020-07-28:
They *mostly* work with BigTreeTech's neopixel fork, but it's still bugged and will likely cause more support issues if we add support until they've been fixed.

rselec, 2020-07-28:
The bigtreetech lib https://github.com/bigtreetech/Adafruit_NeoPixel doesn't work for me. The LEDs go on but don't change brightness or colour.

The lib I linked in the first post: https://github.com/ccccmagicboy/Adafruit_NeoPixel#meeb_3dp_use works for me. LEDs are off in the beginning, change colour from blue over purple to red while heating and became "white" during print. When they become green when the print finishes.

Could we make the used lib depened on the environment option?

I could also provide measuring equipment for the timings. Saleae Logic clone and a Rigol DS1054Z is available. But C++ knowledge is limited, especially on 32-Bit ARMs...

thisiskeithb, 2020-07-28:
Have you tested it with other SKR Mini E3 boards? It may work on the SKR Mini E3 V2, but other `STM32F1` boards have had issues with it. I made the suggestion to @rhapsodyv they add "Comment out this line to continue" like we've done for other sanity checks so users can still bypass the check if they want to proceed at their own risk.

rselec, 2020-07-28:
Hi, I don't have one of the other SKR Mini E3 boards.
But I've looked in the schematics of the older v1.2 version. There is a akward mosfet on the data pin of the neopixel connector, which most propably causes issues. They removed it on the v2.
So I would say, if it uses bitbanging on a gpio it have to work on any STM32F1 board as long as they installed a proper levelshifter to 5V level.

If I find some time, I could make some measurements about the timings of the data signal. For the neopixel LEDs, they have to be pretty exact to work with all LED strips.

thisiskeithb, 2020-07-30:
There’s now a sanity check until NeoPixels are fully supported: https://github.com/MarlinFirmware/Marlin/pull/18833

closed 2020-07-30 by thisiskeithb