MarlinFirmware/Marlin #18840: [BUG] COREXY only moving in X- direction

Protean-Man, 2020-07-28:
I am trying to figure out what is happening with my machine and it's movement. 

**Problem:** When moving in any direction on the X/Y axis the machine only moves in X- direction.
**Example:** Telling it to move >>> X+50 = X-50, Y-30 = X-30, Y+60 = X-60. 
Every movement is in the negative on the X axis. 

M119 shows all endstops open and fine. 
Homing X will home to Xmin but it is unable to backtrack for second endstop trigger since all movements are in the X- direction. 
Homing Y will move in the X- direction

I am confused and have even tested by switched off of CoreXY to regular X & Y separate movement and it moves in the correct directions on each motor. 

Z axis seems to do the same thing with movement and homing, only moving in a single direction.
I tried with a fresh Marlin build and set up CoreXY only, without anything else, and the same issue arises. 

Anyone else have this issue?

boelle, 2020-07-29:
@Protean-Man 

Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

Protean-Man, 2020-08-05:
I figured this one out. It was a faulty stepper driver. I don't know why this issue would happen because of it but when replacing with a new unused driver the issue disappeared - nothing else was changed in Marlin.

Just to recap what I saw with the faulty driver:
All movements were in the X- direction. 
When moving Y-, Y+, & X+ the carriage would only move in the X- direction.
The machine is CoreXY and was working perfectly for some time before hand. 
The driver died and the issue appeared.
Replacing the faulty driver with a new one fixed the issue.

closed 2020-08-05 by Protean-Man