MarlinFirmware/Marlin #18843: [FR] Z homing deceleration

VanessaE, 2020-07-29:
### Description

Marlin should decelerate on approach of the probe or end stop, when homing along Z (I don't see a particular need  for this along X/Y).

If you have a printer that can move along Z rather fast, such as a delta or a CoreXY, the sudden, instantaneous stop of the can induce vibration in the carriage(s) and whatever moves with it.

A delta's relatively low-mass, so there won't be much to worry about, but in a CoreXY, it's the bed that moves, and a bed can get pretty heavy on larger machines.  Both of my Hypercubes home Z at 100 mm/s (with much lower post-bump speeds), with 500 mm/s² acceleration on the smaller machine and 100 mm/s² on the large one.

This can lead to incorrect homing or ABL probing if the vibration does not settle before the second approach, and depending on the machine's design, the sudden stop can produce an undesirable \*clunk\* sound, or potentially pull the Z screw out of the motor coupler if the bed is massive enough.

For example, on my larger machine, that sudden stop after a long 100 mm/s home travel is enough to make the bed bounce a little, which in turn makes its Z probe pin bounce.  This is especially true after a print, as I have my end-of-print G-code set to park the bed at the bottom of the build volume, thus necessitating a long homing move on the next print, which means it'll definitely be doing 100 mm/s when it triggers the probe and slams-on the brakes.

Here's a short video on Youtube demonstrating it:

https://youtu.be/9LJKUuTFOQI

Play the video once through so you can hear the probe's pin rattle on the glass when it hits.  If you pause at 11s, then use comma and period to roll it back and forth frame by frame, you'll see the bed and even the probe's pin bouncing.

Of course, Marlin can't know where it is just after power-up, or if Z has turned off by `M84` or idle time-out.  But if it already knows where it is because it's been homed since then (or thinks it knows, so let's assume the user keeps her grubby paws off the Z screw(s) :stuck_out_tongue:), it should be able to anticipate the approach of a probe/endstop while homing, and decelerate.

Obviously I could just limit the Z travel speed, but that wastes time.

Others would say I should drive the bed with screws from opposing sides instead of sticking with this cantilevered setup, and I may do that eventually anyway (this printer is still WIP), but while that would decrease the effect, it won't eliminate it.

### Feature Workflow

1. Power the machine on.  Marlin doesn't know where it is.
2. Home all axes.
3. Use your host or a `G1` command to jog Z to its maximum limit, putting it as far as possible from the probe/endstop.
4. Crank up the Z homing speed and set a reasonable acceleration that the hardware is happy with.
5. Home Z, observe how harshly it stops when the probe/endstop triggers on the initial approach.
6. Marlin now knows where it's at along Z.  Imagine at this point that you've just printed something.
7. Move Z back to max, as before.
8. Home Z again.  Now, since Marlin knows where it is, it should be able to anticipate when it's about to hit the probe/endstop, and should begin decelerating, i.e. do what it would normally do in a `G1` move.

#### Additional Information

* Provide pictures or links that demonstrate a similar feature or concept.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

thinkyhead, 2020-07-29:
My job is to come up with caveats. So here is the first one off the top of my head….

In this new behavior mode, on _**any failure**_ to trigger the endstop or the probe, instead of immediately assuming a failure, Marlin will have to do another (full length) move in the **same direction** and **try to hit the endstop or probe again**.

This might be a rare condition, but it seems like this additional "try again" move could lead to additional damage and frustration.

VanessaE, 2020-07-29:
I'm not entirely sure where "try again" comes from here?  Any time I have a homing failure, like if a motor stalls, Marlin just aborts without trying again, tells me to reset the printer.  Maybe there's some option you're thinking of, that I don't use?

I figure, in practice, if Marlin were to overestimate the distance to the probe/endstop, it's still gonna hit and trigger it when it gets there, and hit the brakes, even if it's still going at full speed -- just like now.

On the other hand, if it underestimates, then it'll have slowed down to the jerk speed by the time it gets there.  I Imagine this ought to trigger a homing failure, but I'm not sure what would be a reasonable threshold; I suppose in if Marlin were to \*only\* use deceleration during a "perfect" guess, and always slightly overestimate the distance (by say 1 mm), then it could just throw a normal homing error if it managed to slow down to the configured jerk speed before triggering the probe/endstop.

VanessaE, 2020-07-29:
Oh, it occurred to be just after I submitted that last remark, maybe you're thinking of what I wrote under "Feature Workflow"?  I meant that more as a list of steps to reproduce the undesired behavior, rather than how the feature should behave, since the description already covers that part.

"Sorry about that, Chief."