MarlinFirmware/Marlin #18847: [BUG] Wrong indication for printer statistic & Service periob

trengtor, 2020-07-29:
### Bug Description

Wrong indication for printer statistic (2.0.5.3)

[image]

[image]

[image]

Also servise period message dislayed for each ON of printer. Resetting the interval does not help clear this message.

### My Configurations

[For Marlin 2.0.5.3.zip](https://github.com/MarlinFirmware/Marlin/files/4994394/For.Marlin.2.0.5.3.zip)

SKR mini E3 V1.2 & BTT TFT24 V1.1

### Steps to Reproduce

1. Enable in configs

#define PRINTCOUNTER

/**
 * Advanced Print Counter settings
 */
#if ENABLED(PRINTCOUNTER)
  #define SERVICE_WARNING_BUZZES  5
  // Activate up to 3 service interval watchdogs
  #define SERVICE_NAME_1      "Service 100hr"
  #define SERVICE_INTERVAL_1  100 // print hours
  //#define SERVICE_NAME_2      "Service L"
  //#define SERVICE_INTERVAL_2  200 // print hours
  //#define SERVICE_NAME_3      "Service 3"
  //#define SERVICE_INTERVAL_3    1 // print hours
#endif

2. Compile & flash into board

3. Check in 12864 emulation mode

ellensp, 2020-07-29:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

trengtor, 2020-07-29:
> Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

Ok, but in some days

trengtor, 2020-07-29:
> Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

Some problem with compilation. Config files  attached.

[image]

[Marlin-bugfix-2.0.x configs.zip](https://github.com/MarlinFirmware/Marlin/files/4996720/Marlin-bugfix-2.0.x.configs.zip)

closed 2020-09-03 by github-actions[bot]