MarlinFirmware/Marlin #18848: [BUG] Laser control mixing up percentage and pwm

ryanaukes, 2020-07-29:
I think I encountered a bug in marlin 2.0.6.
I tried using the laser control feature. All the controls work, but the output does not match the set value.
It seems like it mixes up percentages and pwm values.
I am using a 5v output and set the laser control to use pwm, so 0 should be 0V, 255 should be 5v.
What I am getting is the following:
Laser power 0 -> output 0V
Laser power 50 -> output 2.5V
Laser power 100 -> output 5V
Laser power 101 ->output 0.05V
Laser power 200 -> output 5V
Laser power 255 -> output 2,75V

I tried to attach the configuration_adv.h file, but Github won't let me upload that file. How can I share my settings? I am not really used to using Github.

ellensp, 2020-07-29:
Config files need to be in a zip archive. then you just drop the zip in the boy of a reply.
Output is PWM, you need a scope, 
"Most current DMM's have a display update of 3 readings per sec."  it will miss most of the 5v pulses

ryanaukes, 2020-07-29:
Thanks for your reply. I know a multimeter is not optimal for measuring pwm duty cycle, but I have no osciloscope. 
However, i have been measuring it this way before and it always worked (it took the weighted average voltage).
When I connect the laser, it barely turns on when the power is set to anything below 255. 
That's why I measured the voltages and discovered this problem.   I will try to upload my config files later today!

ellensp, 2020-07-29:
Still need config files.

boelle, 2020-07-29:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

ryanaukes, 2020-07-29:
I updated my post, this time using the right format and included the config files. Sorry, I am not used to using Github. I hope everything is right now.

ellensp, 2020-07-29:
Well first thing I notice is you have Dual X and Dual Y stepper motors and 1 extruder...  so you seem have clearly modified the pins file to add in an extra stepper driver, and perhaps the laser pin...
Please provide your modified pins.h file also.

ryanaukes, 2020-07-29:
First of all, thanks for checking! That's correct. I am using a CNC machine with two steppers and two endstops on both the x and y axis. Here is the pins file. I am using the servo pins to control relays, so I am using pin 44 as the laser pwm pin. 
[pins_RAMPS.zip](https://github.com/MarlinFirmware/Marlin/files/4996704/pins_RAMPS.zip)

ellensp, 2020-07-30:
you can set extruders to 0 in marlin...

ellensp, 2020-07-30:
It does look like something is wrong

I just have  LED on pin 44.

If I use pin 44 just as a PWM I see the following in M43 and the led is quite bright.
 
PIN:  44   Port: L5        <unused/unknown>PWM:   240                           TIMER5C   PWM:   240    WGM: 1    COM5C: 0    CS: 3    TCCR5A: 9    TCCR5B: 3    TIMSK5: 0  

But if enable laser and use M3 S240  the led is dull and m43 give the following.

PIN:  44   Port: L5        SPINDLE_LASER_PWM_PIN                  PWM:   240    TIMER5C   PWM:   240    WGM: 14    COM5C: 0    CS: 1    TCCR5A: 10    TCCR5B: 25    TIMSK5: 0  

This is something to do with setting SPINDLE_LASER_FREQUENCY
If I turn this off  I get expected results.  

So that might be a short term fix while I look further.

ellensp, 2020-07-30:
With SPINDLE_LASER_FREQUENCY commented out.
M3 S1, results as expected.
[image]
M3 S128, results as expected.
[image]
M3 S254, results as expected.
[image]

With SPINDLE_LASER_FREQUENCY enabled and set to 495hz  (a freq Its already show mega can do) 

M3 S10, (s1 was to glitched to capture) results are abnormal.
[image]
M3 S128, results are abnormal.
[image]
M3 S254, results are abnormal.
[image]

ryanaukes, 2020-07-30:
Thanks! For the time being I will disable SPINDLE_LASER_FREQUENCY. Where can I find updates on when this bug is fixed? 
Thanks again!

ellensp, 2020-07-31:
I have it working on test machine..  just checking it doesn't break anything else!
Updates will be posted here

ellensp, 2020-07-31:
The above PR fixes it.

ellensp, 2020-08-03:
This has been merged into bugfix..  now it needs retested.

ryanaukes, 2020-08-04:
Thanks!. I added the change of code to my firmware and now the M3 command works as expected. I have one question however. I am using inline laser control, but now the laser is turned of when I do a G1 movement. Is this normal? The inline controll works fine. I attached a gcode file. On every M3 S60 command, the laser turns on, but it turns off directly after that on the next G1 movement.
[smiley outline.zip](https://github.com/MarlinFirmware/Marlin/files/5021556/smiley.outline.zip)

ellensp, 2020-08-05:
That doesn't seem correct. 

The way I read things when M3 is used Laser/spindle should stay on till M5 is given.

@jediminer543 and @shitcreek  can you guys confirm or deny?  (need someone who actually uses a laser!)

ryanaukes, 2020-08-07:
Thanks! I hope the other people you tagged can verify it and confirm it or tell me what I am doing wrong

ellensp, 2020-08-09:
Since you opened this as a new issue, closing this one as finished.

closed 2020-08-09 by ellensp

shitcreek, 2020-10-28:
Sorry, just saw this now.

With `LASER_POWER_INLINE` it allows you to use the `I` parameter with `M3`, thus `M3 Sxxx I` don't immediately turn on the laser until there is a move command such as G1.
With `LASER_MOVE_POWER`, `M3` is not needed as the laser can be triggered with the `S` parameter eg `G1 Xxx Yxx S255`
With `LASER_MOVE_G0_OFF` and `LASER_MOVE_G28_OFF`, `M5` is not needed to turn off the laser.
`G1` shouldn't be turning off the laser unless it's `G1 ... S0` and you're using `LASER_MOVE_POWER`
Also it should be noted `S0` with `M3` or say `G1` shouldn't turn off the laser if you're using `CUTTER_POWER_RELATIVE`

I hope it clears it up.