MarlinFirmware/Marlin #18849: [BUG] (short description)

bzzmartt02, 2020-07-29:
### Bug Description

Hi! I've just updated my old Anet A8 to the newest firmware (2.0.6) and I'm having troubles with the power loss recovery. The problem is after turning my printer off (on purpose, so I can check if it works) and then turn it on again after 20 sec or so, the power loss recovery start as it should - homes y and x, waits till it heats up again. Then goes back to the print but it also raises the Z axis by about a cm. On the screen it shows the height it should be printing at, however, in reality it is actually that same height + another 10mm.

### My Configurations

I have included zip of my configs. I've commented on every setting I have changed with "//MAG" so if you look for that you can see every setting I have touched or were planning on changing.

### Steps to Reproduce

1. Start a print as normal
2. Wait for the printer to finish a few layers
3. Turn the printer off by disconnecting it from the mains (to simulate power outage).
4. Turn the printer on after 10 sec
5. Resume the print and watch it fail

**Expected behavior:** I expect it to continue on the same "Z" height

**Actual behavior:** Continues on the same spot but 10mm higher.

trustfm, 2020-07-30:
To me 2.0.6 does not manage to raise the gantry when power loss occurs. Instead this worked perfectly on 2.0.5.3

bzzmartt02, 2020-07-30:
That's quite interesting. I uploaded a clean version of the the firware twice and still have that problem. I haven't checked whether I would still have it if I disable the manual mesh bed leveling, as I really don't want to disable that feature.

bzzmartt02, 2020-07-31:
Just a quick update. This only happens when manual mesh bed leveling is enabled. Can someone help?

bzzmartt02, 2020-08-08:
Is the post getting neglected because I have missed some information or what?

boelle, 2020-08-08:
no, its prob because only you and @trustfm discovered this one

sjasonsmith, 2020-08-08:
Posts get neglected because there are thousands of users willing to report bugs, but only a small number of people that actually fix them.

In my example, I'm busy during the week. I _might_ have time to look at things on the weekend, but even if I do have time I already have other issues I'm working on. I might spend 20 hours on it one weekend, then not have time for a month. That's just the way volunteer labor goes.

I will also always gravitate toward issues that will be easy for me to reproduce on my hardware. I'm willing to rearrange my hardware to work on things, but that is an extra time commitment that I don't always feel like doing (again, volunteer labor!).

In general, if you are polite and helpful somebody will eventually look at the issue, it can just take some time to find the right person with overlapping interest, experience, and availability.

sjasonsmith, 2020-08-08:
Power-loss recovery is honestly on the fringes of support as well. There are a few features that are wildly popular in the community, but not widely used by the people actively fixing bugs. People will always tend to fix the features that they use. Examples of these are power-loss recovery and sensorless homing.

Of course if anybody really is passionate about power-loss recovery, they are more than welcome to jump in a debug/fix issues themselves! Users are more than welcome to post pull-requests to fix things.

bzzmartt02, 2020-08-09:
Alright! Thanks for your response! Sorry if I sounded offensive, that really wasn't my intention. I was genuinely curious whether I submitted it correctly. I have tried edditing a few lines in the power loss recovery code, but due to my limitted knowledge I couldn't it get to work properly.

bzzmartt02, 2020-09-10:
I'm adding this comment as the github bot marked this issue as "stale-closing-soon", however, I still haven't found a solution to the problem.

closed 2020-11-02 by github-actions[bot]