MarlinFirmware/Marlin #18850: [FR] (A single step motor shared  for 2 to 8 extruders)

tphan26363, 2020-07-29:
### Description

-  Using 1 stepper to  for all the extruder ( from 1 to 8 extruders) in switching tool head
- Only using movement to switch and dock extruders with the aid of magnets
-Allow to define Stepper "Direction" and "E step" depend on the extruder used
-Extruders shall have independent  heater and temperature sensor defined
-Extruders shall have different nozzle size or the same
-Extruders shall have different Nozzle offset
-Use T0, T1 ... Tn to switch extruders not stepper
-Separate extruder from The Stepper.  The stepper is assumed to be a virtual class and can be assign to multiple steppers or a single stepper as in :
+ E0 ->E8   use  Stepper 0 to stepper 8
+E0--> E8  use Stepper 0( this is possible because the tool head maybe only used one at a time)
+
Also may be filament size can be assign independently

### Feature Workflow

GMagician, 2020-07-29:
This seems an extension of `SWITCHING_EXTRUDER` with no servo.
I used it a long time ago with Dondolo (2 extruders, 2 thermistors, 2 heaters, 1 stepper), but it requires a servo (or more servos when extruders are > 2), so some coding is required

tphan26363, 2020-07-29:
There is no need for servo.  It all switching by a move and magnet only
https://www.facebook.com/100033475262422/videos/302337124225461/

GMagician, 2020-07-29:
@tphan26363 

> This seems an extension of SWITCHING_EXTRUDER **with no servo**.

tphan26363, 2020-07-29:
It is not.  SWITCHING_EXTRUDER required separated E_STEPPER. I can make it into a single stepper but direction and e step/mm does not work

GMagician, 2020-07-29:
ok not understand anything then

tphan26363, 2020-07-29:
In Conditional_LCD
#elif ENABLED(SWITCHING_TOOLHEAD)
  #define E_STEPPERS     EXTRUDERS   
  #define E_MANUAL       EXTRUDERS

tphan26363, 2020-08-02:
Adding to the list.  The xyz tool offset does not seem to be handle correctly with Switching tool head

tphan26363, 2020-08-03:
Here is a working concept that I made.
https://www.facebook.com/100033475262422/videos/pcb.628432048103938/311553433303830/?type=3&ifg=1&__tn__=HH-R&eid=ARC-IYyzTZ2nurpU8tLBDmeqr2q-jceFKKIp9QG9Bt5qkoHl55zSW0vQEVdf5lHi7CKI6LqK7WZQuBxR
The model can extend the number of hotend without needed much in modification. I have change the 2.0.5.3 code to adapt to my model.  But it is better for the Github to support these changes

DerAndere1, 2020-12-22:
I coded a solution for switching tools using a dedicated stepper with the T0 / T1 comands . The commit is here:

https://github.com/DerAndere1/Marlin/commit/fd6efc3a8b5813c629d637ca84422820caf2b94e

tphan26363, 2020-12-25:
Thank you very much,  I guess now this feature may be able to support by GitHub.  I have made some code that works with 3 or more tools.  But I did some terrible thing to the code( which I don't understand all of them so I just removed and change a bunch of code).  This is can help me get to the right direction