MarlinFirmware/Marlin #18851: [BUG] (CompilingIssueFTDI_Display)

TB1405, 2020-07-29:
Hi 
i tried with the newest Marlin branch 2.0.6 to get a ftdi screen (FT800CB) work with a SKR1.4 Turbo and it works!
But I had to change a few things to make it compile successfully. 

- **First one** This also happend with my other display a VM800B43A  and fysetc S6. 
following error message appears. 
`Marlin\src\lcd\extui\lib\ftdi_eve_touch_ui\screens\bed_mesh_screen.cpp:131:11: error: 'VERTEX_FORMAT' was not declared in this scope; did you mean 'FTDI_FT810_DL::VERTEX_FORMAT'?  `

//.cmd(VERTEX_FORMAT(0)) I commented that out and then I could continue compiling  

- **Second one** is that following defintions missing in the LPC1768 hal.h 

@ellensp has proposed the following solution at the beginning of the year and with this one it works 
https://github.com/MarlinFirmware/Marlin/issues/16534#issuecomment-573382828

Add the following to src/HAL/HAL_LPC1768/HAL.h (I added at line 50, after includes)
`
#define strcat_P(dest, src) strcat((dest), (src))
#define pgm_read_ptr_far pgm_read_ptr
#define pgm_read_ptr_near pgm_read_ptr
#define pgm_read_word_far pgm_read_word
`

- **third one** is we have to use SPI_MODE3 instead off SPI_MODE0 

following error message appears 
`Marlin\src\lcd\extui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\basic\spi.cpp:38:64: error: 'SPI_MODE0' was not declared in this scope; did you mean 'SPI_MODE3'?`
i changed it to SPI_MODE3 and it compiles now succesfully

[ConfigForFTDIandSKR1.4.zip](https://github.com/MarlinFirmware/Marlin/files/4995674/ConfigForFTDIandSKR1.4.zip)

RudolphRiedel, 2020-07-29:
Can't really do anything about the second and third issue as I am using a Fysetc Cheetah to test my builds.

However the first issue, after enabling MESH_BED_LEVELING in Configuration.h I am seeing this as well.

Since I am currently working on adding new EVE displays anyways, I'll have a look.
But as I know what VERTEX_FORMAT does in FT81x and up I can only hope that this is easily fixable.
What puzzles me though is that there is only one single VERTEX_FORMAT command in the entire source, at first glance it looks like it can't be doing anything anways.

TB1405, 2020-07-29:
Okay. the problems 2 and 3 are already solved with my proposed solution. but somebody would have to make the whole thing into a PR. Unfortunately, I didn't manage to do that.

the strange thing is i don't use mesh bedleveling i use bilinear leveling also in the attached config files. 

Are you familiar with ftdi display ? 
Because I have the problem that if I move the cable a little bit the display fails. 
I've also tried twisting the cables but I think the SPI bus is just very sensitive to interference.

RudolphRiedel, 2020-07-29:
>the strange thing is i don't use mesh bedleveling i use bilinear leveling also in the attached config files.

I just looked for the HAS_MESH define and what is triggering setting it, the screen should get activated on any autobed leveling.

>Are you familiar with ftdi display?

A little. :-)
https://github.com/RudolphRiedel/FT800-FT813
https://github.com/RudolphRiedel/EVE_display-adapter

Well, I do know these a lot better than Marlin.

>Because I have the problem that if I move the cable a little bit the display fails.
>I've also tried twisting the cables but I think the SPI bus is just very sensitive to interference.

Can you measure how fast the SPI clock actually is with your configuration?
And how long is your cable?

And now going back to the top, what exactly is a "SKR1.4 Turbo"?
From that it shares a repository with the normal SKR 1.4 I would guess the "Turbo" is that is has the LPC1769 instead of the LPC1768? That is quite a stretch. :-)

> third one is we have to use SPI_MODE3 instead off SPI_MODE0
>
> following error message appears
> Marlin\src\lcd\extui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\basic\spi.cpp:38:64: error: 'SPI_MODE0' was not [...]

Should have read this more carefully.
This actually looks like a bug in HAL/LPC1768/include/SPI.h

Not only does this hardly make any sense:

---
#define MSBFIRST 1
#define SPI_MODE3 0

class SPISettings {
---

What this means is that you can only use one SPI-mode, mode 0.
But it is called mode 3.

So changing this in ftdi_eve_ib/basic/spi.cpp is not correct.
Changing it in spi.cpp to "SPI_MODE3" while actually mode 0 is needed would break it for every other architecture that
has a correct set of defines for the SPI like STM32F1.

TB1405, 2020-07-29:
> A little. :-)
> https://github.com/RudolphRiedel/FT800-FT813
> https://github.com/RudolphRiedel/EVE_display-adapter
> 
> Well, I do know these a lot better than Marlin.

looks more like you are a ftdi expert! :)
 did you also use a 4.3inch screen ? 
on both screens the marlin boot logo is only displayed half 

> Can you measure how fast the SPI clock actually is with your configuration?
> And how long is your cable?

I think i need a good oscilloscope for measure the frequency i only have a DYI osci from aliexpress with a STM32 Chip i think this want work.  
i tried different calbe length and cables from 5cm to 25cm i also tried it with a flat ribbon cable but it was with all cables very instable. This problem i also had with the Fysetc S6 and a VM800B43A on my second printer. I use here a CAT5 ethernet cable and this actual works. 
but i think the FT800CB from Hayou is more is even more sensitive to interference signals. 

> And now going back to the top, what exactly is a "SKR1.4 Turbo"?
> From that it shares a repository with the normal SKR 1.4 I would guess the "Turbo" is that is has the LPC1769 instead of the LPC1768? That is quite a stretch. :-)

Yes correct it is a SKR1.4 with LPC1769. BTT it calls Turbo because the clock frequency is higher. 

> Should have read this more carefully.
> This actually looks like a bug in HAL/LPC1768/include/SPI.h
>
> Not only does this hardly make any sense:
>
> #define MSBFIRST 1
> #define SPI_MODE3 0
>
> ## class [...]

Okay I have to admit, I don't know that much about it. I just tried it with SPI_Mode_3 and was glad that it really works and the display is display something

RudolphRiedel, 2020-07-30:
>did you also use a 4.3inch screen ?
Yes, but not FT80x.

>on both screens the marlin boot logo is only displayed half
Okay, that is odd as well.

>but i think the FT800CB from Hayou is more is even more sensitive to interference signals.
It should not as it has integrated logic to buffer the signals.
I have connected a FYSETC Cheetah with STM32F103 to one of my adapter boards with the original CR-10 cable that is about ~20cm long.
And the display is doing just fine with hardware SPI at 8MHz.
How do you supply the display? EXP2 on the SKR 1.4 has no 5V pin.

I'll have a look, I should have at least one display from Haoyu in my collection.
I stopped using these years ago though when I switched over to cap-touch displays.

>I just tried it with SPI_Mode_3 and was glad that it really works
It works for the LPC176x because that is the only define in the HAL and incorrectly points to mode 0.
I'll look into adding a workaround to Extui but I consider Marlin/Marlin/src/HAL/LPC1768/include/SPI.h to be at fault.
@thinkyhead - I checked the history for that file but that did not help much, is there someone this could be assigned to?
I do not like to poke around in HAL files for boards I can not test with controllers that I never used.

TB1405, 2020-07-30:
> Okay, that is odd as well.
after work i could post a foto maybe it helps to find the problem

> It should not as it has integrated logic to buffer the signals.
> I have connected a FYSETC Cheetah with STM32F103 to one of my adapter boards with the original CR-10 cable that is about ~20cm long. [...]

i will try it with a cat5 ethernet cable with twisted cables. 
I use the 5V Pin from the EXP2.

RudolphRiedel, 2020-07-30:
You could also try to reduce the clock for the SPI to 4MHz or less or use software-spi.
Where have you configured the pin-connections for  Extui anyways?
I am looking at pins_BTT_SKR_V1_4.h in which I would expect the definitions but it is not there.

The correct way to do this would be to add an #if ENABLED(TOUCH_UI_FTDI_EVE) / #endif block like in pins_FYSETC_CHEETAH.h
To this you could add this line as well:
#define CLCD_HW_SPI_SPEED 8000000 >> SPI_SPEED

TB1405, 2020-07-30:
> You could also try to reduce the clock for the SPI to 4MHz or less or use software-spi.
> Where have you configured the pin-connections for Extui anyways?
> I am looking at pins_BTT_SKR_V1_4.h in which I would expect the definitions but it is not there.

I tried it also with the SW_SPI but same problem. 
How could i reduce the SPI speed ? 

the function #define OTHER_PIN_LAYOUT in conf.adv does not work. Thats why i added the #define SKR_Turbo_TFT_PINMAP  i copied the definitions from S6_TFT_PINMAP  and changed it for the SKR

RudolphRiedel, 2020-07-30:
As I wrote, the pin-configuration is supposed to be in the pins_** files now.
The "// Mappings for boards with a standard RepRapDiscount Display connector" block in Configuration_adv.h is merely a relic from the original implementation.

>I tried it also with the SW_SPI but same problem.

Now that is really strange since software SPI should be well below 4MHz.

>How could i reduce the SPI speed ?

#define CLCD_HW_SPI_SPEED 4000000 >> SPI_SPEED

But I am not sure if this even works for the LPC176x.

TB1405, 2020-07-30:
> #define CLCD_HW_SPI_SPEED 4000000 >> SPI_SPEED

with this the display runs much better but with the sd card modul also connected it is bad again. 
but now i have a cat cable i will check it with this.

TB1405, 2020-07-30:
alright iam finshed with crimping the CAT5 cable and now it works also with 8MHZ and SD Card Modul. 
But i killed with a short circuit my Display driver board :( (the backlight inverter)
i need a RT9293B but is not available in germany 
[image]
i tried to use 5V as backlight voltage but didnt work. 
regarding this datasheet the backlight voltage is 20V
http://www.haoyuelectronics.com/Attachment/HY43-LCD/LCD%20DataSheet.pdf
i think i now have to wait :(

RudolphRiedel, 2020-07-30:
Also 5V ist für die soweit schon richtig.
Und wie der gestorben ist wüsste ich schon gerne um das zu vermeiden. :-)
Leider finde ich den Chip auch nicht bei den üblichen Verdächtigen.

TB1405, 2020-07-30:
Laut Datasheet von dem LCD  benötigen sie 19,8 Volt, ich schätze mal die sind in Reihe geschaltet. Hab mir natürlich auch schon überlegt von den 24V abgreifen und ein Widerstand davor, aber wenn das schief geht ist alles hin... 
[image]

Das CAT5 Kabel hat ja eine Abschirmung aus Aluminium. Diese muss an den Chip gekommen sein dann hats kurz geraucht und gerochen und vorbei 🥇 
Das Display und Druckerboard haben aber zum Glück nichts abbekommen. 

Ja ist meistens so das man das Zeug hier nicht bekommt, oder so teuer das es sich nicht lohnt.

RudolphRiedel, 2020-07-30:
Ja, sowas ist hochgradig lästig, dazu noch kurz vor Wochenende.

And to catch on with a potential audience.. :-)
I still wonder what is up with the SPI there.
I gave up using the modules from Haoyu but only because of the panels while I kind of like their driver board.

TB1405, 2020-07-30:
Glaub es nicht läuft tatsächlich mit Vorwiderstand und 24V 
[image]
[image]
Das blaue hat den Kurzschluss verursacht naja also eigentlich meine unvorsicht. :D

RudolphRiedel, 2020-07-30:
Hehe, wenn man verzweifelt ist. :-)
So wie das aussieht könnte der Widerstand aber noch ein paar Ohm größer sein.

sjasonsmith, 2020-08-09:
What is the status of this? I see pictures of working LCD screens above. Has this already been resolved, or are there action items pending?

RudolphRiedel, 2020-08-09:
No, this is not fixed.

I have not looked further into fixing issue 1 yet.

Issue 2 is with src/HAL/HAL_LPC1768/HAL.h

Issue 3 is with src/HAL/LPC1768/include/SPI.h:
`#include <stdint.h>

#define MSBFIRST 1
#define SPI_MODE3 0

class SPISettings {`

One possible fix may be:
#define SPI_MODE0 0

But I am not sure, it may break something else and I can not test it.

RudolphRiedel, 2020-08-09:
Okay, I tried to fix the first issue with bed_mesh_screen.cpp.
@marciot I am afraid this is one for you as you wrote it

` cmd.cmd(SAVE_CONTEXT())
     .cmd(VERTEX_FORMAT(0))
     .cmd(TAG_MASK(false))
     .cmd(SAVE_CONTEXT());

  for (uint8_t y = 0; y < rows; y++) {
    for (uint8_t x = 0; x < cols; x++) {
      if (ISVAL(x,y)) {
       const bool hasLeftSegment  = x < cols - 1 && ISVAL(x+1,y);
       const bool hasRightSegment = y < rows - 1 && ISVAL(x,y+1);
       if (hasLeftSegment || hasRightSegment) {
         cmd.cmd(BEGIN(LINE_STRIP));
         if (hasLeftSegment)  cmd.cmd(VERTEX2F(TRANSFORM(x + 1, y    , HEIGHT(x + 1, y    ))));
         cmd.cmd(                     VERTEX2F(TRANSFORM(x    , y    , HEIGHT(x    , y    ))));
         if (hasRightSegment) cmd.cmd(VERTEX2F(TRANSFORM(x    , y + 1, HEIGHT(x    , y + 1))));
       }
      }
    }`

We need to remove the line " .cmd(VERTEX_FORMAT(0))" as stated somewhere above as this does not exist for FT80x.
What it does is to change the precision for the coordinates of VERTEX2F from the default 1/16 pixels to 1/1 pixels meaning that when removing the line the X and Y coordinates need to be multiplied by 16.
So far so good, VERTEX_FORMAT is not used anywhere else and so all the coordinates get multiplied by 16.

But what do all these macros really do?

if (hasLeftSegment)  cmd.cmd(VERTEX2F(TRANSFORM(x + 1, y    , HEIGHT(x + 1, y    ))));

  #define TRANSFORM_5(X,Y,Z)  (X), (Y)                                                                   // No transform
  #define TRANSFORM_4(X,Y,Z)  TRANSFORM_5((X)/(Z),(Y)/-(Z), 0)                                           // Perspective
  #define TRANSFORM_3(X,Y,Z)  TRANSFORM_4((X), (Z), (Y))                                                 // Swap Z and Y
  #define TRANSFORM_2(X,Y,Z)  TRANSFORM_3((X), (Y) + 2.5, (Z) - 1)                                       // Translate
  #define TRANSFORM(X,Y,Z)    TRANSFORM_2(float(X)/(cols-1) - 0.5, float(Y)/(rows-1)  - 0.5, (Z))        // Normalize
  #define VALUE(X,Y)         (data ? data[X][Y] : 0)
  #define ISVAL(X,Y)         (data ? !isnan(VALUE(X,Y)) : true)
  #define HEIGHT(X,Y)        (ISVAL(X,Y) ? (VALUE(X,Y) - val_min) * scale_z : 0)

Maybe it is too hot here but I tried a few things, did not work.

marciot, 2020-08-09:
@RudolphRiedel: I can fix it. I haven’t been testing on the FTDI800 which is why this snuck up.

The reason to use VERTEX_FORMAT is that on the larger resolution displays there isn’t enough resolution in VERTEX2F to span the entire display without overflow. So to support both the FT800 and FT810 it is necessary to make the use of VERTEX_FORMAT conditional on display size.

The macros form various stages in a 3D transformation pipeline. I’ll make the use of VERTEX_FORMAT conditional and add one more stage the pipeline to scale the values.

RudolphRiedel, 2020-08-09:
Excellent, thank you. :-)

VERTEX2F however has a range of -1024 to 1023, VERTEX2II is the one that is limited to 0...511.

There is at least one EVE module with 10.1" 1280 x 800 announced but I am pretty sure we won't see a feature request to add support for this one to Marlin. :-)

I have a protoype EVE module on my desk with 7" 1024 x 600 that I could try out but the backlight current alone could be bad news for the common printer board. :-)

marciot, 2020-08-10:
@RudolphRiedel, @TB1405: I pushed a fix for the third issue (the VERTEX_FORMAT incompatibility). 

Let me know what you guys find with the other two issues. I actually have a SKR board I've been meaning to try out, but I won't be able to get to it for a while.

RudolphRiedel, 2020-08-10:
I can confirm that it works, I manually changed the source and the lines and dots on the "Level Bed" screen are back to normal on my 800x480 display with BT815.
I completely missed the "#undef  TRANSFORM_5" and redefinition of TRANSFORM_5 before.

marciot, 2020-08-10:
> I can confirm that it works, I manually changed the source and the lines and dots on the "Level Bed" screen are back to normal on my 800x480 display with BT815.
> I completely missed the "#undef TRANSFORM_5" and redefinition of TRANSFORM_5 before.

Yeah. its easy to miss. I tried expanding the comments a bit. I actually do that because it auto-scales the grid to fit the display area. Since we're dealing with multiple display resolutions, it gets a bit complicated. I'm also trying to precompute as much as I can during compile time, for the benefit of 8-bit micros.

It would be nice if C++ had better compile-time programming capabilities. There are constexpr functions, but they are not yet powerful enough to layout a GUI at compile time the way I wanted to do it.

closed 2020-09-15 by github-actions[bot]