MarlinFirmware/Marlin #18852: Enabling IMPROVE_HOMING_RELIABILITY and running G28 causes X Y and Z crashing

Crunch69, 2020-07-29:
### Bug Description

Enabling IMPROVE_HOMING_RELIABILITY in Configuration_adv and running Homing G28 causes X Y and Z crashing

### My Configurations
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/4996392/Configuration_adv.zip)

### Steps to Reproduce

1. enable IMPROVE_HOMING_RELIABILITY 
2. Run G28

**Expected behavior:** Should Home extruder

**Actual behavior:** Y axis moves forward beyond it endstop X axis runs beyond its endstop and Z axis lowers off build plate

AFprinter, 2020-09-13:
This is happening to me as well.  After about 4-5 consecutive prints homing is being inverted and the axis just crash into their max extremes and then start bed leveling in crazy spots until marlin freezes and resets. 

BTT SKR1.4 Turbo w/ TMC 2209,

boelle, 2020-09-14:
@AFprinter not sure if the smart people can make use of it but try and attach your configs

also it would not hurt to use  [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) if not allready doing that.

closed 2020-10-22 by github-actions[bot]