MarlinFirmware/Marlin #18853: [BUG] V 2.0.6 Gradient - Freeze

Avatarsia, 2020-07-29:
Like # 17397

Compiled without any errors.

Start a normal Print - Fine
Change Colors - Fine
Set a gradient in the Display will keep the Printer Freeze Totally.
Not even the restart Button works anymore, just a hard switch off and on works.
Printer A20M

boelle, 2020-07-29:
precise the same as #17397 ?

Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

Avatarsia, 2020-07-29:
[config.zip](https://github.com/MarlinFirmware/Marlin/files/4996639/config.zip)
No not exactly the same, only the error seems to be the same.
As i said, if i try to set a mix (Gradient), the printer freeze totally.

thinkyhead, 2020-07-29:
I have an A20M here that I can flash and test for debugging.

thinkyhead, 2020-07-29:
Generally speaking your configurations look okay, but did you happen to notice that your A20M configurations were built starting from A10M configurations? The default A20M configurations have all the stepper motors going the opposite direction, and a few other settings differ also.

Anyway, I have compiled your configurations (with the correct stepper directions for this machine, plus the BLTouch probe enabled) and I'm about to try printing some objects and playing with the gradient functions to see how it behaves.

Avatarsia, 2020-07-29:
yes first i choosed the wrong one and corrected some stuff.
the bltouch stuff is from the example, i just use the manual bed leveling story.

in general i can print objects.
try the following:
mix menu - gradient - start 0 - end 15mm 
during the heat up time

thinkyhead, 2020-07-29:
I'll post some videos of my testing as I go along. Here I am at the start of my first test print….

https://youtu.be/PFnhSBBToeA

Avatarsia, 2020-07-30:
i cut the part where i start the print from sd card and heating up
https://photos.app.goo.gl/KPqFz9koSnSngQWC8

thinkyhead, 2020-07-30:
Sure enough, the machine froze up at the point where the gradient would have started to have some effect, at a point a bit higher than 22mm. At this point it's a complete mystery, but it should be easy to reproduce and hopefully shouldn't be too hard to figure out where and why it's glitching out.

Avatarsia, 2020-09-09:
please not when not fixed

closed 2020-10-17 by github-actions[bot]