MarlinFirmware/Marlin #18855: #error "ESP3D_WIFISUPPORT or WIFISUPPORT requires an ESP32 controller."

Ffatstuo, 2020-07-29:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4996866/Marlin.zip)
Bug Description
ESP3D_WIFISUPPORT Does not work

Error message produced
Processing LPC1769 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.3.zip; board: nxp_lpc1769; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
Getting g++ path from cache
arm-none-eabi-g++.exe -D__MARLIN_FIRMWARE__ -DU8G_HAL_LINKS -w -dM -E -x c++ buildroot/share/PlatformIO/scripts/common-dependencies.h
Adding lib_deps for HAS_TRINAMIC_CONFIG... 
Adding src_filter for HAS_TRINAMIC_CONFIG... 
Adding lib_deps for HAS_GRAPHICAL_LCD...     
Adding src_filter for HAS_GRAPHICAL_LCD...   
Adding lib_deps for HAS_LCD_MENU... 
Adding src_filter for HAS_LCD_MENU...
Adding lib_deps for AUTO_BED_LEVELING_(3POINT|(BI)?LINEAR)...
Adding src_filter for AUTO_BED_LEVELING_(3POINT|(BI)?LINEAR)... 
Adding lib_deps for (ESP3D_)?WIFISUPPORT...
Ignoring libs for (ESP3D_)?WIFISUPPORT...
Ignoring libs: ['ESPAsyncTCP', 'lvgl', 'SailfishLCD', 'Arduino-L6470', 'Adafruit MAX31865 library', 'LiquidTWI2', 'TMC26XStepper', 'SlowSoftI2CMaster']
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1769.html
PLATFORM: NXP Arduino LPC176x 0.1.3 > NXP LPC1769
HARDWARE: LPC1769 120MHz, 31.97KB RAM, 464KB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink)
PACKAGES:
 - framework-arduino-lpc176x 0.2.2
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ strict
Found 6 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Servo> 1.0.0
|-- <LiquidCrystal> 1.0.0
|-- <Adafruit NeoPixel> 1.5.0
|-- <TMCStepper> 0.7.1
|-- <U8glib-HAL> 0.4.1
Building in release mode

Unable to find destination disk (Autodetect Error)
Please select it in platformio.ini using the upload_port keyword (https://docs.platformio.org/en/latest/projectconf/section_env_upload.html) or copy the firmware (.pio/build/LPC1769/firmware.bin) manually to the appropriate disk

Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\DebugMonitor.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\HAL.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\HAL_SPI.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\eeprom_flash.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\eeprom_sdcard.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\eeprom_wired.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\main.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\timers.cpp.o
In file included from Marlin\src\HAL\LPC1768\../../core/../inc/MarlinConfig.h:41,
                 from Marlin\src\HAL\LPC1768\../../core/serial.h:24,
                 from Marlin\src\HAL\LPC1768\DebugMonitor.cpp:25:
Marlin\src\HAL\LPC1768\../../core/../inc/SanityCheck.h:3051:4: error: #error "ESP3D_WIFISUPPORT or WIFISUPPORT requires an ESP32 controller."
 3051 |   #error "ESP3D_WIFISUPPORT or WIFISUPPORT requires an ESP32 controller."
      |    ^~~~~
In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:41,
                 from Marlin\src\HAL\LPC1768\HAL.cpp:24:
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:3051:4: error: #error "ESP3D_WIFISUPPORT or WIFISUPPORT requires an ESP32 controller."
 3051 |   #error "ESP3D_WIFISUPPORT or WIFISUPPORT requires an ESP32 controller."
      |    ^~~~~
In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:41,
                 from Marlin\src\HAL\LPC1768\HAL_SPI.cpp:51:
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:3051:4: error: #error "ESP3D_WIFISUPPORT or WIFISUPPORT requires an ESP32 controller."
 3051 |   #error "ESP3D_WIFISUPPORT or WIFISUPPORT requires an ESP32 controller."
      |    ^~~~~
In file included from Marlin\src\HAL\LPC1768\../../sd/../inc/MarlinConfig.h:41,
                 from Marlin\src\HAL\LPC1768\../../sd/cardreader.h:24,
                 from Marlin\src\HAL\LPC1768\main.cpp:38:
Marlin\src\HAL\LPC1768\../../sd/../inc/SanityCheck.h:3051:4: error: #error "ESP3D_WIFISUPPORT or WIFISUPPORT requires an ESP32 controller."
 3051 |   #error "ESP3D_WIFISUPPORT or WIFISUPPORT requires an ESP32 controller."
      |    ^~~~~
In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:41,
                 from Marlin\src\HAL\LPC1768\eeprom_sdcard.cpp:25:
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:3051:4: error: #error "ESP3D_WIFISUPPORT or WIFISUPPORT requires an ESP32 controller."
 3051 |   #error "ESP3D_WIFISUPPORT or WIFISUPPORT requires an ESP32 controller."
      |    ^~~~~
In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:41,
                 from Marlin\src\HAL\LPC1768\eeprom_flash.cpp:39:
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:3051:4: error: #error "ESP3D_WIFISUPPORT or WIFISUPPORT requires an ESP32 controller."
 3051 |   #error "ESP3D_WIFISUPPORT or WIFISUPPORT requires an ESP32 controller."
      |    ^~~~~
In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:41,
                 from Marlin\src\HAL\LPC1768\eeprom_wired.cpp:24:
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:3051:4: error: #error "ESP3D_WIFISUPPORT or WIFISUPPORT requires an ESP32 controller."
 3051 |   #error "ESP3D_WIFISUPPORT or WIFISUPPORT requires an ESP32 controller."
      |    ^~~~~
In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:41,
                 from Marlin\src\HAL\LPC1768\timers.cpp:31:
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:3051:4: error: #error "ESP3D_WIFISUPPORT or WIFISUPPORT requires an ESP32 controller."
 3051 |   #error "ESP3D_WIFISUPPORT or WIFISUPPORT requires an ESP32 controller."
      |    ^~~~~
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\DebugMonitor.cpp.o] Error 1
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\HAL.cpp.o] Error 1
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\HAL_SPI.cpp.o] Error 1
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\main.cpp.o] Error 1
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\eeprom_flash.cpp.o] Error 1
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\eeprom_sdcard.cpp.o] Error 1
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\eeprom_wired.cpp.o] Error 1
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\timers.cpp.o] Error 1
=============================================================================== [FAILED] Took 4.79 seconds ===============================================================================
Steps to Reproduce 

#define SERIAL_PORT 3

/**
 * Select a secondary serial port on the board to use for communication with the host.
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT_2 -1

Platform ini set : LPC1769
Board skr 1.4 turbo

thisiskeithb, 2020-07-29:
> Board skr 1.4 turbo

As the error message states, you need an ESP32-based motherboard to enable `ESP3D_WIFISUPPORT`/`WIFISUPPORT`. The SKR 1.4 Turbo is an LPC1769-based motherboard.

___
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-07-29 by thisiskeithb

thinkyhead, 2020-07-29:
Full instructions on installing an ESP32 wifi module on an SKR 1.4 (not turbo) are at https://youtu.be/NGgzw-XayEo

In this particular tutorial Chris Riley chooses to set `SERIAL_PORT` to 3 and `SERIAL_PORT_2` to -1.

jjtronics, 2020-07-31:
Hello, i have the same problem, in marlin 2.0.5.4 works very well

radek8, 2020-08-01:
> > Board skr 1.4 turbo
> 
> As the error message states, you need an ESP32-based motherboard to enable `ESP3D_WIFISUPPORT`/`WIFISUPPORT`. The SKR 1.4 Turbo is an LPC1769-based motherboard.
> 

So why Marlin version 14 days old worked with wifi card ESP32 on SKR 1.4 board and now there is a problem
I have the same problem compiling a new version of Marlin.

ellensp, 2020-08-01:
Because people where enabling Marlin ESP3D_WIFISUPPORT or WIFISUPPORT incorrectly. They are not required for a stand alone ESP3D/ESP01 wifi modules. You just need a serial port for those. 

The new build environment automatically fetches includes libraries based on what is enabled in config files. (before it used to only grab what was listed in platformio.ini) If you have either of these two enabled it downloads and tries to use the libraries. Since these two libraries only work on esp32 based controllers marlin unsurprisingly fails to compile.

radek8, 2020-08-01:
@Ffatstuo
After commenting on these lines, everything is fine

Marlin\src\inc\SanityCheck.h

#if EITHER(ESP3D_WIFISUPPORT, WIFISUPPORT) && DISABLED(ARDUINO_ARCH_ESP32)
#error "ESP3D_WIFISUPPORT or WIFISUPPORT requires an ESP32 controller."
#endif

radek8, 2020-08-01:
> Protože lidé, kteří povolili Marlin ESP3D_WIFISUPPORT nebo WIFISUPPORT nesprávně. Nejsou vyžadovány pro samostatné WiFi moduly ESP3D / ESP01. Pro ně stačí sériový port.
> 
> Nové prostředí sestavení automaticky vyvolá knihovny na základě toho, co je povoleno v konfiguračních souborech. (předtím to bylo jen chytit, co bylo uvedeno v platformio.ini) Pokud máte některou z těchto dvou povoleno, stáhne a pokusí se použít knihovny. Vzhledem k tomu, že tyto dvě knihovny fungují pouze na řadičích založených na esp32, marlin nepřekvapuje.

If I used WIFI only on the serial port, as you write,
  without #define ESP3D_WIFISUPPORT permission.
Marlin worked poorly. commands that I sent via wifi (for example G28) had a delay of sometimes 30 seconds.

radek8, 2020-08-01:
# 18792
end
# 18808
bad work