MarlinFirmware/Marlin #18860: [BUG] (short description)

jeffeb3, 2020-07-30:
### Bug Description

I am getting a compile error when building for Archim boards (1&2) in arduino with bugfix-2.0.x and with 2.0.6. The same code builds in platformio.

I have confirmed this with arduino 1.8.12 on my Linux desktop (ubuntu 18.04). But I am also trying to build this automatically. Here is the output from our builder:

https://github.com/jeffeb3/MarlinBuilder/runs/926010778?check_suite_focus=true#step:9:2068

It looks like an error in HAL.h, but I'm not sure why it only affects Archim (the rambo, ramps, skr boards are all fine). I am not sure why it is only a problem in Arduino. Hopefully it is an easy fix.

### My Configurations

[Archim_Broken.zip](https://github.com/MarlinFirmware/Marlin/files/4998621/Archim_Broken.zip)

### Steps to Reproduce

1. Configure for Archim
2. Install the board manager: https://raw.githubusercontent.com/ultimachine/ArduinoAddons/master/package_ultimachine_index.json
3. Install archim board
4. Choose Archim1 from the boards menu.
5. Compile

**Expected behavior:** [What you expect to happen]

The build succeeds

**Actual behavior:** [What actually happens]

These errors:

This link is a little weird. I hoped it would point to the exact issue. But it starts on line 2068:

https://github.com/jeffeb3/MarlinBuilder/runs/926010778?check_suite_focus=true#step:9:2068

#### Additional Information

cr20-123, 2020-07-30:
Is that a 32bit board?

ellensp, 2020-07-30:
Please update description

jeffeb3, 2020-07-30:
> Is that a 32bit board?
> 
> After some googling, it looks like those boards use an Atmel SAM3X8E which is a 32bit processor. Arduino IDE is only for 8bit boards. With a 32bit board, you need to use PlatformIO and VSCode or Atom.

Ok. I assumed arduino was the target because in 2.0.5.4, it builds in arduino and not in platformio.

Sorry for the missing description.

cr20-123, 2020-07-30:
I could be wrong, didn't realize there are libraries for SAM chips for Arduino.

thisiskeithb, 2020-07-30:
Some 32-bit boards/configs can still be built with Arduino IDE, but it's recommended to use VSCode + PlatformIO.

AFprinter, 2020-07-31:
> 
> 
> Some 32-bit boards/configs can still be built with Arduino IDE, but it's recommended to use VSCode + PlatformIO.

This is true but the Archim line of boards retained it's ability to be compiled in Arduino IDE I believe due to the unique pinout for the board. For the record I haven't been able to build in Arduino IDE for the Archim 2 since the end of May.

boelle, 2020-07-31:
@ellensp @thisiskeithb is this an issue with marlin or the IDE ? if it's the later can it called a bug with marlin?

AFprinter, 2020-07-31:
> @ellensp @thisiskeithb is this an issue with marlin or the IDE ? if it's the later can it called a bug with marlin?

This may be related to the issue I'm having in my thread https://github.com/MarlinFirmware/Marlin/issues/18816#issue-666658645.  Something may be broken with compiling Marlin with regards to this board.  I'm able to complete using VSCode but it's not functioning properly.  The manufacture recommended method of compiling is Arduino IDE and as of late May something in the coding changed and is no longer working with Arduino either.

jeffeb3, 2020-07-31:
I don't have an Archim 2, so I can't say anything about running, unfortunately.

I can say:

2.0.5.4 builds with arduino, but not platformio.
2.0.6 builds with platformio but not arduino. 

I haven't even looked closely at the error. I assume someone familiar would immediately recognize it. 

Fundamentally, unless Marlin is willing to just completely give up on arduino building, the boards that can be built in arduino should be tested with the arduino toolchain. Pull requests should be required to pass those builds, or else end users are going to keep finding the errors, reporting issues, and have to convince developers it isn't their fault.

jeffeb3, 2020-08-31:
Is Marlin on arduino officially dead? I know there are many troubles with arduino in windows. Troubles that are very hard to fix. It would be nice to know if we should be working to keep it alive, or pushing it out.

boelle, 2020-09-15:
vscode+platform.io is the tool

arduino ide has bugs that we cant fix

jeffeb3, 2020-09-15:
I understand the preference. I prefer it too. But is arduino definitely dead? Is that the project's preferred step forward? If so, is this written down somewhere?

There is a lot of documentation for how to build with arduino. We have a parallel doc for platformio. But people still pick up the arduino instructions first (it shouldn't be surprising, after years of marketting it to hobby and beginners). 

Until it is officially deprecated, issues like this should be fixed.

boelle, 2020-09-16:
> But is arduino definitely dead? Is that the project's preferred step forward? If so, is this written down somewhere?

Our maintainer @thinkyhead does not use it, nor do i personally due to bugs in arduino ide that we cant fix, arduino ide guys have to do that and they dont

So i would say yes to the first 2, but its not written down anywhere, but when people mention arduino ide we say to use vscode+platform.io

jeffeb3, 2020-09-16:
In the readme, the first suggestion is to use arduino. The first instructions are for arduino, and they point [here](https://marlinfw.org/docs/basics/install_arduino.html). I understand pointing people to platformio as a work around. I also understand that developers prefer it. I also understand that not every board can work with arduino. I also see the new CreateProcess error in windows as of 2.6.0, and it seems to me like the writing is on the wall. I see all of these things, and I don't think this issue is probably the right place to argue about it.

But, it isn't sustainable to direct users to arduino, have them get lots of problems (which many probably assume is there fault) and then come here or another forum and be told they need to use platformio.

Because I love Marlin, I hope there is a decision about the future of Marlin in arduino. If it's dead, then we should be clear about that. If support is still wanted, then we shouldn't be responding to all the arduino issues with "use platformio".

Until there is a clear deprecation of arduino, this is still an issue.

sjasonsmith, 2020-09-16:
@jeffeb3 I agree with you. I don't use Arduino either, but until @thinkyhead announces it is dead and changes the documentation, we shouldn't simply disregard issues like this.

In this case, it appears to be an actual code issue of some sort, not the normal Arduino problems related to path lengths, etc.

sjasonsmith, 2020-09-16:
@jeffeb3 when was the last time you tried this? I was able to reproduce that error with 2.6.0 on Windows, but not with 2.6.0.1 or bugfix-2.0.x.

jeffeb3, 2020-09-16:
Thanks. It does build with bugfix and 2.0.6.1 in our builder. I haven't had them on since 2.0.6.

https://github.com/V1EngineeringInc/MarlinBuilder/pull/23/checks?check_run_id=1124244240

closed 2020-09-16 by jeffeb3

sjasonsmith, 2020-09-16:
Great! Thanks for the update.