MarlinFirmware/Marlin #18864: [2.0.x] CR-10S No media/No SD-card detected.

JalteJKR, 2020-07-30:
I have a CR-10S that I recently updated to Marlin 2.0.5.3 (I used the stock CR-10S configuration). But now the printer won't register the SD-card and the "Init. sd card" option is gone, so I can't do it manually either.

Here are my configuration files:
[Configuration-Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/5002705/Configuration-Configuration_adv.zip)

[image]

ellensp, 2020-07-31:
There where many SD card issues with 5.0.5.3  please use bugfix or at lease latest release 2.0.6

boelle, 2020-07-31:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

JalteJKR, 2020-07-31:
Thanks for the answers, I will try and update to 2.0.6, and if that doesn't work then I will try the bugfix. I will report back once I've tried 👍

sjasonsmith, 2020-08-17:
@JalteJKR have you resolved this?

Jimmy8881, 2020-08-29:
@JalteJKR on your configuration_adv
try changing:
 //#define SDCARD_CONNECTION LCD
to   #define SDCARD_CONNECTION ONBOARD

closed 2020-09-23 by github-actions[bot]

JalteJKR, 2020-11-05:
Hello again, sorry that I never reported back, but I had a lot of issues updating marlin since I kept getting a "file name or extension too long" error. I have finally come around that and have successfully updated to marlin 2.0.x bugfix yesterday... But the problem is still there.

I will try doing what @Jimmy8881 suggested when I come home today, and I'm actually pretty confident that will fix the problem after looking it up in the code :)

[image]

_This time I WILL report back!_

JalteJKR, 2020-11-05:
I was right, It finally works!

Many many thanks to @Jimmy8881 for the solution 👍

djardi, 2020-12-31:
This is not a complaint; I am very happy with Marlin and grateful for people's work. I am adding a comment in the hopes of bumping this issue to a higher priority. This is a significant problem for anybody with a Creality CR-10S printer. When you install Marlin 2.0.x, the SD card reader is disabled. I experienced this problem first with the branch of Marlin that I downloaded from TH3D and then with Marlin 2.0.x downloaded from github. Until I found this solution, I thought that I was going to have to choose between using old firmware or losing access to the SD reader. The fix is incredibly easy (alter one line of code in the Configuration_adv.h file as described above). This completely fixes the problem.

Unless this creates another problem that I am not aware of, it could be quickly fixed by a github volunteer who can edit the configuration file examples (Creality --> CR-10S --> CrealityV1 --> Configuration_adv.h). This would be a great service for those of us who own this printer.

Thanks for considering this change to Marlin!