MarlinFirmware/Marlin #18867: [FR] BABYSTEP_MBL_Z_OFFSET for persistent babystepping and feedback to hosts

PrintEngineering, 2020-07-31:
I have an Artillery Sidewinder. My host is not Marlin Mode capable, therefore implementing babystepping with MESH_BED_LEVELING is like trying to hit the toilet bowl with the lights out. I can get a pretty good feel for the floorplan, but I might miss and hit the seat now and then. Worse yet, the changes are not persistent, as would be with ABL by defining BABYSTEP_ZPROBE_OFFSET.

For my purposes I had found BABYSTEP_DISPLAY_TOTAL which appears to display a step count, and would solve half my problem though the endeavor would require computation and prior query to axis steps/mm to work properly. Additionally, it does not display said values if a Z parameter is sent, only a naked M290 echos the value, so getting the value to confirm the command requires yet another command. That whole process is more than a bit cumbersome. 

But this new proposed feature should be implemented to make babystepping persistent when using Manual Mesh and give hosts feedback and confirmation that babysteps have been received and executed. (equivalent to BABYSTEP_ZPROBE_OFFSET for ABL offsets, but always displayed. Also, BABYSTEP_ZPROBE_OFFSET should always display ZPROBE_OFFSET for a similar benefit to ABL enabled variations of hostss like mine that would have to send a follow up command to query the results of the babystep move).

Minimizing host interaction is especially important to me, as I often run a dual-host setup on the same serial port and baystepping from the non Marlin Mode capable TFT while Octoprint is printing on that same serial port is already a high interruption risk event - thankfully I am ADVANCED_OK and line number checksum enabled to correct for errors. 

configuration_adv.h:
#define BABYSTEP_MBL_Z_OFFSET 

Suggested Changes to M290.cpp (note that both mentioned SERIAL_ECHOs could probably be put at the end of the function instead of being duplicated later in the function for naked M290s, depending on the intent of isolating them to naked commands to begin with):

void GcodeSuite::M290() {
  #if ENABLED(BABYSTEP_XY)
    LOOP_XYZ(a)
      if (parser.seenval(XYZ_CHAR(a)) || (a == Z_AXIS && parser.seenval('S'))) {
        const float offs = constrain(parser.value_axis_units((AxisEnum)a), -2, 2);
        babystep.add_mm((AxisEnum)a, offs);
        #if ENABLED(BABYSTEP_ZPROBE_OFFSET)
          if (a == Z_AXIS && (!parser.seen('P') || parser.value_bool())) mod_probe_offset(offs);
        #endif
      }
  #else
    if (parser.seenval('Z') || parser.seenval('S')) {
      const float offs = constrain(parser.value_axis_units(Z_AXIS), -2, 2);
//*************NEW BEGIN********************
      #if ENABLED(MESH_BED_LEVELING)
      #if ENABLED(BABYSTEP_MBL_Z_OFFSET)
      mbl.z_offset += offs;      
      #endif
      SERIAL_ECHOLNPAIR("MBL Adjust Z", mbl.z_offset);
      #endif
//*************NEW END********************
      #if ENABLED(BABYSTEP_ZPROBE_OFFSET)
        if (!parser.seen('P') || parser.value_bool()) mod_probe_offset(offs);
        SERIAL_ECHOLNPAIR(STR_PROBE_OFFSET " " STR_Z, probe.offset.z);//<******ALSO SUGGESTED NEW********
      #endif

boelle, 2020-07-31:
if possible submit a PR with the change

GMagician, 2020-07-31:
Have you tryed with BABYSTEP_ZPROBE_OFFSET?

PrintEngineering, 2020-08-01:
> if possible submit a PR with the change

I tried but I have made so many other changes to my fork that I do not know how to select only the relevant changes for the PR. Do you know how to cherry pick differences?  I did update the original post to hopefully make it easier to parse if not.

thinkyhead, 2020-08-02:
This change can't work, or it would already be there.

When Mesh Bed Leveling is enabled the `mbl.z_offset` is applied to Z as the nozzle moves. Since Babystepping also physically moves the nozzle up and down, by modifying the `mbl.z_offset` this change will cause the Z axis to move by double the babystep distance, which is not what you want.

In contrast, the probe offset which is used by Auto Bed Leveling (`M851 Z`) can be modified during movement by babystepping because that Z offset is not used during movement, only during probing.

closed 2020-08-02 by thinkyhead

PrintEngineering, 2020-08-02:
> This change can't work, or it would already be there.
>
> When Mesh Bed Leveling is enabled the `mbl.z_offset` is applied to Z as the nozzle moves. Since Babystepping also physically moves the nozzle [...]

Except it does not apply the new Z offset until the next homing event. Simply sending a G29 S4 Zxx.x results in no movement. The behavior is exactly like setting the PROBE_OFFSET on the fly, which also does not take effect until the next homing move. So for exactly the same reason BABYSTEP_PROBE_OFFSET works, so does this. What happens when you home and the new offset is taken into account is that babystepping is also reset to zero so there is a never a moment when both are applied simultaneously as you describe. 

If the intended behavior is that changes to mbl.z_offset are immediately taken into account, it is not working properly.

PrintEngineering, 2020-08-02:
Unless you are saying that changes to mbl.z_offset is applied to future moves under all circumstances, even prior to the next homing move?

If that is the case then it makes it very difficult to implement a menu for setting the Z offset because you have to generate a superficial movement, and then undo it, just to end up in the same position you began calibration

PrintEngineering, 2020-08-02:
It would also make it irrelevant to use M290 for babystepping, as G29 S4 Z would then be the only way to make babystepping changes persistent. I think that would be a very inconvenient and inconsistent implementation. If this is the case then I will direct my energy and solving it from that angle instead.

PrintEngineering, 2020-08-02:
Indeed it does move by double *when in motion from other moves*. That explains the inconsistent behavior. So... In order to make the changes persistent I have to also *NOT* apply babystepping, but in a way that will force movement to occur and still correct the current position. Maybe forcing it to its own current position after setting mbl.z_offset will correct the behavior for *when the print timer is not running*. It all seems so messy. I will think of something and resubmit something with the correct behavior

thinkyhead, 2020-08-02:
The thing about babysteps is that it needs to be pretty responsive, otherwise it doesn't serve its function.

So, while `M290` could simply **modify `mbl.z_offset` and not babystep** to achieve the same Z height and final effect, the side-effect is you wouldn't get regular babysteps. The motion would be uneven Z (based on how long each segment is and the movement speed, and so on).

However, there is a "trick" that I can think of in terms of making `M290` both produce regular babystepping and also modify `mbl.z_offset` without getting the extra motion.

The main constraint is we definitely want Z to move regularly and responsively and do real babystepping, but now we also want the planner to respect these babysteps whenever it makes new blocks to send to the stepper routine.

All moves at the stepper class level are relative, and there's one point where the planner decides how many steps to apply for the given destination. It does this by translating the new requested position into steps and then it subtracts from its last recorded position to get the distance….

So the trick is… 🥁  …to **add babysteps (as they are applied) onto the real stepper position**.

That way, when the planner is figuring the number of steps to apply on the next move, it will account for the difference applied to `mbl.z_offset`. The only thing that needs to be ensured is that the stepper interrupt is disabled when the counts are being modified.

PrintEngineering, 2020-08-02:
What if we just make mbl.z_offset behave like probe.offset.z? That seems like it would make the most sense and be the most consistent.

What I mean is to stop applying the *current* mbl.z_offset to each new movement, but load the saved mbl.z_offset value upon homing and use that value for all future moves during that homing cycle.

Then when homing occurs and babystep counts are reset to zero, the previously adjusted mbl.z_offset will be simultaneously applied and the behavior is 100% predictable and consistent. 

In that scenario we are letting the mechanisms already in place for babystepping do all of the messy work!

PrintEngineering, 2020-08-02:
I have tried to locate the place where mbl.z_offset is queried by the planner for current moves, but I can find no reference to actually *using*  the value of mbl.z_offset beyond displaying it. If you can point me to the right place to identify its pseudonym(s) I will be happy to make a working prototype of the changes.

PrintEngineering, 2020-08-02:
I think I found it. I'm currently looking at get_z() in mesh_bed_leveling.h

thinkyhead, 2020-08-03:
> What I mean is to stop applying the current `mbl.z_offset` to each new movement

That is considered a fundamental aspect of the implementation by its authors.

PrintEngineering, 2020-08-03:
I found a solution and I think it still honors the spirit of the original intent of the authors, as it still applies the EEPROM stored mbl.z_offset value (plus the babysteps) to every move. It just avoids *simultaneously* applying the babystep adjustment with the changes to the mbl.z_offset, as the difference between the the eeprom stored mbl.z_offset and the current mbl.z_offset is functionally equivalent to the babystepping value until homing when babysteps are zeroed out and the currently *applied* mbl.z_offset is set to the currently *adjusted* mbl.z_offset. I hope that makes sense because I feel like I could have explained it better.

I also allow the mbl.z_offset to be applied when leveling is declared invalid, as it is a scalar value across the bed that represents the true zero position. So even if you do not set a mesh, or there is a mesh error, you can still use the Z offset. And I found it disturbing that the z offset was not automatically applied after homing, so I added that position offset as well. 

I will try to show you how I made it work. There may be better ways to pull this off, as I am sort of hacking at your code here. If you are willing to accept the potential functionality of these modifications, then maybe you can adapt the changes to be more to your liking. 

This solution builds upon the original functionality and adds practical usefulness.

Give me a moment to figure out how to share the changes I have made.

PrintEngineering, 2020-08-03:
After looking at all of this, I think it would be better communicated through a pull request. Since it is already typed out, I'll still leave it here.

/*Configuration.h*/
    #define MESH_BED_LEVELING
    #ifdef MESH_BED_LEVELING
      #define BABYSTEP_MBL_Z_OFFSET
    #endif

/*mesh_bed_leveling.cpp*/
//line 37
  float mesh_bed_leveling::z_offset,
        #if ENABLED(BABYSTEP_MBL_Z_OFFSET)
          mesh_bed_leveling::z_offset_start,
        #endif

//near line 50
  void mesh_bed_leveling::reset() {
    z_offset 
    #if ENABLED(BABYSTEP_MBL_Z_OFFSET)
      = z_offset_start
    #endif
    = 0;
    ZERO(z_values);
    #if ENABLED(EXTENSIBLE_UI)
      GRID_LOOP(x, y) ExtUI::onMeshUpdate(x, y, 0);
    #endif
  }

/*mesh_bed_leveling.h*/
//line 41
class mesh_bed_leveling {
public:
  static float z_offset,
               #if ENABLED(BABYSTEP_MBL_Z_OFFSET)
                 z_offset_start,
               #endif

//near line 105
  static float get_z(const xy_pos_t &pos
    #if ENABLED(ENABLE_LEVELING_FADE_HEIGHT)
      , const float &factor=1.0f
    #endif
  ) {
    #if DISABLED(ENABLE_LEVELING_FADE_HEIGHT)
      constexpr float factor = 1.0f;
    #endif
    const xy_int8_t ind = cell_indexes(pos);
    const float x1 = index_to_xpos[ind.x], x2 = index_to_xpos[ind.x+1],
                y1 = index_to_xpos[ind.y], y2 = index_to_xpos[ind.y+1],
                z1 = calc_z0(pos.x, x1, z_values[ind.x][ind.y  ], x2, z_values[ind.x+1][ind.y  ]),
                z2 = calc_z0(pos.x, x1, z_values[ind.x][ind.y+1], x2, z_values[ind.x+1][ind.y+1]);

    return 
          #if ENABLED(BABYSTEP_MBL_Z_OFFSET)
            z_offset_start
          #else
            z_offset
          #endif 
          + calc_z0(pos.y, y1, z1, y2, z2) * factor;
  }

/*M290.cpp*/
void GcodeSuite::M290() {
  #if ENABLED(BABYSTEP_XY)
    LOOP_XYZ(a)
      if (parser.seenval(XYZ_CHAR(a)) || (a == Z_AXIS && parser.seenval('S'))) {
        const float offs = constrain(parser.value_axis_units((AxisEnum)a), -2, 2);
        babystep.add_mm((AxisEnum)a, offs);
        #if ENABLED(BABYSTEP_ZPROBE_OFFSET)
          if (a == Z_AXIS && (!parser.seen('P') || parser.value_bool())) mod_probe_offset(offs);
        #endif 
        #if ENABLED(BABYSTEP_MBL_Z_OFFSET)
          mbl.z_offset += offs; 
        #endif
      }
  #else
    if (parser.seenval('Z') || parser.seenval('S')) {
      const float offs = constrain(parser.value_axis_units(Z_AXIS), -2, 2);
      babystep.add_mm(Z_AXIS, offs);    
      #if ENABLED(BABYSTEP_ZPROBE_OFFSET)
        if (!parser.seen('P') || parser.value_bool()) mod_probe_offset(offs);
      #endif
      #if ENABLED(BABYSTEP_MBL_Z_OFFSET)
        mbl.z_offset += offs; 
      #endif

    }
  #endif

/*configuration_store.cpp*/
//line 1507 in the function _load()
        #if ENABLED(MESH_BED_LEVELING)
          if (!validating) mbl.z_offset =
          #if ENABLED(BABYSTEP_MBL_Z_OFFSET)
            mbl.z_offset_start =
          #endif
           dummyf;

/*motion.cpp*/
//near line 1419 in the function set_axis_is_at_home()
  #if HAS_POSITION_SHIFT 
    float diff = 0;
    
    #if ENABLED(BABYSTEP_MBL_Z_OFFSET)
      diff = 0-mbl.z_offset;
      mbl.z_offset_start = mbl.z_offset;
    #endif

    position_shift[axis] = diff;
    update_workspace_offset(axis);
  #endif

/*planner.cpp*/
//in the function apply_leveling()
  void Planner::apply_leveling(xyz_pos_t &raw) {
    if (!leveling_active){
      #if ENABLED(BABYSTEP_MBL_Z_OFFSET)      
        raw.z += mbl.z_offset_start;
      #endif
      return;
    } 

//at the end of the if (leveling_active){} of unapply_leveling()
   if (leveling_active) { }
    #if ENABLED(BABYSTEP_MBL_Z_OFFSET)
      else raw.z -= mbl.z_offset_start;
    #endif
  }

PrintEngineering, 2020-08-03:
This code seems to function not only in the original capacity, but allows the mbl.z_offset to function in an equivalent capacity to probe.offset.z. Nothing of the original functionality is given up, as far as I can tell. And a few tests indicate that it works (unless I have missed something again). The nice part is that with MBL we have a functional Z offset that can be globally applied and manipulated just like with probe.offset.z! I feel like that is something that is really lacking!

PrintEngineering, 2020-08-03:
*I updated the code above because it only worked with an Invalid Mesh. Now, the mbl.z_offset is globally applied on homing. I am fairly certain there is alot of extraneous code happening here, but it does work. This is just a proof of concept at this point. I will continue testing to see what I can distill it down to before submitting a PR. Also open to suggestions.

PrintEngineering, 2020-08-05:
*I have updated the code again because the previous edit failed to reflect the correct final axis position. I don't understand your code well enough for an elegant solution yet, so upon homing I resorted to:

  if(axis == Z_AXIS)babystep.add_mm(Z_AXIS, mbl.z_offset); 

Though I was able to hijack the machinery for babysteps instead of building new functionality from scratch, I do consider this pretty hacky since it basically merges babysteps and z offset and I get the feeling you will find that in poor taste. I am not even going to bother with a PR because I doubt it will be approved like this but I really hope that you consider implementing some functionally similar solution. It is worlds easier and more functional to use.