MarlinFirmware/Marlin #18869: [BUG]Delta generic config do not get NUM_AXIS_N correctly.

yoch2015, 2020-07-31:
### Bug Description
The latest version of Marlin-bugfix-1.1.x with the delta generic config.
The board is MKS_GEN_L and if the extruder is 2, it will be caught on a sanity check.

#define EXTRUDERS 2
#define SINGLENOZZLE
#define DEFAULT_AXIS_STEPS_PER_UNIT   { DEFAULT_XYZ_STEPS_PER_UNIT, DEFAULT_XYZ_STEPS_PER_UNIT, DEFAULT_XYZ_STEPS_PER_UNIT, 150.9, 188.62 }

Fixing SanityCheck.h avoids the error, so it seems that you are not getting NUM_AXIS_N correctly.

// static_assert(COUNT(sanity_arr_1) <= NUM_AXIS_N, "DEFAULT_AXIS_STEPS_PER_UNIT has too many elements.");
static_assert(COUNT(sanity_arr_1) <= 5, "DEFAULT_AXIS_STEPS_PER_UNIT has too many elements.");

**Expected behavior:** [What you expect to happen]
Error Avoidance .

**Actual behavior:** [What actually happens]
static assertion failed: DEFAULT_AXIS_STEPS_PER_UNIT has too many elements.

[config.zip](https://github.com/MarlinFirmware/Marlin/files/5004790/config.zip)

thisiskeithb, 2020-07-31:
`1.1.x` is no longer receiving updates.

Please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

yoch2015, 2020-07-31:
Oh, I see.
I'm sorry.
Close.

closed 2020-07-31 by yoch2015

yoch2015, 2020-07-31:
It was the same in bugfix-2.0.x.
The XYZE variable is not set correctly.
Adding the following to Configuration.h solves the problem.
#define XYZE 5

yoch2015, 2020-07-31:
oops,
#define DISTINCT_E_FACTORS
Was missing.