MarlinFirmware/Marlin #18874: Archim 2 Watchdog not working

ZS1FJK, 2020-07-31:
Hello

My Printer set on fire and burnt down about 8 months ago. I noticed a week earlier to that the printer had hung and kept the outputs on. I then upgraded and flashed new firmware and rechecked all the wiring and formatted the sd card. After a weak of running 8 h prints successfully i thought the issue was resolved. It was not. 

I have now replaced all wires on my printer and installed new Archim 2 board and got the printer working. It hung again 3 times today so far .

steps to follow

Load stl file onto sd card. 
Start print.

Problems

Printer hangs and display goes blank. all Outputs stay on. 
Printer dose not reset. 

I have now replaced the cables that run from the screen to the board in a hope that this was the issue. I Think the watchdog on the archim 2 is not working correctly and is dangerous.

ellensp, 2020-07-31:
Please attach your configuration files.

ZS1FJK, 2020-07-31:
Sorry I thought i had done this.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5007801/Marlin.zip)

ellensp, 2020-07-31:
"display goes blank." is interesting... as the displays normally just freezes with the last displayed data if cpu crashes...

ZS1FJK, 2020-07-31:
Yes i thought this was strange. That is why i have changed the cables between the main board and screen. I will test for the next week and see if i can get it to hang again. 

What concerns me the most is the output staying fully on. My understanding is that the watch dog should prevent this from happening and reset even in the case of the a faulty memory card ,faulty display or faulty wiring between display and main board. 

I can perhaps try reproduce the fault by unplugging a cable while in operation. Don't know if this could blow something tho.

Thank you for the fast response.

ellensp, 2020-07-31:
Been doing some testing.

I don't have a real Archim 2, but have a Due (which is very close)
I used your configs and env DUE_archim 
I changed thermisors to type 998 (25c always)
I rigged the gcode G1 with a C parameter that goes into an infinite loop. This can be used to trigger the watchdog
In testing I quickly (if you don't do this quickly you trigger heating errors) send a M104 S100 and M140 S100 and finally G1 C0 and...     

The board seems to not reliably reboot. 

When it does reboot, it works as expected.
When it doesn't? It starts a reboot, disabled USB CDC, enables USB CDC, but then seem to just stop. This is while IO lines like heaters are still active. In this state you cannot talk to the board.

Things of note: 
One of the flags used when setting up watchdog is WDT_MR_WDRPROC  which the comments tell me 
"WDT fault resets processor only - We want to keep PIO controller state"
But do we?  I don't think we want this.... 
I think that because USB is still initialized from before the reboot is causing the issue.

If I comment out this flag it seems to reboot reliably and turn off IO's reliably.

ZS1FJK, 2020-08-03:
I have run several short prints without any problems after changing the cables from the screen to the main board. Tomorrow morning i will be buying a smoke alarm and putting it onto my printer and connecting it to my house alarm. 

I noticed another thread about the watchdog not working [here](https://github.com/MarlinFirmware/Marlin/issues/13558)

ellensp, 2020-09-03:
Still need looking into.

ZS1FJK, 2020-09-03:
> 
> 
> Still need looking into.

I agree. Rather dangerous bug this.

ZS1FJK, 2020-10-04:
I don't think this issue has been looked into.

ellensp, 2020-11-04:
go away bot!  (this comment is a good as any other)

sjasonsmith, 2020-11-04:
Does anybody know what all boards could be used to reproduce this? Is it specific to Archim, or is a DUE sufficient?

ellensp, 2020-11-04:
I was testing on a due. It has same lockup.

sjasonsmith, 2020-11-04:
@V1EngineeringInc and @jeffeb3 have both made recent contributions toward Archim support. I wonder if one of them would be able to help with this issue.

jeffeb3, 2020-11-04:
We worked on it in the V1 forums:

https://forum.v1engineering.com/t/crashing-to-blank-lcd-screen-on-archim-board/20971?u=jeffeb3

That is a long thread and we were learning as we went. But the symptoms sound the same. We ended up with the solution for the hanging on an archim 1 board here:

https://github.com/MarlinFirmware/Marlin/pull/19596

I believe this should also fix the archim 2, since it is mostly the same except for the drivers. 

This fix is in 2.0.7.1. 

It's really hard for me to say for sure it is fixed, but anecdotally, it looks like it is working. But fires are scary. Please be careful. If my printer had hung like this. I would have replaced the board and not looked back.

sjasonsmith, 2020-11-10:
If you enable `MARLIN_DEV_MODE` I added a `D100` command which can be used to verify watchdogs are working.
Just send a `D100` command over serial, and it will attempt to hard-hang the board for 10 seconds. It disables heaters before hanging just to be safe. If the watchdog is working, it should reboot. If the watchdog is broken, then it will unblock and report back a failure message to the console.

@jeffeb3 was the actual fix all in the variant, such that it might improve Archim boards but would not have improved every board in the HAL (such as DUE)?

jeffeb3, 2020-11-10:
The fix we were targeting was the frequency change and that ended up in the arduino_due_x variant. I have no idea which boards use that or even how that is related to the watchdog. I also don't own an archim, so I can't test.

closed 2020-12-18 by github-actions[bot]