MarlinFirmware/Marlin #18876: [FR] Separate K values for speeding up and slowing down

Liamnerf, 2020-07-31:
### Description

For the last few month I've been attempting to get LA tuned with my printer. In my testing I've found that at very low speeds (say, 5mm/s) and jerk values (also 5) LA advance doesn't work properly. Running the calibration pattern results in no good uniform lines.

analysing the lines though, it seems like being able to set separate k values for when the nozzle speeds up and slows down would allow for perfect tuning. anyone else think this would he the case?

### Feature Workflow

1. Set K factor for acceleration moves 
2. Set K factor for deceleration moves

XDA-Bam, 2020-08-02:
This has been discussed, tested and rejected in #16880 . I agree that it sounds logical and should in theory give superior results. Sadly, it doesn't work as expected in practice 😐 

The way LA is currently implemented, there is no relevant improvement possible using different K-factors for acceleration and deceleration. The main problem is, that the timing doesn't change with different K and thereby, delays and hysteresis behaviours of the filament supply can't be accounted for. But these are responsible for the different behaviours in acceleration and deceleration. To actually improve printing results, you would need to look ahead a couple of millimeters or centimeters in the planner and adjust the filament flow accordingly. This currently isn't possible due to the architecture of the Marlin planner.

In case you changed some stuff in the code yourself and got a better result for the test pattern, I encourage you to print some "real" parts with your code. You will see that they are either not improved at all, or that there are regions which look a bit better and others, which look worse. At least that's what I got in my tests.

dbuezas, 2024-03-10:
I have this exact issue (blob when slowing down) and it is visible in real prints too.
It's been a while, has there been any new solution to this?

dbuezas, 2024-08-18:
Informed by https://github.com/MarlinFirmware/Marlin/issues/16880 and some logging/debugging, I'm fairly confident I found the reason behind the different best values for k on acceleration vs deceleration. My answer is jerk:
The exit speed of the fast segment is higher than the initial rate of the subsequent slow segment, and this is by design (this is what jerk is for). But the current linear advance algorithm doesn't handle these discontinuities, and that results in extra unaccounted for pressure in the nozzle when starting the slow segment on the right.

While experimenting [with a new faster LA algorithm](https://github.com/MarlinFirmware/Marlin/pull/27352), I added logic to account for extrusion speed discontinuities and the right blobs are almost completely gone, which results in higher quality, particularly seams and top surfaces.

@XDA-Bam, thanks for your investigations in the topic, particularly for posting about failed experiments. I think you may have found the actual reason back then since you mentioned the lack of planner lookahead.