MarlinFirmware/Marlin #18878: [BUG] "ESP3D_WIFISUPPORT or WIFISUPPORT requires an ESP32 controller."

jjtronics, 2020-07-31:
### Bug Description

ESP3D_WIFISUPPORT Does not work

Error message produced

### My Configurations

BTT SKR 1.4 TURBO
### Steps to Reproduce

#define SERIAL_PORT 3

/**
 * Select a secondary serial port on the board to use for communication with the host.
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT_2 -1

**Expected behavior:** [What you expect to happen]
I have BTT SKR1.4 Turbo and ESP01S and i use ESP3D_WIFISUPPORT on Marlin 2.0.54 and work very well but i migrate to Marlin 2.0.6 and it's not possible to compile with ESP3D_WIFISUPPORT enabled.
I have this error : 
ESP3D_WIFISUPPORT or WIFISUPPORT requires an ESP32 controller.
But in Marlin 2.0.5.4 work very well with ESP01S (ESP 8266 not ESP32).

Thank you very munch for you help and sorry for my english

thisiskeithb, 2020-07-31:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/18855

thisiskeithb, 2020-07-31:
> BTT SKR 1.4 TURBO

As the error message states, you need an ESP32-based motherboard to enable `ESP3D_WIFISUPPORT`/`WIFISUPPORT`. The SKR 1.4 Turbo is an LPC1769-based motherboard.

Full instructions on installing an ESP32 wifi module on an SKR 1.4 (not turbo) are at https://youtu.be/NGgzw-XayEo

In this particular tutorial Chris Riley chooses to set `SERIAL_PORT` to `3` and `SERIAL_PORT_2` to `-1`.

___
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-07-31 by thisiskeithb

jjtronics, 2020-07-31:
Thank you, i found the problem with the TouchUI (Esp is connecter on the screen) and Marlin 2.0.6 i set a value in the ini file
serial_always_on: 1 and my ESP3D work again !

radek8, 2020-08-01:
I have the same problem
## This is a problem in Marlin.
It worked very well and now cannot be compiled.

radek8, 2020-08-01:
The problem is here
https://github.com/MarlinFirmware/Marlin/pull/18792
and
https://github.com/MarlinFirmware/Marlin/pull/18808

After commenting on these lines, everything is fine

#if EITHER(ESP3D_WIFISUPPORT, WIFISUPPORT) && DISABLED(ARDUINO_ARCH_ESP32)
  #error "ESP3D_WIFISUPPORT or WIFISUPPORT requires an ESP32 controller."
#endif

ellensp, 2020-08-02:
Problem is users ESP3D_WIFISUPPORT and WIFISUPPORT are exclusively for esp32 based controllers.
NOT FOR STAND ALONE esp01/esp3d modules which only require a serial port

It is NOT for LPC1768/9 based controllers.

ellensp, 2020-08-02:
Re watch the video https://youtu.be/NGgzw-XayEo 

This video shows how you need to modify the lpc1768/9 framework and change the serial pins used to by serial port 3. As BTT in their Infinite wisdom decided to use non standard serial pins. 4.28 and 4.29
Where as the default pins for the framework are as follows.
```
Port    |   TX  |  RX
   --   |  --   |    --
Serial  | P0_02 | P0_03
Serial1 | P0_15 | P0_16
Serial2 | P0_10 | P0_11
Serial3 | P0_00 | P0_01

```