MarlinFirmware/Marlin #18882: Nozzle temperature fluctuates despite pid[BUG] (short description)

Nissal1983, 2020-08-01:
The temperature of the nozzle fluctuates between 8-11 degrees despite the Pid, ​​which was not the case with 2.0.5.4. Tested on an Anycubic i3 Mega

boelle, 2020-08-01:
@Nissal1983 please add [BUG] to start of title

oneeyepizzaguy404, 2020-08-01:
I am having similar issues 

No matter what I do im stuck at a 5c droop when my fans come on tuned with the fans on. 

Printer seems to be ignoring the functional thresh hold as well and you can see it cycle via the led as soon as it starts heating. 

No matter what values i enter via m301 the behavior does not change.

The bed behaves correctly.

50watt heater, silicone sock, skr 1.3 
Official marlin release 2.0.6.

Nissal1983, 2020-08-01:
My configuration.h and configuration_adv.h
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5010742/Configuration.zip)

Nissal1983, 2020-08-02:
> I am having similar issues
>
> No matter what I do im stuck at a 5c droop when my fans come on tuned with the fans on.
>
> Printer seems to be ignoring the functional thresh hold as well and you can see [...]

you solved the pid problem with you, how and where did you go?

oneeyepizzaguy404, 2020-08-02:
So my problem was in the pins files if/then statements in the bbt common pins. The exttruder pin p2_04 was also being assigned a parts fan function if you only have one hotend. The sanity check missed this.

Nissal1983, 2020-08-02:
Thank you. i had a similar problem. I'm closing here

closed 2020-08-02 by Nissal1983