MarlinFirmware/Marlin #18885: [BUG] Ender 3V2 : BLTouch keep "clicking" Stowing and Deploying continuously upon switching on printer.

fleek, 2020-08-01:
<### Bug Description

I just uploaded a newly compiled version of the bugfix-2.0.x branch today and the BLTouch is going through this continuous cycle of stowing and deploying the pin upon startup and it never stops until I do a autohome.

Previously it would go through a few cycles and then stow the detection pin.

### My Configurations

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5010479/Marlin.zip)

### Steps to Reproduce

1. Turn on printer.

**Expected behavior:** [What you expect to happen]

BLTouch will go through a few cycles of deploying and stowing and stop.

**Actual behavior:** [What actually happens]

BLTouch will keep cycling on deploying and stowing the detection pin.

#### Additional Information

I have been using the same configuration files and the previous compile maybe a week before was exhibiting the correct behavior, I don't know what has changed to the startup code or the BLTouch code.

CRCinAU, 2020-08-06:
I haven't had any reports of this - but normally, this means you've got the output plugged into a sensor port that isn't configured for the probe...

ie zstop when it should be PROBE or other way around...

boelle, 2020-08-08:
@fleek did you check what @CRCinAU suggested?

fleek, 2020-08-08:
@boelle My BLTouch was working fine until some recent commits were put in place, as for what @CRCinAU has alluded to, I don't quite agree because it was working fine.  I was pretty busy for these couple of days, so I didn't really have time to step through the code to see whats wrong, but I suspect that something was committed for some other displays was affected this.  This is because the clicking seems looks like a timed sequence rather than random with a period of 1 sec.

boelle, 2020-08-08:
strange one, i never seen that

but i dont use an Ender so might be that, all mine does is the selftest at power up

sjasonsmith, 2020-08-08:
If the behavior changed in a recent commit I suggest you go back and try to locate the commit where the problem began.

It sounds like your probe is running a self-test command instead of the expected “stow” command. Self-test should stop after 10 cycles for a legit BLTouch. It might run forever on clones or older models.

Is there any chance you have any equipment to be able to observe the output from the servo pin?

CRCinAU, 2020-08-08:
The new code seems to be global - and I haven't seen any reports of this on an Ender 3 V2 or SKR board. Unless there's a specific option that causes this - and I have tried some pretty tight settings on the BLTouch, I'm not sure it'd be a marlin issue.

sjasonsmith, 2020-08-09:
@fleek also make sure you carefully check your wiring. If the power connection is flaky it might be repeating the power-on self-test over and over. BLTouch wiring is very fragile, so it is possible it became damaged and you only think it corresponded to a firmware change.

sjasonsmith, 2020-08-17:
@fleek did you resolve this issue?

fleek, 2020-08-17:
@sjasonsmith too busy with work to tinker with the machine these past two weeks.. will try to find time to look at it.

boelle, 2020-09-11:
@fleek had time to look at it?

closed 2020-09-23 by github-actions[bot]

sjasonsmith, 2020-09-23:
@fleek this report was closed out. Once you have time to spend with this again, try one of the various support forums available. I hang our frequently on Discord, but you have multiple options available.

- MarlinFW.org [Marlin Documentation](http://marlinfw.org/)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you could report a new issue or comment on this one if it needs to be re-opened. (https://github.com/MarlinFirmware/Marlin/issues/new/choose).