MarlinFirmware/Marlin #18887: [Bug] Nozzle temperature pid values make no change to the control loop

oneeyepizzaguy404, 2020-08-01:
No matter what I do im stuck at a 5c droop when my fans come on tuned with the fans on.

Printer seems to be ignoring the functional thresh hold as well and you can see it cycle via the led as soon as it starts heating.

No matter what values i enter via m301 the behavior does not change.
P1000 I0 D0/ P10 I1 D10 ect

The bed behaves correctly. 
Note sure but might be helpful only my machines set up as coreyx are doing this not the Cartesian printers. 

50watt heater 24, silicone sock, skr 1.3 
Official marlin release 2.0.6
[bug.zip](https://github.com/MarlinFirmware/Marlin/files/5010497/bug.zip)

qwewer0, 2020-08-01:
Did you do a pid autotune?

oneeyepizzaguy404, 2020-08-01:
M303 e0 s230(tried 215) c12 

It will tune correctly. And hold temp while in self tune m303. While fans are running at z0 in the center of the build plate.

I take the values from the the tune and put them in via m301 px ix dx. Save with m500.

When setting the temp via the screen or gcode when printing it will drop 5+c once the fans come on.  Eventually it will time out do to a watchdog error. It is never able to gain any ground when not in self tune. 

I have also just tried puting in m301 p i and d vaules to see if i can get a change.

oneeyepizzaguy404, 2020-08-01:
Found my issuse it was a conflict with the if then logic in the skr commons

closed 2020-08-02 by oneeyepizzaguy404