MarlinFirmware/Marlin #18888: bltouch for G29 et G34 does not go to the point planned for deployment

boubouexpress, 2020-08-01:
hello
the bltouch works perfectly for g28
after a G28 it is decent to return the probe

for bed level by LCD screen:
he deploys and after the G28 he decent to enter the probe,the probe is not extended to perform the G29
to avoid any damage I had to activate manual probe.
same problem for auto Z alignment ,which has in addition an additional problem when it finishes it decent too low to retract the probe,that works perfectly with the G29

GMagician, 2020-08-01:
try to compile with debug option, use M111 S255 and report log you get

boubouexpress, 2020-08-01:
echo:DEBUG:ECHO,INFO,ERRORS,DRYRUN,COMMUNICATION,LEVELING

GMagician, 2020-08-01:
I don't need configuration but log when firmware is compiled with `DEBUG_LEVELING_FEATURE`

boubouexpress, 2020-08-01:
Connecting...
Printer is now online.
>>>M111
SENDING:M111
echo:DEBUG:INFO,ERRORS,LEVELING
>>>G28
SENDING:G28
>>> G28 8 X306.00 Y243.00 Z9.71
Machine Type: Cartesian
Probe: TOUCH_MI_PROBE
Probe Offset X-51.00 Y10.00 Z-4.52 (Left-Back & Below Nozzle)
Auto Bed Leveling: BILINEAR (enabled)
ABL Adjustment Z-0.194
  current_position= X306.00 Y243.00 Z9.71 : Leveling ON
  current_position= X306.00 Y243.00 Z9.52 : ...Now OFF
  current_position= X306.00 Y243.00 Z9.52 : sync_plan_position
Raise Z (before homing) by 10.00
>>> do_blocking_move_to 8 X306.00 Y243.00 Z9.52
>  X306.00 Y243.00 Z10.00
<<< do_blocking_move_to 8 X306.00 Y243.00 Z10.00
>>> homeaxis(X)
Home 1 Fast:
>>> do_homing_move 8 X306.00 Y243.00 Z10.00
...(X, -771.00, [50.00])
echo:busy: processing
echo:busy: processing
echo:busy: processing
<<< do_homing_move 8 X306.00 Y243.00 Z10.00
Move Away:
>>> do_homing_move 8 X306.00 Y243.00 Z10.00
...(X, 20.00, [50.00])
<<< do_homing_move 8 X306.00 Y243.00 Z10.00
Home 2 Slow:
>>> do_homing_move 8 X306.00 Y243.00 Z10.00
...(X, -40.00, 12.50)
echo:busy: processing
<<< do_homing_move 8 X306.00 Y243.00 Z10.00
>>> set_axis_is_at_home(X)
Axis X home_offset = 0.00 position_shift = 0.00
> home_offset[X] = 0.00
  current_position= X6.00 Y243.00 Z10.00 :
<<< set_axis_is_at_home(X)
  current_position= X6.00 Y243.00 Z10.00 : sync_plan_position
  current_position= X6.00 Y243.00 Z10.00 : > AFTER set_axis_is_at_home
<<< homeaxis(X)
>>> homeaxis(Y)
Home 1 Fast:
>>> do_homing_move 8 X6.00 Y243.00 Z10.00
...(Y, -760.50, [50.00])
echo:busy: processing
echo:busy: processing
<<< do_homing_move 8 X6.00 Y243.00 Z10.00
Move Away:
>>> do_homing_move 8 X6.00 Y243.00 Z10.00
...(Y, 20.00, [50.00])
echo:busy: processing
<<< do_homing_move 8 X6.00 Y243.00 Z10.00
Home 2 Slow:
>>> do_homing_move 8 X6.00 Y243.00 Z10.00
...(Y, -40.00, 12.50)
echo:busy: processing
<<< do_homing_move 8 X6.00 Y243.00 Z10.00
>>> set_axis_is_at_home(Y)
Axis Y home_offset = 0.00 position_shift = 0.00
> home_offset[Y] = 0.00
  current_position= X6.00 Y-5.00 Z10.00 :
<<< set_axis_is_at_home(Y)
  current_position= X6.00 Y-5.00 Z10.00 : sync_plan_position
  current_position= X6.00 Y-5.00 Z10.00 : > AFTER set_axis_is_at_home
<<< homeaxis(Y)
[... 869 lines cut ...]
  current_position= X306.00 Y243.00 Z4.52 : > AFTER set_axis_is_at_home
  current_position= X306.00 Y243.00 Z4.52 : Probe::set_deployed
deploy: 0
Probe::do_z_raise(5.00)
>>> do_blocking_move_to 8 X306.00 Y243.00 Z4.52
>  X306.00 Y243.00 Z9.52
<<< do_blocking_move_to 8 X306.00 Y243.00 Z9.52
>>> do_blocking_move_to 8 X306.00 Y243.00 Z9.52
>  X306.00 Y243.00 Z0.50
echo:busy: processing
<<< do_blocking_move_to 8 X306.00 Y243.00 Z0.50
>>> do_blocking_move_to 8 X306.00 Y243.00 Z0.50
>  X306.00 Y243.00 Z9.52
<<< do_blocking_move_to 8 X306.00 Y243.00 Z9.52
>>> do_blocking_move_to 8 X306.00 Y243.00 Z9.52
>  X306.00 Y243.00 Z9.52
<<< do_blocking_move_to 8 X306.00 Y243.00 Z9.52
<<< homeaxis(Z)
<<< home_z_safely 8 X306.00 Y243.00 Z9.52
  current_position= X306.00 Y243.00 Z9.52 : sync_plan_position
<<< G28 8 X306.00 Y243.00 Z9.52
  current_position= X306.00 Y243.00 Z9.52 : Leveling OFF
  current_position= X306.00 Y243.00 Z9.71 : ...Now ON
  current_position= X306.00 Y243.00 Z9.71 : sync_plan_position
<<< G34 8 X306.00 Y243.00 Z9.71

boubouexpress, 2020-08-01:
X91.00 Y243.00 Z0.50
echo:busy: processing
<<< do_blocking_move_to 8� X91.00 Y243.00 Z0.50
do_blocking_move_to 8� X91.00 Y243.00 Z0.50

boubouexpress, 2020-08-01:
however it indicates the correct value 0.5 I do not understand who goes much lower has deformed the bed

boubouexpress, 2020-08-05:
after several modifications and always try the problem for me it is a bug in marlin

boubouexpress, 2020-08-11:
no one for this problem
it is not normal that the touchmi retractation problem occurs only on G34

GMagician, 2020-08-11:
I'm sorry I have no hardware to test. With bltouch G34 works perfectly, don't know which differences your probe has

sjasonsmith, 2020-09-11:
I actually bought a Touch Mi just so I could understand how it works and try to reproduce issues like this.

closed 2020-10-19 by github-actions[bot]