MarlinFirmware/Marlin #18889: [BUG] (short description)

anm8tor, 2020-08-01:
### Bug Description

Error compiling Marlin 2.0.06 for Anycubic Chiron

### My Configurations

[anm8tor_M206_configs.zip](https://github.com/MarlinFirmware/Marlin/files/5010782/anm8tor_M206_configs.zip)

### Steps to Reproduce

1. Change Config.h and Config_adv.h settings to match known Chiron settings from supplied files on Anycubic's GitHub
2. Open Arduino IDE 1.8.13 (Windows Store 1.8.42.0), ensure Board is "Arduino Mega or Mega 2560" and Processor is "ATMega 2560 (Mega 2560), then select Compile
3. Receive Errors and fail to compile

**Expected behavior:** [What you expect to happen]

Compile firmware successfully

**Actual behavior:** [What actually happens]

Error compiling for board Arduino Mega or Mega 2560

#### Additional Information

none

ellensp, 2020-08-01:
No error logs provided..

"BOARD_TRIGORILLA_14"

I get this error, but cant say its its the same error they are getting

"Marlin/src/HAL/AVR/../../inc/SanityCheck.h:2009:6: error: #error "Z2_USE_ENDSTOP has been assigned to a nonexistent endstop!""

fleek, 2020-08-01:
Anycubic Chiron's  is not a fork from Marlin and was not updated for two years.  You should ask Anycubic for support and not us.

ellensp, 2020-08-01:
I'm using real actual marlin, and what they do provide is 20006 config files. I don't think they are using a foreign repo.

ellensp, 2020-08-01:
Its is a rogue conditional.

#if PIN_EXISTS(Z2_MIN)
  #define HAS_Z2_MIN 1
#endif
#if PIN_EXISTS(Z2_MAX)
  #define HAS_Z2_MAX 1
#endif

with 

  #elif !HAS_Z2_MIN && !HAS_Z2_MAX
    #error "Z2_USE_ENDSTOP has been assigned to a nonexistent endstop!"

In sanitycheck.h  it is now assuming that Z2_MIN_pin and Z2_MAX_pin exsist, when infact most use 
Z2_USE_ENDSTOP to define what endstop to use.

fleek, 2020-08-01:
@ellensp I am sorry, but I checked their github, they are not forked from our Marlin repo.  I think the bug reporter should have used the config files from our Configurations repo rather than from their github and hope that everything works.

ellensp, 2020-08-01:
Config has     #define Z2_USE_ENDSTOP          _ZMAX_
But this board doesn't have ZMAX endstops  it only has a Z_STOP_PIN 

define this to use a real pin.

closed 2020-08-01 by ellensp

anm8tor, 2020-08-02:
Yes, I inadvertently changed XMAX on that line to ZMAX based on settings reviewed on a known working version of 2.0.0. Thanks for helping everyone!