MarlinFirmware/Marlin #18890: Enabling Filament Runout Detection fetaure for Robin_Nano_v1.2 with lvgl_fsmc option results an error about advanced_pause_feature requiring an lcd_controller

halilbil, 2020-08-01:
### Bug Description
I added a filament runout sensor to my printer which has a robin nano v1.2 and a fsmc connected 3.5" tft. When I try to compile, I get an error message that states "advanced_pause_feature currently requires an lcd controller" although I enabled an lcd option in Configuration.h

### My Configurations

I enabled following features in the configuration files:

Configuration.h:
#define TFT_LVGL_UI_FSMC
#define FILAMENT_RUNOUT_SENSOR
#define NOZZLE_PARK_FEATURE

Configuration_adv.h
#define ADVANCED_PAUSE_FEATURE

My config files can be found below:
[https://drive.google.com/file/d/1vQ6hRMDVjwUngJNwjTTHx9QD3_YC6wlZ/view?usp=sharing](url)

### Steps to Reproduce

1. In pilatformio.ini set default_envs = mks_robin_nano35
2. Use attached Configuration.h and Configuration_adv.h when compiling

**Expected behavior:** Successfull compilation

**Actual behavior:** Error during compilation

rhapsodyv, 2020-08-01:
It’s a bug. Maybe the message is a bit misleading.

Right now, only the text menu implements all Marlin features.

The LVGL ui don’t support all those features. 

If you change the FILAMENT_RUNOUT_SCRIPT to “M25” instead of “M600” it will compile fine.

halilbil, 2020-08-02:
Thanks for the reply. I was able to compile it when I changed FILAMENT_RUNOUT_SCRIPT to M25 and also disable ADVANCED_PAUSE_FEATURE.

Now a side question. Since you said only the text menu implements all Marlin features, I tried to compile the software by disabling TFT_LVGL_UI_FSMC and enabling FSMC_GRAPHICAL_TFT and TOUCH_BUTTONS. This resulted in a partly filled screen. The screen was also irresponsive to any touch. I guess this is because the screen that comes with Robin Nano v1.2 has a resolution of 480x320 and default resolution in Marlin is different that that. I am guessing I have to alter ultralcd_DOGM.h file but I am not sure. 

A very similar question is also mentioned in issue number #18899.

closed 2020-08-03 by halilbil

thinkyhead, 2020-08-04:
Earlier today there was a patch made for screen pixel sizes, so give the latest code a try. If it still doesn't work for your setup, then try defining `FSMC_UPSCALE` with a value of 3 in your configuration, or maybe there will be a better place for it….

rhapsodyv, 2020-08-04:
> Earlier today there was a patch made for screen pixel sizes, so give the latest code a try. If it still doesn't work for your setup, then try defining `FSMC_UPSCALE` with a value of 3 in your configuration, or maybe there will be a better place for it….

His problem was fixed in: #18901.