MarlinFirmware/Marlin #18891: Compile Error CR-10 Mini with BLTouch

dfiorucci, 2020-08-01:
[#](url
[Compile Issue.zip](https://github.com/MarlinFirmware/Marlin/files/5010895/Compile.Issue.zip)
I receive this error message every time I try to compile fro the CR-10 Mini.
I've tried with the modified version that is attached and also the stock configuration files.
Windows 7 64 bit
Arduino 1.8.13
Marlin 2.0.6
I also received the same error on 
Windows 10 64 bit
Arduino 1.8.13
Marlin 2.0.6

Same files will compile in Marlin 1.1.9 but will not upload due to this error.

Upload Error Message Marlin 1.1.9
Sketch uses 128228 bytes (98%) of program storage space. Maximum is 130048 bytes.

Global variables use 4663 bytes (28%) of dynamic memory, leaving 11721 bytes for local variables. Maximum is 16384 bytes.

avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x65

avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x63

avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x68

avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x6f

avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x3a

avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x54

avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x46

An error occurred while uploading the sketch

avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x20

avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x69

avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x6e

Error Message Marlin 2.0.6
Arduino: 1.8.13 (Windows 7), Board: "Sanguino, ATmega1284 or ATmega1284P (16 MHz)"

fork/exec C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc.exe: The filename or extension is too long.

Error compiling for board Sanguino.

fleek, 2020-08-01:
Thats nothing have to do with Marlin, your board cannot take the compiled size of the code, either you disable some functions or upgrade to a new board with more memory.

dfiorucci, 2020-08-01:
I don't think that is the issue in the fact that the stock configurations that don't have BLTouch enabled produce the same error.
Also you see that I'm able to compile with Marlin 1.19 just have an issue uploading for some reason and yes I've put the boot loader on the board.
Marlin 1.1.9 Compile results
Sketch uses 128228 bytes (98%) of program storage space. Maximum is 130048 bytes.
Global variables use 4663 bytes (28%) of dynamic memory, leaving 11721 bytes for local variables. Maximum is 16384 bytes.

fleek, 2020-08-01:
Are you able to upload that 1.1.9 compile?

dfiorucci, 2020-08-01:
No I get the error message that I stated in the first message
Upload Error Message Marlin 1.1.9
Sketch uses 128228 bytes (98%) of program storage space. Maximum is 130048 bytes.

Global variables use 4663 bytes (28%) of dynamic memory, leaving 11721 bytes for local variables. Maximum is 16384 bytes.

avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x65

avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x63

avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x68

avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x6f

avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x3a

avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x54

avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x46

An error occurred while uploading the sketch

avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x20

avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x69

avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x6e

dfiorucci, 2020-08-01:
The port is recognized and I'm able to confirm it in device manager.

fleek, 2020-08-01:
though it is at 98%, thats a measure based on the assumption that you have full 128K memory, but the bootloader itself will take about 2k at least.  The only way to upload that is to forego your bootloader and upload via ISP.

dfiorucci, 2020-08-01:
baud rate is 115200

dfiorucci, 2020-08-01:
I have the ISP burner that came with the Creality3d kit that is what I used to burn the boot loader with using progisp72.exe
So what is the process for loading this file that I've modified in Marlin via the ISP?
Can I save the Marlin configuration to a hex file so I can use the  progisp72.exe?
I'm able to load files with that.

dfiorucci, 2020-08-01:
I think I found it "Export Compiled Binary"

ellensp, 2020-08-01:
Not a support forum. Not an marlin issue.
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-08-01 by ellensp

fleek, 2020-08-01:
Once you overwrite the boot loader, you can only load via ISP and not longer via arduino IDE anymore.  you can use the progisp72.exe program to load the hex file.  But I would suggest disabling some functions like debug/SDSupport etc, there are quite a lot of comments in the configuration file to save some bytes.

dfiorucci, 2020-08-01:
Well I was guessing that it was a bug since it compiled in an earlier version but not the new version.