MarlinFirmware/Marlin #18894: Printer Crashes after the 1st layer

RosalieWessels, 2020-08-01:
### Printer Crashes at the beginning of the print

Hi all, I am converting an old Makerbot Replicator 2 to Marlin Firmware and everything works. The printer heats up, auto bed levels, and starts the print. However, after a couple minutes (usually after the 1st or 2nd layer), the printer crashes and stops. It does not continue printing and I have to restart it again. It crashed for every single print, I have not had any successful prints yet.

Here is a video showing the issue:
[Link to the Video](https://youtu.be/MzcrX3wcA-Q)
The camera was started right when the print started. After about 1.5 minutes, the printer crashes and the lcd screen freezes. 

Here are photos of a couple of prints that crashed:
[image]

### My Configurations

I have all of my code here: https://github.com/RosalieWessels/Marlin_MakerbotReplicator2

My models are sliced with Cura and printed in PLA. 

I tried hotend temperatures of 200, 210, and 220 degrees. My print speed is around 50 or 60 mm/s.

Here is a sample sliced file that was used:
[benchy4TEST.zip](https://github.com/MarlinFirmware/Marlin/files/5011380/benchy4TEST.zip)

**Expected behavior:** I expect it to finish the print.

**Actual behavior:** It crashes after a couple of minutes. The LCD Screen is also freezing.

thisiskeithb, 2020-08-02:
Please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

boelle, 2020-08-03:
@RosalieWessels 

Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

boelle, 2020-08-03:
Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

sjasonsmith, 2020-08-17:
@RosalieWessels do you have any update on this?

If you have a serial connection to the board it may provide some information prior to the crash.

RosalieWessels, 2020-08-18:
Hi all,

I was very busy so I didn't have a chance to look at it again. I will continue to work on it this week and see if I can upgrade to the bugfix or figure out where the problem is.

boelle, 2020-09-11:
@RosalieWessels what did you figure out?

closed 2020-09-23 by github-actions[bot]