MarlinFirmware/Marlin #18895: [FR] Set separate minimum planner speeds for outer/inner wall, infill, etc.

Liamnerf, 2020-08-02:
### Description

In the last few months I've been testing getting linear advance tuned as perfectly as possible, with junction deviation and direct drive. I've run what must be over 100 calibration patterns by now, and I've learnt a lot about how it behaves. 

The key thing I've learnt, at least with my printer, is that linear advance doesn't play nice if you set the slow speed in the calibration pattern to under 15mm/s, and the fast speed to over 25mm/s. With my printer, I want the outer and inner wall speeds to be slow and fast, respectively. Some example values that wouldn't break LA would be a speed range of 8-20mm/s for the outer wall, and a speed range of 20-60mm/s for the inner wall. I've run calibration patterns with these speeds and found that they produce consistent lines with no visible flaws, even upon very close inspection. 

To translate these settings over to practial printing, in the case of the outer wall, means setting the print speed to 20mm/s, and the minimum planner speed in the firmware to 8mm/s. However, setting the speed for the inner wall is where my issue arises. I can set the inner wall speed to 60mm/s, but the minimum planner speed must remain at 8mm/s for the outer wall. this results in a speed range of 8-60mm/s for the inner wall, which breaks linear advance. 

I could increase the minimum planner speed to 20mm/s and adjust the outer wall speed to, say, 30mm/s, but this means ringing, less consistent layers, poor performance on fine details, etc. on the other hand, I could drop the inner wall speed to 20mm/s, but that would lead to painfully slow prints. 

The solution to this issue would be to allow for the minimum planner speed to be set separately for the inner and outer walls. I imagine this would require a plugin for slicers as well. 

I could potentially do this classic jerk, but, in the case of the inner wall, to get a minimum speed of 20mm/s, I would need to set the jerk to 40mm/s, which in the case off a 180 degree direction change would keep the nozzle's speed at 20mm/s constantly. However, this jerk would be very high for my printer and would likely cause it to lose steps.

XDA-Bam, 2020-08-02:
Sorry, but I think you are misinterpreting the usage of `MINIMUM_PLANNER_SPEED`. Have you read the documentation? It says
```c++
// Minimum planner junction speed. Sets the default minimum speed the planner plans for at the end
// of the buffer and all stops. This should not be much greater than zero and should only be changed
// if unwanted behavior is observed on a user's machine when running at very slow speeds.
```
The value is essentially just a safe value which avoids division by zero and other oddities that might occur if you actually use zero speeds in the planner. Also, Marlin and the planner do not know, whether you are printing an outer wall, inner wall, skin or infilll at any given point in time. Marlin just sees "move from A to B with U mm/s, extrude E mm of filament".

The things you want to achieve are all handled by the slicer. At least in Cura you can define different speeds for outer walls, inner walls, etc. And you can fine tune much, much more in the slicer and it's way more flexible (no need to re-flash the firmware, etc.). This is not something you should do in Marlin.

Liamnerf, 2020-08-02:
> Sorry, but I think you are misinterpreting the usage of `MINIMUM_PLANNER_SPEED`. Have you read the documentation? It says
>
> ```c++
> // Minimum planner junction speed. Sets the default minimum speed [...]

I see, thanks for the info. Had had seen that, but I didn't fully understand it. My understanding of the minimum planner speed was based on this post I read:

https://reprap.org/forum/read.php?1,739819

In that link it's stated that if a junction is below a certain angle, the minimum planner speed is used for that corner. I realise this description for is for smoothieware, but I figured that their version and marlin's version of minimum planner speeds was virtually the same. Is this not the case? 

Please forgive my ignorance, I've not fully come to grips with how marlin works yet.

XDA-Bam, 2020-08-02:
`MINIMUM_PLANNER_SPEED` is also used in a similar way in Marlin for angles below `cos = 0.999999f`, which is about 0.08°. But besides that, it is used in some other places which are not related to JD. Therefore, you shouldn't change the defaults unless you have a very specific problem with your machine which can be traced to this parameter. I couldn't imagine what that problem would be, but nonetheless, it's a configurable parameter. It is not intended for what you are trying to achieve, though 😀 

Also keep in mind: If you increase the minimum speed too far - be it in Smoothieware or Marlin - you may exceed your machines maximum acceleration!

Liamnerf, 2020-08-02:
> `MINIMUM_PLANNER_SPEED` is also used in a similar way in Marlin for angles above `cos = 0.999999f`, which is less than about 0.08°. But besides that, it is used in some other places which are not related to JD. Therefore, you shouldn't change the defaults unless you have a very specific problem with your machine which can be traced to this parameter. I couldn't imagine what that problem would be, but nonetheless, it's a configurable parameter. It is not intended for what you are trying to achieve, though 😀
> 
> Also keep in mind: If you increase the minimum speed too far - be it in Smoothieware or Marlin - you may exceed your machines maximum acceleration!

thanks for the tips 🙂 Do you know of anyway I can achieve what I'm trying to do? To ensure that the print head speed never dips under 8mm/s? when printing?

XDA-Bam, 2020-08-02:
No, sorry. As I said, this technically may violate the acceleration limits of your machine. That's probably why something like this can't be configured in Marlin. As far as I know, you can only increase your acceleration limits and junction deviation somewhat, to avoid dipping below 8 mm/s too frequently. And set the minimum desired speed in your slicer, of course. In Cura, you should watch out for the minimum layer time, too - as that may force lower speeds on very short layers.

On smooth curves, even smaller ones, this speed should rarely be a problem. On 0° junctions (a reversing of the print direction), you will inevitably hit 0 mm/s at the junction. You can't avoid this anyway - it's physics.

Liamnerf, 2020-08-02:
> No, sorry. As I said, this technically may violate the acceleration limits of your machine. That's probably why something like this can't be configured in Marlin. As far as I know, you can only increase your acceleration limits and junction deviation somewhat, to avoid dipping below 8 mm/s too frequently. And set the minimum desired speed in your slicer, of course. In Cura, you should watch out for the minimum layer time, too - as that may force lower speeds on very short layers.
> 
> On smooth curves, even smaller ones, 8 mm/s should rarely be a problem. On 0° junctions (a reversing of the print direction), you will inevitably hit 0 mm/s at the junction. You can't avoid this anyway - it's physics.

I see. Would you happen to know how junction speed is calculated when using JD, or where I could find the maths please? If not I’ll try to find it online.

 I imagine that if I plug a worst case scenario junction angle, say 1 degree, into whatever the formula is, I can find the speed that that junction would be taken at - I then could adjust the JD and acceleration values to ensure that that speed is 8mm/s. As for a junction of 0 degrees, luckily I won’t have to worry about that, because that direction change would never happen while printing the walls, which is where maintaining linear advance matters. Does that idea sound feasible to you?

Liamnerf, 2020-08-02:
> I see. Would you happen to know how junction speed is calculated when using JD, or where I could find the maths please? If not I’ll try to find it online.
>
> I imagine that if I plug a worst case [...]

Nevermind about my question, I've found the equation online. happy days.

XDA-Bam, 2020-08-08:
Look [here](https://github.com/MarlinFirmware/Marlin/blob/fd9c6983d30f70171d9d9a9c55962689fa4cdd85/Marlin/src/module/planner.cpp#L2354) for the junction speed^2 and [here](https://github.com/MarlinFirmware/Marlin/blob/fd9c6983d30f70171d9d9a9c55962689fa4cdd85/Marlin/src/module/planner.h#L963) for the acceleration in that equation. The acceleration is at least euqal to the maximum accel of all axes currently moving and typically larger than that (due to the division by an element of a unit vector).

If this FR is obsolete now, please consider closing it.

Liamnerf, 2020-08-08:
Thanks for the info, I wasn’t able to find it myself. I will close this now.

closed 2020-08-08 by Liamnerf