MarlinFirmware/Marlin #18896: Printer feed rate percentage changes with every move the printer makes

mchaos, 2020-08-02:
### Bug Description

Whenever I print, with each move the printer makes the feed rate decreases untill the printer is at a crawl. If I run an m244 command back to 100% it willjump to 100%  and decrease as it makes moves. 

Latest bug fix. Btt sir es dip v1.1. octoprint.

### My Configurations

Will upload later. I am in the basement with printer and laptop up stairs.

### Steps to Reproduce

1. [First Step]  slice with prusa slic3r. Same config as always. Works fine on anet.
2. [Second Step] upload to octo
3. [and so on...] Print

**Expected behavior:** [What you expect to happen] damn thing not to slow down

**Actual behavior:** [What actually happens] ... Printer feed rate goes down a bit with each move.

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

mchaos, 2020-08-02:
[config.zip](https://github.com/MarlinFirmware/Marlin/files/5011508/config.zip)

mchaos, 2020-08-02:
I am trying pronterface and getting the same result. i keep running it and then all of a sudden it works right...

ellensp, 2020-08-02:
You have a lcd connected, does the LCD FR indicator also say FR is dropping?

Can you attache small gcode example, Just in case something weird is going on in gcode...

mchaos, 2020-08-02:
I don't have the lcd connected. I have not modded it to work yet.  It;s clear that it is because if I do an m220 s100 it speeds back up and slowly lowers a little with each move.

mchaos, 2020-08-02:
[test.zip](https://github.com/MarlinFirmware/Marlin/files/5011541/test.zip)

GMagician, 2020-08-02:
> G1 X110.000 Y119.368 F12000.000
> G1 F1800

ellensp, 2020-08-02:
that is just first_layer_speed = 30

GMagician, 2020-08-02:
I also see G1 F3415 in next layers

ellensp, 2020-08-02:
Try disabling unused zonestar lcd for now.

mchaos, 2020-08-02:
I did.. no change

mchaos, 2020-08-02:
So I'll try to do a vid later but let me explain it so everyone understands what it's doing. It's like every command executed the printer rate drops by 5% until it's moving at like 0 which is so slow.  Once I stop the print, it will continue to be at this slow speed until I run m220 s100 and it will run normal speed.

mchaos, 2020-08-02:
There's a setting in adv that sets the minimum feed rate. I'm going to try that at 100 ND see if it stops it from lowering so I can print until I figure this out

ellensp, 2020-08-02:
could you sprinkle M220 into your gcode (like perhaps after every gcode) and log output? see if  freedrate percentage is actually changing?  or just looks like it is from symptoms.

mchaos, 2020-08-03:
I can.. but its for sure changing. If I run a print, it will slow down every move and if I stop the print and I manually send a command to move, it will still move at the last speed rate until I m220 s100 or better.  But I will do it to rule it out.

mchaos, 2020-08-03:
Okay so there is something telling it to lower the feed rate.. Without even printing, it lowers on its own... every time I run m220 its lower... now this is weird.

mchaos, 2020-08-03:
So I it was late last night, and I don't remember for sure. I know I disabled the LCD in the firmware but I don't remember if that was after I was done for the night or if I installed it. It would make sense that a button thinks it is stuck, but for now I set the minimum speed to 100 in firmware and it seems to be staying at 100. I'll raise it up during print and see if it goes back down to 100. if it doesn't then I never loaded the FW after disabling LCD and that was it. if not, I have a work around.

mchaos, 2020-08-03:
son of a biscuit jointer.... Damn Zonestar LCD...  ... There is only one man who would dare give me the raspberry.... ZONE STAR...  That was it.. What a totally strange thing. Man it was driving me nuts... Feed rate stays where I put it after ensuring firmware with Zonestar LCD disabled.  Although it does say there is something wrong there.  Thanks everyone for helping me troubleshoot.

mchaos, 2020-08-03:
Oh yeah.. need to hit the close, thanks again..

closed 2020-08-03 by mchaos

ellensp, 2020-08-03:
I suspect this is happening as you don't have the zonestar connected.  But the controller doesn't know this.  So it happily keeps reading the ADC port. Every time it reads the ADC is it reads a down arrow. Since as far as the controller is concerned your on the main LCD the default action for a down arrow is to,  you guessed it, it lowers the feedrate!